WorldFamousElectronics/PulseSensorPlayground #20: BPM is 150+

eomsanghyun, 2017-09-19:
I use BPM alternative source code on arduino 101(intel curie) board.
BPM is not stable...
[image]
What is problem??
I want stable value. Please help me.. :(

joelSensor, 2017-11-13:
Please let me know what example sketch you are using?

yury-g, 2018-01-10:
eomsanghyun,

Yes, the PulseSensor works with the Arduino 101.  We just created a Playground Library, with lots of projects already coded out. Install our Library: 
https://github.com/WorldFamousElectronics/PulseSensorPlayground/

Then open the Example titled  "PulseSensor_BPM_Alternative"
[image]

closed 2018-01-13 by yury-g