WorldFamousElectronics/PulseSensorPlayground #29: Is there a way to simply get a numeric BPM

SparkMike77, 2017-12-22:
I'm looking for how to just pull the BPM number, not plot it on a graph but all the BPM sketches have a graphing output.

I'm assuming instead of calling pulseSensor.outputBeat() there's a way just to call for the numeric BPM?

joelSensor, 2017-12-22:
There is a way to just get BPM number.
Please check out the PulseSensor Playground Tools
https://github.com/WorldFamousElectronics/PulseSensorPlayground/blob/master/resources/PulseSenaor%20Playground%20Tools.md

a call to getBeatsPerMinute() will return the latest BPM value (int). You can simply comment out the call to pulseSensor.outputBeat() and just do you own thing instead.

yury-g, 2018-01-10:
SparkMike77, 
I drafted some code here for you.  It's not pretty, but it does the behavior you requested. 
Try this code in a new Sketch, or just run it as an ".ino" file in Arduino. 
[https://github.com/yury-g/ReadMePlayground/blob/master/PulseSensor_BPM_Serial_Print.ino](https://github.com/yury-g/ReadMePlayground/blob/master/PulseSensor_BPM_Serial_Print.ino
)

This is a screengrab of it working correctly: 
Let me know if it helps!
[image]

yury-g, 2018-01-13:
Mike, 
Try the sketch at this link. It is BPM and BPM-only via our new Library.  ; ) 

https://github.com/WorldFamousElectronics/PulseSensorPlayground/blob/master/examples/Getting_BPM_to_Monitor/Getting_BPM_to_Monitor

SparkMike77, 2018-02-06:
Thanks Guys, I had completely forgotten to check back and was hammering on my own library (really poorly).  This looks exactly like what I need!

haarers, 2018-07-18:
Is there somewhere that I can see the code for the function

getBeatsPerMinute()

Thanks!

biomurph, 2018-07-18:
The actual function is pretty simple, and it's on line 87 here
https://github.com/WorldFamousElectronics/PulseSensorPlayground/blob/master/src/utility/PulseSensor.cpp

The function that determines BPM is called processLatestSample, on line 122 of the same file.

closed 2018-07-18 by biomurph

haarers, 2018-07-18:
> master/src/utility/PulseSensor.cpp

Thank you, Joel.   I appreciate this.    -Shawn Haarer