WorldFamousElectronics/PulseSensorPlayground #40: STOP PulseSensor.

AiWOL, 2018-02-20:
Command PulseSensorPlayground.:begin() to START sequence.
How do I STOP the execution PulseSensor sequence?

biomurph, 2018-02-20:
Hi AiWOL,

That is a great feature request!
I'll see what I can do to incorporate that into our library.

bneedhamia, 2018-02-21:
AiWOL: Hi, it's so exciting that you're doing cool things with the Pulse Sensor. Can I ask a little about what you want to do? What is it you want to stop happening?

Thanks,
Brad

AiWOL, 2018-02-22:
After pressing the Button, the necessary parameters are measured and the data is calculated. After that it is necessary to stop the sensor to save battery.

-------------------------------------------------------------------------------------
'? ?????????. ????. ???????. ????????? ???, ? ??. ???? ?? ?? ?????????? - ?? ??????? ? '?????????'..."

biomurph, 2018-02-22:
@AiWOL Pulse Sensor only draws about 4mA when it's on. You could easily attach the Vdd pin (red wire) to any GPIO, and use digitalWrite HIGH and LOW to turn on and off the Pulse Sensor. 

@bneedhamia It still might be a good idea to look into what the risks and rewards are in having a software defined end() function. I know it can be tricky, but since the GPIO power is a workable fix there is not a huge hurry to do this.

rylRAE, 2019-04-15:
Hi guys, just want to ask if there's new updates on how to disable begin() for a while because it disrupts the GPS encoding when merged in the code.

biomurph, 2019-04-15:
@rylRAE 
I have some code that needs to be tested. It will likely be the end of the week before that is available, and we will be pushing a new version of the library then.

rylRAE, 2019-04-16:
That would be highly appreciated @biomurph . I do love the library and the sensor for its accuracy. I do hope to have the update soon. God bless guys and well done for you efforts!

biomurph, 2019-04-16:
In the mean time, you can use the PulseSensor_BPM_Alternative code example to start with.
in lieu of a stop() method, you can just not call the sawNewSample() since the timing of the signal acquisition is done in software.

biomurph, 2019-05-03:
@rylRAE 
I now have the library working with new methods
`pause()` 
`resume()`
`isPaused`
Please check out the [Toolbox](https://github.com/WorldFamousElectronics/PulseSensorPlayground/blob/master/resources/PulseSensor%20Playground%20Tools.md) for more info.
To use this version of the Playground, you will need to either clone or download from the repository main page, as it is not formally released yet. If you encounter any bugs, please open a new issue, as it will be concerning new code.

By the way, what Arduino platform are you targeting?

closed 2019-06-13 by biomurph

OliverosA, 2020-05-24:
Thank you very much, I had problems using the pulse sensor together with a fingerprint sensor, but when using the .pause () method of the pulse sensor when entering the fingerprint menu it works without any problem!