WorldFamousElectronics/PulseSensorPlayground #41: MKR1000 compatibility issue

mlavrador, 2018-02-21:
I think I found a compatibility issue between the PulseSensorPlayground library and the MKR1000. Different from the NANO, where the last version of PulseSensorPlayground (1.4.4) works perfectly with all provided examples, with MKR1000 I cannot compile examples whenever using PulseSensorPlayground versions above 1.2.3.
Please see the error below, when trying to compile "PullseSensor_BPM_Alternative" with 1.4.4:
-------------------------------------------------------------------------------------------

C:\Users\marce\OneDrive\Documents\Arduino\libraries\PulseSensor_Playground\src\PulseSensorPlayground.cpp: In member function 'int PulseSensorPlayground::getPulseAmplitude(int)':

C:\Users\marce\OneDrive\Documents\Arduino\libraries\PulseSensor_Playground\src\PulseSensorPlayground.cpp:213:5: error: return-statement with no value, in function returning 'int' [-fpermissive]

     return; // out of range.

     ^

C:\Users\marce\OneDrive\Documents\Arduino\libraries\PulseSensor_Playground\src\PulseSensorPlayground.cpp: In member function 'long unsigned int PulseSensorPlayground::getLastBeatTime(int)':

C:\Users\marce\OneDrive\Documents\Arduino\libraries\PulseSensor_Playground\src\PulseSensorPlayground.cpp:220:5: error: return-statement with no value, in function returning 'long unsigned int' [-fpermissive]

     return; // out of range.

     ^

exit status 1
Error compiling for board Arduino/Genuino MKR1000.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

bneedhamia, 2018-02-21:
Good catch!  The two errors are because there are return; statements that instead must return something. One solution would be to change those two return statements to return(-1); and document that those functions return -1 for sensor numbers out of range.

biomurph, 2018-02-22:
@mlavrador @bneedhamia 
Yep, I have a library update that will release and return a -1 in these cases.
Thanks for always making it better!!

biomurph, 2018-02-22:
Aaaaand done.
New release 1.4.5 fixes this bug.... methinks

closed 2018-02-22 by biomurph

mlavrador, 2018-02-22:
Almost there..
After updating to 1.4.5,  I have an error with the "-1". Please see below.

******

:\Users\marce\OneDrive\Documents\Arduino\libraries\PulseSensorPlayground\src\PulseSensorPlayground.cpp: In member function 'void PulseSensorPlayground::setThreshold(int, int)':

C:\Users\marce\OneDrive\Documents\Arduino\libraries\PulseSensorPlayground\src\PulseSensorPlayground.cpp:194:13: error: return-statement with a value, in function returning 'void' [-fpermissive]

     return -1; // out of range.

             ^

exit status 1
Error compiling for board Arduino/Genuino MKR1000.

biomurph, 2018-02-22:
Oh, holy crow. I over corrected!

biomurph, 2018-02-22:
Ok re released the release of the release.

sigh

That's what happens when you get sculptors managing code bases!

mlavrador, 2018-02-22:
Now works great!!!  Thank you very much!