WorldFamousElectronics/PulseSensorPlayground #44: getBeatsPerMinute is providing the o/p even when I am not touching the Sensor

jindalak, 2018-02-22:
Hi,
   I am using the I2C 20X4 display to show the heart beat and what is happening that, first it shows 0 as heard beat, then after sometime it changes to any number up to 200+ also, even when I don't touch the sensor. My expectation is that I should have the heart beat as 0 and when I touch the sensor, then it should print the correct heartbeat on the LCD. I have used the following code and without even touching the sensor I am getting o/p on the display.

I have not included the LCD code. I did search on the internet on this issue, but I could not find any concrete solution.

#define USE_ARDUINO_INTERRUPTS true
#include <PulseSensorPlayground.h>     

const int PulseWire = 0;
void setup()
{
  pulseSensor.analogInput(PulseWire);
  pulseSensor.setThreshold(Threshold);
  if (pulseSensor.begin()) {
    Serial.println("We created a pulseSensor Object !");  
  }
}
void loop()
{

  int myBPM = pulseSensor.getBeatsPerMinute();  
  if (pulseSensor.sawStartOfBeat()) {          
  Serial.println("♥  A HeartBeat Happened ! "); 
    Serial.print("BPM: ");                       
    Serial.println(myBPM);                        
  }
}

biomurph, 2018-02-22:
I can't really troubleshoot your code unless you provide it.
Also will need to see the humidity sensor code.
Please provide a link to git repository for your code.

jindalak, 2018-02-22:
Thanks a lot for a quick response. I am able to solve the issue for DHT22 sensor after using the cactus_io_DHT22. Earlier I was using the dht.h library. 

https://github.com/jindalak/Monitor.git

But I am still facing the issue for Heart Beat Rate

Thanks

biomurph, 2018-02-23:
Ok, I think I see a couple of issues.
First, don't use the line
Serial.println(analogRead(pulseWire);
The Pulse Sensor library takes care of the analog read part. Instead, use
pulseSensor.outputSample();
That will print the latest sample.

Second, you have a lot of serialprint lines in your code. I recommend you crank up your baud rate. Use something like 250000baud.

The DHT22 library you're using turns off interrupts to do it's readings, and then turns interrupts back on. This will cause an issue with the Pulse Sensor readings, since we're using interrupts to time the analog reads that we do. 

Here's a possible fix: Don't run the DHT22 functions so often. The humidity is not going to change that rapidly, and so you don't really need high resolution samples. That will just give you noise. 
Instead, read the DHT22 like every minute. Then when you're not reading the DHT22, run the Pulse Sensor. Now, we don't have an end() function in our library (yet), but the Pulse Sensor only draws about 4mA, so you can power it from one of your unused GPIOs. You can then turn the Pulse Sensor 'on' and 'off' by setting that pin HIGH and LOW. Further, you will likely want to use the BPM_Alternative code that we have, since it does not rely on interrupts. Using the BPM_Alternative code, you can 'turn on' and 'turn off' the Pulse Sensor sample reading by either calling or not calling sawNewSample(). When you're reading the Pulse Sensor, be aware that you can't do much else, because we are trying to time the readings at 500Hz (2mS). So any writes to the serial port or the lcd between Pulse Sensor analog reads will need to be quick. 

Anyhoo, I think what you need to do is isolate your Pulse Sensor and your DHT22 activity from eachother so they don't cause problems.

Hope that helps!

jindalak, 2018-02-24:
Thanks a lot for your reply. I have cleaned up the code and put all Serial print statements under the debug if condition, so it will not run by default now.

I am in a process of implementing a switch between the PulseSensor and DHT, so that just one will be on for now.

But I am having hard time understanding/getting the pulse sensor only show a reading only if there is a pulse.

I have modified some examples for only pulse sensor and have these in my git repository. I don't know what am I missing.

https://github.com/jindalak/Monitor.git

I think if I separate out these two I will not have DHT interfere the Pulse. But I need to get the Pulse started working correctly. Please let me know if I am missing anything.

wramang, 2019-02-18:
same problem i can't make 0 BPM when my finger not touch sensor

biomurph, 2019-02-19:
@wramang 
This issue is a year old. Your comment lacks enough information to even begin to troubleshoot.
Please open a new issue and use your words to clarify the problem fully.
What's your hardware? What code are you using? etc..

closed 2019-02-19 by biomurph