WorldFamousElectronics/PulseSensorPlayground #45: PulseSensorPlayground does not work on Arduino Mega 2650

Williamvdb, 2018-03-21:
I am using your example sketch called Getting_BPM_to_Monitor, it works on my Arduino UNO, but not on my Arduino Mega 2560. Why is that? 

It does not even Serial print "We created a pulseSensor Object!" indicating that pulseSensor.begin() failed. The Serial Monitor is completely empty.

biomurph, 2018-03-22:
We don't yet support use of interrupts on the Mega. You will need to run the BPM_Alternative code example. If all you want is BPM on the serial port, then you will need to add a variable at the top of the sketch 

`int myBPM;`

Then modify part of the code in the loop to be something like this

`if (pulseSensor.sawStartOfBeat()) {
        myBPM = pulsesensor.getBeatsPerMinute();
        Serial.print("BPM: ");
        Serial.println(myBPM);
      }`

bhavindave9, 2018-05-08:
so can we expect the support for Mega anytime soon? would love to get my hand dirty on.

bhavindave9, 2018-05-08:
I checked the arduino mega 2560 data sheet .Everything (timer section that we are using) is same as uno.
so what is the problem here? how we can use this existing code on mega using interrupts . would really appreciate if you can help .

bhavindave9, 2018-05-09:
I will definitely try it and will let you know the results

VANHUUTHANHTRUNG, 2019-03-08:
I have the same issue, did you fix this?

  If I set `#define USE_ARDUINO_INTERRUPTS true` :
  I can run the program without error but it return nothing on the monitor and plotter

  If I set `#define USE_ARDUINO_INTERRUPTS false` :
  The program just return straight lines which means it can not measure anything.

I tried to return raw value from the sensor, it was fine so there is no problem with the sensor.
If I want to use functions in PulseSensorPlayground.h to get BPM, how should I do it? I tried to figure something in PulseSensorPlayground.h and PulseSensorPlayground.cpp but it's too hard for me.

biomurph, 2019-03-08:
I think I have a mega2650around here somewhere. I'll see if I can replicate your problem.

luto65, 2019-04-28:
I have also a MEGA and still having the same problem

biomurph, 2019-04-29:
I will be able to fix the issue this week.

biomurph, 2019-04-30:
@luto65 @VANHUUTHANHTRUNG @bhavindave9 

The latest release of PulseSensor Playground supports Arduino MEGA boards. Please let us know if you run into any bugs. I've tested it with and without including the Servo library, though have not used with servo on MEGA. 
You can update your library through the Arduino Library manager. Latest is version 1.4.13

closed 2019-04-30 by biomurph

luto65, 2019-05-05:
I tried it today and unfortunately does not work still.