WorldFamousElectronics/PulseSensorPlayground #52: Errors in compiling the library for Adafruit Feather M0

ejams, 2018-05-14:
I'm trying to use the pulse sensor library with an Adafruit Feather M0 but there are various issues when compiling; I have attached the error output from Arduino IDE.
[Errors.txt](https://github.com/WorldFamousElectronics/PulseSensorPlayground/files/2002688/Errors.txt)

biomurph, 2018-05-16:
@ejams 
I think I fixed it, but I don't actually have a Feather M0. I have a Feather ESP...
I did adjust the code to accommodate the M0, and I think it works. 
At least it compiles :)
Please download the Playground library from this repo into your Arduino/libraries folder, and then give it a shot. You should use the BPM_Alternative example. Let me know how it goes.

ejams, 2018-05-22:
@biomurph I don't actually have access to the Feather M0 I was using anymore as I was on a tight schedule and completed the project with an Uno instead. It sounds like you fixed it though, so you can probably close the issue now but I appreciate the quick action to get it working and I'll give it a shot in the future if I get a chance :+1:

closed 2018-05-22 by biomurph