WorldFamousElectronics/PulseSensorPlayground #57: PulseSensor with arm based device?

SalvatoreLicitra, 2018-05-28:
Hi I want to ask you if there is possible to use the library and the examples with a stm32 board (nucleo F401RE) because I saw that the libraries uses the Arduino avr interrupts, but my board is an Arm Cortex one so I cannot use the functions "cli()" and "sei()"; I already try to use "__enable_irq()" and "__disable_irq()" with the "Getting_BPM_to_Monitor" example, but the program never receive the BPM.
Is there something that i can do?

SalvatoreLicitra, 2018-05-29:
Ok, thanks very much,!

biomurph, 2018-05-30:
Hi,

I'm able to get the Nucleo to program with Arduino, but I don't have the serial port working yet. Do you have any info on how to get the Nucleo to output serial over the USB port? If not, I will have to dig up an FTDI cable to make a go of it.

SalvatoreLicitra, 2018-05-30:
Hi,
I'm not sure to understood the question. When the Nucleo board are connected via usb, computer automatically assign the serial port to this, if you use windows you can see that using the panel "device management"; if you don't see the (COM) port assignment you can try to fix it installing the drivers. If the problem is that the code doesn't compile because an error like "Nucleo-64 doesn't recognized" or similar one check if the serial port (COMx if you use windows) is the correct one and select "STLink" as "upload method".

biomurph, 2018-05-30:
No, I get that the Nucleo is connected, but at least in my board files the
USB is not used as the serial connection. It uses digital IO pins instead.
I have an FTDI to get it working. I will try in a bit and let you know the
result.

Joel
~~~
Making the world safe for robots
www.pulsesensor.com
www.openbci.com

On Wed, May 30, 2018 at 2:19 PM, Salvatore Licitra <notifications@github.com
> wrote:

> Hi,
> I'm not sure to understood the question. When the Nucleo board are
> connected via usb, computer automatically assign the serial port to this,
> if you use windows you can see that using the panel "device management"; if
> you don't see the (COM) port assignment you can try to fix it installing
> the drivers. If the problem is that the code doesn't compile because an
> error like "Nucleo-64 doesn't recognized" or similar one check if the
> serial port (COMx if you use windows) is the correct one and select
> "STLink" as "upload method".
>
> —
> You are receiving this because you commented.
> Reply to this email directly, view it on GitHub
> <https://github.com/WorldFamousElectronics/PulseSensorPlayground/issues/57#issuecomment-393265667>,
> or mute the thread
> <https://github.com/notifications/unsubscribe-auth/AF_jGfMqcj2pm1yfyMRTaKzLgHam5Ebxks5t3uKwgaJpZM4UPf3A>
> .
>

SalvatoreLicitra, 2018-05-30:
Ok, I'm waiting for your updates.

biomurph, 2018-06-04:
@SalvatoreLicitra 
I updated the library so that it will work with STM32 boards. There is an example sketch that targets a Nucleo F401RE board. Please check the comments at the top of the sketch for info on how to configure the board parameters in Arduino. I used an FTDI brekaout to make the serial connection, and I found that I could see the Pulse Sensor signal, BPM, and IBI data clearly.
Make sure that you power the Pulse Sensor with the same voltage that your Nucleo power jumper is set to.

SalvatoreLicitra, 2018-06-04:
Hi, i tried the example and it works, thanks a lot!

closed 2018-06-04 by SalvatoreLicitra