WorldFamousElectronics/PulseSensorPlayground #61: pulseSensor.sawStartOfBeat is always false

Tvde1, 2018-06-02:
Here is my (short) sketch:

```
#define USE_ARDUINO_INTERRUPTS false
#include <PulseSensorPlayground.h>

PulseSensorPlayground pulseSensor;

void setup() {
	Serial.begin(115200);

	pulseSensor.analogInput(A0);

	pulseSensor.setSerial(Serial);
	pulseSensor.setOutputTypeSERIAL_PLOTTER);
	pulseSensor.setThreshold(550);

	samplesUntilReport = SAMPLES_PER_SERIAL_SAMPLE;

	if (!pulseSensor.begin()) {
		Serial.println("Cannot begin.");
	}
}

void loop() {
	if (pulseSensor.sawStartOfBeat()) {
		pulseSensor.outputBeat();
	}
}
```

This will never output anything while previously using `pulseSensor.outputSample();` would output valid bpm and a nice heartbeat to the graph.

I tested this with the serial monitor on an ESP8266.

biomurph, 2018-06-03:
Try running the BPM_Alternative example sketch from the Playground.
I'm  pretty sure you need to run the sawNewSample() function in order to get the sawStartOfBeat() to work right.

Tvde1, 2018-06-03:
I just put it in there in the loop?

biomurph, 2018-06-04:
First, use the example BPM_Alternative code, and then work from that if you need to make any adjustments.

closed 2018-06-05 by biomurph