WorldFamousElectronics/PulseSensorPlayground #63: Connect PulseSensor to nRF51822

francesco98, 2018-06-07:
Hi,

I want to connect the Pulse Sensor to nRF51822. Is it possible?

Thank you

biomurph, 2018-06-07:
I have not used the nRF51822 directly, however I have used the [Simblee](https://www.simblee.com/) and [RFduino](http://www.rfduino.com/) made by RFDigital (now bought up by someone else). Both of those are built on the nRF51822 glass. There should be no problem connecting the Pulse Sensor to one of the analog pins. You will likely want to use the BPM_Alternative code example if you are going to run BLE or Gazelle stack, as the radio will stomp on your interrupts (in my experience). I don't see any issue with using either BLE or Gazelle to send the data over air. What you will have to do is buffer the samples (BLE allows 20 bytes per radio packet, Gazelle allows 32 bytes per radio packet). Pulse Sensor samples at 500Hz, 2mS, so you should be fine if you collect samples to fill the packet, and then send over air. On the other side, you will have to unpack, but that's trivial.

francesco98, 2018-06-07:
Thanks for your support. I'm using the alternative code, but I'm receiving these errors:
- `error: 'cli' was not declared in this scope`
- `error: 'sei' was not declared in this scope`

How can I resolve? Thanks.

biomurph, 2018-06-07:
Ah, ok. Are you using the Arduino IDE? If so, what board are you targeting?

francesco98, 2018-06-07:
The target is: **Generic nRF1**

biomurph, 2018-06-07:
Can you link me to the json for the board cores?

francesco98, 2018-06-07:
Where can I find it? I'm using this: [https://github.com/sandeepmistry/arduino-nRF5](https://github.com/sandeepmistry/arduino-nRF5)

francesco98, 2018-06-07:
Ok thanks, but I have another problem..
I'm connecting the pulse sensor to ATTiny85, but I'm receiving this error:
`PulseSensorPlayground.cpp:28: undefined reference to `operator new[](unsigned int)'`

Sorry for my many question..

biomurph, 2018-06-08:
That seems a strange error. I am able to compile targeting a Gemma board. What board are you using?
ATTiny85 don't have any hardware serial port, so you need to use the Software Serial Demo example code to get started.

francesco98, 2018-06-09:
I have solved by replacing `new PulseSensor[SensorCount]` with `(PulseSensor*) malloc(sizeof(PulseSensor))`

biomurph, 2018-06-09:
Great! We'll look into rolling that change in the next release.

closed 2018-06-09 by biomurph