WorldFamousElectronics/PulseSensorPlayground #71: teensy 3.2

stephanschulz, 2018-09-20:
love the sensor and you library. thank you.

do you by any chance have plans to make the interrupt version work for the teensy 3.2 ?

I will give it a try myself using the IntervalTimer that ships with the teensy. But I thought you people might have something already in the pipeline.

thanks.

biomurph, 2018-09-21:
It would be great to have the interrupt happen on the Teensy.
I don't have the bandwidth right now to get into it, but if you take a stab at it I can do an assist and we can roll it into the Playground and credit your work. What say you?

biomurph, 2019-03-18:
@stephanschulz 
I have taken some stabs at upgrading the Pulse Sensor Playground library to support interrupts on Teensy, but still not quite there yet.
For now, please use the `PulseSensor_BPM_Alternative` code example.

closed 2019-03-18 by biomurph