WorldFamousElectronics/PulseSensorPlayground #72: pulse sesnsor does not work on ESP-WROOM-32

haroonahmad, 2018-10-08:
Plzz help me to resolve the issue.  MY ESP32 is not working.

biomurph, 2018-10-09:
@haroonahmad 
We need more information from you in order to help you.
Please tell us how you are connecting the Pulse Sensor to the ESP-WROOM-32?
What is the code that you are using?
Is there any documentation that you can send about the ESP module that you are using? Example code that you can share?

biomurph, 2019-06-13:
@haroonahmad 
The main issue that I've found with the ESP32 based boards is that the analogWrite function is not supported yet. The ESP32 does have 2 DAC pins, so I assume that eventually it will be fixed.
Until then, I have updated the library to ignore the analogWrite functions when compiling for the ESP32 architecture.

closed 2019-06-13 by biomurph