WorldFamousElectronics/PulseSensorPlayground #73: Does not work with ESP32 NodeMCU board

mujtabachang, 2018-10-23:
Are there any plans to make this library with ESP32?

dnguyenv, 2019-02-14:
Same issue on my ESP32S
```
Users/duynguyen/Documents/Arduino/libraries/PulseSensor_Playground/src/utility/PulseSensor.cpp: In member function 'void PulseSensor::initializeLEDs()':
/Users/duynguyen/Documents/Arduino/libraries/PulseSensor_Playground/src/utility/PulseSensor.cpp:215:27: error: 'analogWrite' was not declared in this scope
     analogWrite(FadePin, 0); // turn off the LED.
                           ^
/Users/<user>/Documents/Arduino/libraries/PulseSensor_Playground/src/utility/PulseSensor.cpp: In member function 'void PulseSensor::updateLEDs()':
/Users/user/Documents/Arduino/libraries/PulseSensor_Playground/src/utility/PulseSensor.cpp:225:48: error: 'analogWrite' was not declared in this scope
     analogWrite(FadePin, FadeLevel / FADE_SCALE);
                                                ^
```

dnguyenv, 2019-02-14:
Forgot to post the code. The code I use is from Examples >> PulseSensor Playground >> Pulsensor_BPM_Alternative. I tried Examples >> PulseSensor Playground >> Pulsensor_BPM as well, no luck.

```
/*
   Sketch to handle each sample read from a PulseSensor.
   Typically used when you don't want to use interrupts
   to read PulseSensor voltages.

   Here is a link to the tutorial that discusses this code
   https://pulsesensor.com/pages/getting-advanced

   Copyright World Famous Electronics LLC - see LICENSE
   Contributors:
     Joel Murphy, https://pulsesensor.com
     Yury Gitman, https://pulsesensor.com
     Bradford Needham, @bneedhamia, https://bluepapertech.com

   Licensed under the MIT License, a copy of which
   should have been included with this software.

   This software is not intended for medical use.
*/

/*
   Every Sketch that uses the PulseSensor Playground must
   define USE_ARDUINO_INTERRUPTS before including PulseSensorPlayground.h.
   Here, #define USE_ARDUINO_INTERRUPTS false tells the library to
   not use interrupts to read data from the PulseSensor.

   If you want to use interrupts, simply change the line below
   to read:
     #define USE_ARDUINO_INTERRUPTS true

   Set US_PS_INTERRUPTS to false if either
   1) Your Arduino platform's interrupts aren't yet supported
   by PulseSensor Playground, or
   2) You don't wish to use interrupts because of the side effects.

   NOTE: if US_PS_INTERRUPTS is false, your Sketch must
   call pulse.sawNewSample() at least once every 2 milliseconds
   to accurately read the PulseSensor signal.
*/

#define USE_ARDUINO_INTERRUPTS false
#include <PulseSensorPlayground.h>

/*
   The format of our output.

   Set this to PROCESSING_VISUALIZER if you're going to run
    the Processing Visualizer Sketch.
    See https://github.com/WorldFamousElectronics/PulseSensor_Amped_Processing_Visualizer

   Set this to SERIAL_PLOTTER if you're going to run
    the Arduino IDE's Serial Plotter.
*/
const int OUTPUT_TYPE = SERIAL_PLOTTER;

/*
   Pinout:
     PULSE_INPUT = Analog Input. Connected to the pulse sensor
      purple (signal) wire.
     PULSE_BLINK = digital Output. Connected to an LED (and 220 ohm resistor)
      that will flash on each detected pulse.
     PULSE_FADE = digital Output. PWM pin onnected to an LED (and resistor)
      that will smoothly fade with each pulse.
      NOTE: PULSE_FADE must be a pin that supports PWM.
       If USE_INTERRUPTS is true, Do not use pin 9 or 10 for PULSE_FADE,
       because those pins' PWM interferes with the sample timer.
*/
const int PULSE_INPUT = A0;
const int PULSE_BLINK = 13;    // Pin 13 is the on-board LED
const int PULSE_FADE = 5;
const int THRESHOLD = 550;   // Adjust this number to avoid noise when idle

/*
   samplesUntilReport = the number of samples remaining to read
   until we want to report a sample over the serial connection.

   We want to report a sample value over the serial port
   only once every 20 milliseconds (10 samples) to avoid
   doing Serial output faster than the Arduino can send.
*/
byte samplesUntilReport;
const byte SAMPLES_PER_SERIAL_SAMPLE = 10;

/*
   All the PulseSensor Playground functions.
*/
PulseSensorPlayground pulseSensor;

void setup() {
  /*
     Use 115200 baud because that's what the Processing Sketch expects to read,
     and because that speed provides about 11 bytes per millisecond.

     If we used a slower baud rate, we'd likely write bytes faster than
     they can be transmitted, which would mess up the timing
     of readSensor() calls, which would make the pulse measurement
     not work properly.
  */
  Serial.begin(115200);

  // Configure the PulseSensor manager.
  pulseSensor.analogInput(PULSE_INPUT);
  pulseSensor.blinkOnPulse(PULSE_BLINK);
  pulseSensor.fadeOnPulse(PULSE_FADE);

  pulseSensor.setSerial(Serial);
  pulseSensor.setOutputType(OUTPUT_TYPE);
  pulseSensor.setThreshold(THRESHOLD);

  // Skip the first SAMPLES_PER_SERIAL_SAMPLE in the loop().
  samplesUntilReport = SAMPLES_PER_SERIAL_SAMPLE;

  // Now that everything is ready, start reading the PulseSensor signal.
  if (!pulseSensor.begin()) {
    /*
       PulseSensor initialization failed,
       likely because our Arduino platform interrupts
       aren't supported yet.

       If your Sketch hangs here, try changing USE_PS_INTERRUPT to false.
    */
    for(;;) {
      // Flash the led to show things didn't work.
      digitalWrite(PULSE_BLINK, LOW);
      delay(50);
      digitalWrite(PULSE_BLINK, HIGH);
      delay(50);
    }
  }
}

void loop() {

  /*
     See if a sample is ready from the PulseSensor.

     If USE_INTERRUPTS is true, the PulseSensor Playground
     will automatically read and process samples from
     the PulseSensor.

     If USE_INTERRUPTS is false, this call to sawNewSample()
     will, if enough time has passed, read and process a
     sample (analog voltage) from the PulseSensor.
  */
  if (pulseSensor.sawNewSample()) {
    /*
       Every so often, send the latest Sample.
       We don't print every sample, because our baud rate
       won't support that much I/O.
    */
    if (--samplesUntilReport == (byte) 0) {
      samplesUntilReport = SAMPLES_PER_SERIAL_SAMPLE;

      pulseSensor.outputSample();

      /*
         At about the beginning of every heartbeat,
         report the heart rate and inter-beat-interval.
      */
      if (pulseSensor.sawStartOfBeat()) {
        pulseSensor.outputBeat();
      }
    }

    /*******
      Here is a good place to add code that could take up
      to a millisecond or so to run.
    *******/
  }

  /******
     Don't add code here, because it could slow the sampling
     from the PulseSensor.
  ******/
}
```

biomurph, 2019-02-14:
@dnguyenv @mujtabachang 

The error that you're getting on compilation is because the ESP does not use the `analogWrite` function.
In order to fix the library, this will take a bit of time. Since the analogWrite is not critical to the function of the code, I recommend simply commenting out the portion that uses it.

In the library file `PulseSensor Playground/src/utility/PulseSensor.cpp`
find the section below and comment out the same lines that I have commented out with the `//`.
This part of the code is located at the bottom of the file.

void PulseSensor::initializeLEDs() {
  if (BlinkPin >= 0) {
    pinMode(BlinkPin, OUTPUT);
    digitalWrite(BlinkPin, LOW);
  }
  if (FadePin >= 0) {
//    pinMode(FadePin, OUTPUT);
//    analogWrite(FadePin, 0); // turn off the LED.
  }
}

void PulseSensor::updateLEDs() {
  if (BlinkPin >= 0) {
    digitalWrite(BlinkPin, Pulse);
  }

  if (FadePin >= 0) {
//    analogWrite(FadePin, FadeLevel / FADE_SCALE);
  }
}

I don't have an ESP32 in front of me, but I will have tomorrow and next week to troubleshoot this further. Please let me know how it works for you.

dnguyenv, 2019-02-23:
@biomurph commenting out those lines makes the compilation successful. Have you had a chance to trouble shoot the problem? Please let me know if you have updates on the code. Thanks

biomurph, 2019-02-25:
Based on the ESP32 documentation the methods used to drive PWM to fade an LED are quite different from the standard arduino analogWrite function. For now, we will recommend commenting out the lines in the library. Next step is to bracket them with some code that determines the hardware target and do it automatically.

closed 2019-02-25 by biomurph

Marcus-Peterson, 2024-01-10:
I commented out the code according to biomurphs suggestion, it didn't help. I still got the same identical error message.

`\Arduino\libraries\PulseSensor_Playground\src/utility/Interrupts.h:328:5: error: 'sampleTimer' was not declared in this scope
     sampleTimer.stop();`
     
 `\Arduino\libraries\PulseSensor_Playground\src/utility/Interrupts.h:402:5: error: 'sampleTimer' was not declared in this scope
     sampleTimer.start();`

biomurph, 2024-01-10:
@Marcus-Peterson 
This issue is closed. Please create a new issue.

Also, please try this repository branch called 'V2.0 beta' with no modifications and see if that works?