WorldFamousElectronics/PulseSensorPlayground #74: MKR1000_BPM_ALTERNATIVE :Edit

biojisoo, 2018-11-29:
Hello! I test with this sensor with MKR1000

I understood that BPM_ALTERNATIVE code can print BPM on serial monitor

with MKR1000 but I want to see only BPM.. 

I thought that I need to edit pulsesensorserial.cpp to print BPM(not with IBI etc)

so I edited like

/*
   Formatting of Serial output from PulseSensors.
   See https://www.pulsesensor.com to get started.

   Copyright World Famous Electronics LLC - see LICENSE
   Contributors:
     Joel Murphy, https://pulsesensor.com
     Yury Gitman, https://pulsesensor.com
     Bradford Needham, @bneedhamia, https://bluepapertech.com

   Licensed under the MIT License, a copy of which
   should have been included with this software.

   This software is not intended for medical use.
*/
#include "PulseSensorSerialOutput.h"

PulseSensorSerialOutput::PulseSensorSerialOutput() {
  pOutput = NULL;
  OutputType = SERIAL_PLOTTER;
}

void PulseSensorSerialOutput::setSerial(Stream &output) {
  pOutput = &output;
}

Stream *PulseSensorSerialOutput::getSerial() {
  return pOutput;
}

void PulseSensorSerialOutput::setOutputType(byte outputType) {
  OutputType = outputType;
}

void PulseSensorSerialOutput::outputSample(PulseSensor sensors[], int numSensors) {
  if (!pOutput) {
    return;  // no serial output object has been set.
  }

  switch (OutputType) {
    case SERIAL_PLOTTER:
      if (numSensors == 1) {
        pOutput->print(sensors[0].getBeatsPerMinute());
       // pOutput->print(F(","));
      //  pOutput->print(sensors[0].getInterBeatIntervalMs());
        pOutput->print(F(","));
        pOutput->println(sensors[0].getLatestSample());
      } else {
        for (int i = 0; i < numSensors; ++i) {
          if (i != 0) {
            pOutput->print(F(","));
          }
          pOutput->print(sensors[i].getLatestSample());
          // Could output BPM and IBI here.
        }
        pOutput->println();
      }
      break;

    case PROCESSING_VISUALIZER:
      // Don't print bpm and ibi here; they're printed per-beat.
      if (numSensors == 1) {
        outputToSerial('S', sensors[0].getLatestSample());
      } else {
        // PulseSensor 0 = a; #1 = b; #2 = c, etc.
        for(int i = 0; i < numSensors; ++i){
          outputToSerial('a' + i, sensors[i].getLatestSample());
        }
      }
      break;

    default:
      // unknown output type: no output
      break;
  }
}

void PulseSensorSerialOutput::outputBeat(PulseSensor sensors[], int numSensors, int sensorIndex) {
  if (!pOutput) {
    return;  // no serial output object has been set.
  }

  switch (OutputType) {
    case SERIAL_PLOTTER:
      /*
         The plotter doesn't understand occasionally-printed data,
         so we print nothing per-beat.
      */
      break;

    case PROCESSING_VISUALIZER:
      if (numSensors == 1) {
        outputToSerial('B', sensors[sensorIndex].getBeatsPerMinute());
       // outputToSerial('Q', sensors[sensorIndex].getInterBeatIntervalMs());
      } else {
        // PulseSensor 0 = A, M; #1 = B, N; etc.
        outputToSerial('A' + sensorIndex
          , sensors[sensorIndex].getBeatsPerMinute());
        //outputToSerial('M' + sensorIndex
          //, sensors[sensorIndex].getInterBeatIntervalMs());
      }
      break;

    default:
      // unknown output type: no output
      break;
  }
}

    // // testing feedback
    // void printThreshSetting() {
    //
    // }

void PulseSensorSerialOutput::outputToSerial(char symbol, int data) {
  if (!pOutput) {
    return;  // no serial output object has been set.
  }

  pOutput->print(symbol);
  pOutput->println(data);
}
===============================================

but it still prints all in serial monitor.. 

How can I get only BPM value from it to store value in one variable& print in serial monitor?

biojisoo, 2018-11-29:
Oh..! I forgot to check one thing..lol..

I'm editing pulsesensorground.cpp like

/*
   A central Playground object to manage a set of PulseSensors.
   See https://www.pulsesensor.com to get started.

   Copyright World Famous Electronics LLC - see LICENSE
   Contributors:
     Joel Murphy, https://pulsesensor.com
     Yury Gitman, https://pulsesensor.com
     Bradford Needham, @bneedhamia, https://bluepapertech.com

   Licensed under the MIT License, a copy of which
   should have been included with this software.

   This software is not intended for medical use.
*/
#include <PulseSensorPlayground.h>

// Define the "this" pointer for the ISR
PulseSensorPlayground *PulseSensorPlayground::OurThis;

PulseSensorPlayground::PulseSensorPlayground(int numberOfSensors) {
  // Save a static pointer to our playground so the ISR can read it.
  OurThis = this;

  // Dynamically create the array to minimize ram usage.
  SensorCount = (byte) numberOfSensors;
  Sensors = new PulseSensor[SensorCount];

#if PULSE_SENSOR_TIMING_ANALYSIS
  // We want sample timing analysis, so we construct it.
  pTiming = new PulseSensorTimingStatistics(MICROS_PER_READ, 500 * 30L);
#endif // PULSE_SENSOR_TIMING_ANALYSIS
}

boolean PulseSensorPlayground::PulseSensorPlayground::begin() {

  for (int i = 0; i < SensorCount; ++i) {
    Sensors[i].initializeLEDs();
  }

  // Note the time, for non-interrupt sampling and for timing statistics.
  NextSampleMicros = micros() + MICROS_PER_READ;

  SawNewSample = false;

#if PULSE_SENSOR_MEMORY_USAGE
  // Report the RAM usage and hang.
  printMemoryUsage();
  for (;;);
#endif // PULSE_SENSOR_MEMORY_USAGE

  // Lastly, set up and turn on the interrupts.

  if (UsingInterrupts) {
    if (!PulseSensorPlaygroundSetupInterrupt()) {
      // The user requested interrupts, but they aren't supported. Say so.
      return false;
    }
  }

  return true;
}

void PulseSensorPlayground::analogInput(int inputPin, int sensorIndex) {
  if (sensorIndex != constrain(sensorIndex, 0, SensorCount)) {
    return; // out of range.
[... 167 lines cut ...]
  int	data_size	=	(int)&__data_end - (int)&__data_start;
  int	bss_size	=	(int)&__bss_end - (int)&__data_end;
  int	heap_end	=	(int)&stack - (int)&__malloc_margin;
  int	heap_size	=	heap_end - (int)&__bss_end;
  int	stack_size	=	RAMEND - (int)&stack + 1;
  int	available	=	(RAMEND - (int)&__data_start + 1);
  available	-=	data_size + bss_size + heap_size + stack_size;

  Stream *pOut = SerialOutput.getSerial();
  if (pOut) {
    pOut->print(F("data "));
    pOut->println(data_size);
    pOut->print(F("bss "));
    pOut->println(bss_size);
    pOut->print(F("heap "));
    pOut->println(heap_size);
    pOut->print(F("stack "));
    pOut->println(stack_size);
    pOut->print(F("total "));
    pOut->println(data_size + bss_size + heap_size + stack_size);
  }
}
#endif // PULSE_SENSOR_MEMORY_USAGE

and I will check it..!! sorry..!!

biojisoo, 2018-11-29:
after I did edit like ↑, there were error like

arduino:1.8.8 Hourly Build 2018/11/27 05:33 (Windows 10), board:"Arduino/Genuino MKR1000"

C:\Users\jisoo\Documents\Arduino\libraries\PulseSensorPlayground-master\src\PulseSensorPlayground.cpp: In member function 'void PulseSensorPlayground::outputBeat(int)':

C:\Users\jisoo\Documents\Arduino\libraries\PulseSensorPlayground-master\src\PulseSensorPlayground.cpp:204:34: error: no matching function for call to 'PulseSensorSerialOutput::outputBeat(PulseSensor*&)'

   SerialOutput.outputBeat(Sensors);

                                  ^

C:\Users\jisoo\Documents\Arduino\libraries\PulseSensorPlayground-master\src\PulseSensorPlayground.cpp:204:34: note: candidate is:

In file included from C:\Users\jisoo\Documents\Arduino\libraries\PulseSensorPlayground-master\src/PulseSensorPlayground.h:122:0,

                 from C:\Users\jisoo\Documents\Arduino\libraries\PulseSensorPlayground-master\src\PulseSensorPlayground.cpp:16:

C:\Users\jisoo\Documents\Arduino\libraries\PulseSensorPlayground-master\src/utility/PulseSensorSerialOutput.h:72:10: note: void PulseSensorSerialOutput::outputBeat(PulseSensor*, int, int)

     void outputBeat(PulseSensor sensors[], int numberOfSensors, int sensorIndex);

          ^

C:\Users\jisoo\Documents\Arduino\libraries\PulseSensorPlayground-master\src/utility/PulseSensorSerialOutput.h:72:10: note:   candidate expects 3 arguments, 1 provided

exit status 1
board Arduino/Genuino MKR1000 compile error

how can I modify code to do exactly?

biomurph, 2018-11-29:
@biojisoo 
Woah, you really went down into a rabbit hole there! No need to edit the library files. While it's great that you can, it's much easier to get what you want in the sketch. Since all you want to do is print the BPM, use the function call `getBeatsPerMinute()` which will return the latest BPM as an int variable. Starting with the example sketch, just edit that so that you create a variable to receive the BPM, and then print that. You can also use a call to `sawStartOfBeat()` which will return **true** if there is a beat so that you always get the latest and greatest BPM.

You can find that function and other super cool functions in the [PulseSensor Toolbox](https://github.com/WorldFamousElectronics/PulseSensorPlayground/blob/master/resources/PulseSensor%20Playground%20Tools.md)

closed 2018-11-29 by biomurph

biojisoo, 2018-12-01:
Thank you ! I will try it! After it, I will post it on here!

harryhawkes, 2019-10-18:
biojisoo, did you ever get this to work? I am currently trying to get the BPM value to store as a variable with the MKR1000 with their BPM_Alternative script. If so could you post script for this? or even let me know what had to be altered in the example script file? Much appreciated, thanks. This is for a university project that i am trying to make

harryhawkes, 2019-10-18:
@biojisoo