WorldFamousElectronics/PulseSensorPlayground #77: Not working with Beetle Bluno

Virgilius1995, 2019-02-06:
I've tried a couple of versions of your playground as well as switching between A0 and A1 and varying the threshold between 10 and 1024 without any luck. Any ideas?

biomurph, 2019-02-07:
Can you give me some more details about what you mean by 'not working'?
Also, send a pick of your pulsesensor do hardware setup.

Virgilius1995, 2019-02-08:
Since none of the code I could find is working, I've setup a simple script to test the signal for consistent fluctuation with the sensor connected to my finger(also had some other people try):

const int inputPin = A1;
const int threshold = 12;

void setup() {

  Serial.begin(115200);
}

void loop() {

  int signal = (analogRead(inputPin));

  if (signal > threshold)
  {
    Serial.println(signal * 5);
  }

  delay(1);
  Serial.println(10);
}

I've set the script to only plot when the input is higher than 12(since below 12 I get allot of noise). According to the provided documentation the threshold should be between 1 and 1024 but anything above twelve and I get nothing and anything below is just noise.

[image]
[image]

Virgilius1995, 2019-02-08:
I should add that I doubt the authenticity of my sensor since the wires it came solder with are the wrong color
[image]
[image]

biomurph, 2019-02-11:
It does appear that you are using a counterfeit Pulse Sensor.
I'm sorry, but I can't troubleshoot your problem since I have no idea what the hardware is.

Please support original makers by purchasing legit parts from reliable suppliers
www.pulsesensor.com
www.adafruit.com
www.sparkfun.com
[digikey](https://www.digikey.com/products/en/maker-diy-educational/gadgets-gizmos/941?k=pulse%20sensor)
[mouser](https://www.mouser.com/Sensors/Sensor-Development-Tools/_/N-d0qxl?Keyword=pulsesensor&FS=True)
[Amazon](https://www.amazon.com/PulseSensor-com-Original-Pulse-Sensor-project/dp/B01CPP4QM0/ref=sr_1_1?keywords=pulsesensor&qid=1549914583&s=gateway&sr=8-1-spell)

Virgilius1995, 2019-02-11:
Thanks for the info 
I was told it was an original and I'll be in contact with the seller.

biomurph, 2019-02-11:
Oh, interesting!
Please let me know the name of the seller?

Virgilius1995, 2019-02-12:
They've came back to me saying it was possibly a third party release... And they've sent me a replacement as well as saying they're going to make a note on the product. Its a bit difficult for me to order from official sites as I live in South Africa. The site is DIYElectronics... But going back I cant find anywhere indicating it to be original, except for original images being used to advertised and the link to the official site.

biomurph, 2019-02-12:
SparkFun bundles products from their store an ships them to brick and mortar stores all over the world.
Check out their distributors page
https://www.sparkfun.com/distributors
There are some locations in South Africa that they supply. Might have Pulse Sensor and other cool stuff

closed 2019-02-12 by biomurph