WorldFamousElectronics/PulseSensorPlayground #79: How to extract BPM value out and do calculation with it using STM32 board?

Teeblackgold97, 2019-03-03:
Hi, I'm currently using Nucleo64-F401RE development board to interface with pulse sensor. I wanted to design a detection system which can determine if the BPM value is less than assigned threshold, an action will be taken. Since that Nucleo board doesn't support interrupt for pulse sensor, I can't use the getbeatsperminute( ) function. May I know how can I solve this issue? Or I'll just use an Arduino board? Thank you for your attention. Much help will be appreciated.

biomurph, 2019-03-04:
@Teeblackgold97 I am not sure what you mean...

If you start with the PulseSensor_BPM_Alternative.ino sketch. That will return the latest BPM value whenever you call getBeatsPerMinute() function.

LMK if you have issues.

Teeblackgold97, 2019-03-08:
Hi, sorry for the late reply.
May I know where to put getBeatsPerMinute() function in the PulseSensor_BPM_Alternative.ino sketch? If can, how can I declare an integer for it? I want to do a calculation for it, not just serial print the value. Basically, I want to use this method: int myBPM = pulseSensor.getBeatsPerMinute(); which is from Getting_BPM_to_monitor. Since Nucleo can't run Getting_BPM_to_monitor.ino sketch, is there any other way to do it? 
Thank you for your attention, much appreciated.

biomurph, 2019-03-08:
You pretty much have the right idea.
Declare a variable `int myBPM;` or something.
Maybe also create a flag to know if your threshold has be reached `boolean BPMflag;`
Then, in the sketch you will have to check the BPM produced by the library against some threshold (I think that's what you're doing?)
Inside the loop, there is some code that runs when PulseSensor Playground finds the beginning of a heartbeat. That's where you can do your comparison. 

`if (pulseSensor.sawStartOfBeat()) {
  //      pulseSensor.outputBeat();
        myBPM = pulseSensor.getBeatsPerMinute();
        if(myBPM < threshold){
            BPMflag = true;
        }
      }
`

That example is just for demonstration. I have no idea what you are actually trying to do. If you post a link to your code, then further troubleshooting could happen, but the thing you want to do is on the easy side of hard.

Teeblackgold97, 2019-03-16:
Hi, I finally bought a brand new Arduino Uno board to save some of the hassle I had encountered. 
I posted the code below to show you what am I trying to do:
`#include <SPI.h>
#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>
#include <stdlib.h>
#include <math.h> 

#define ADC_ref 5
#define zero_x 1.569    
#define zero_y 1.569
#define zero_z 1.569
#define sensitivity_x 0.3
#define sensitivity_y 0.3
#define sensitivity_z 0.3
#define USE_ARDUINO_INTERRUPTS true

#define SCREEN_WIDTH 128 // OLED display width, in pixels
#define SCREEN_HEIGHT 64 // OLED display height, in pixels

#define OLED_RESET     4 // Reset pin # (or -1 if sharing Arduino reset pin)
Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RESET);

unsigned int value_x;
unsigned int value_y;
unsigned int value_z;

float xv;
float yv;
float zv;
 
float angle_x;
float angle_y;
float angle_z;

int led = 7; //indicating bluetooth connection

float temp;
int hum;
String tempC;
int error;
int pulsePin = 0;                 // Pulse Sensor purple wire connected to analog pin 0
int blinkPin = 13;                // pin to blink led at each beat
int fadePin = 5;
int fadeRate = 0;
// Volatile Variables, used in the interrupt service routine!
volatile int BPM;                   // int that holds raw Analog in 0. updated every 2mS
volatile int Signal;                // holds the incoming raw data
volatile int IBI = 600;             // int that holds the time interval between beats! Must be seeded! 
volatile boolean Pulse = false;     // "True" when heartbeat is detected. "False" when not a "live beat". 
volatile boolean QS = false;        // becomes true when Arduino finds a beat.
volatile int rate[10];                    // array to hold last ten IBI values
volatile unsigned long sampleCounter = 0;          // used to determine pulse timing
volatile unsigned long lastBeatTime = 0;           // used to find IBI
volatile int P =512;                      // used to find peak in pulse wave, seeded
volatile int T = 512;                     // used to find trough in pulse wave, seeded
volatile int thresh = 525;                // used to find instant moment of heart beat, seeded
volatile int amp = 100;                   // used to hold amplitude of pulse waveform, seeded
volatile boolean firstBeat = true;        // used to seed rate array so we startup with reasonable BPM
volatile boolean secondBeat = false;      // used to seed rate array so we startup with reasonable BPM

void setup()
{
  pinMode( led ,OUTPUT);
  
  Serial.begin(115200); //or use default 115200.
  display.begin(SSD1306_SWITCHCAPVCC, 0x3C);
  display.display();
  delay(2000);
[... 101 lines cut ...]
      BPM = 60000/runningTotal;               // how many beats can fit into a minute? that's BPM!
      QS = true;                              // set Quantified Self flag 
      // QS FLAG IS NOT CLEARED INSIDE THIS ISR
    }                       
  }
  if (Signal < thresh && Pulse == true){   // when the values are going down, the beat is over
    digitalWrite(blinkPin,LOW);            // turn off pin 13 LED
    Pulse = false;                         // reset the Pulse flag so we can do it again
    amp = P - T;                           // get amplitude of the pulse wave
    thresh = amp/2 + T;                    // set thresh at 50% of the amplitude
    P = thresh;                            // reset these for next time
    T = thresh;
  }
  if (N > 2500){                           // if 2.5 seconds go by without a beat
    thresh = 512;                          // set thresh default
    P = 512;                               // set P default
    T = 512;                               // set T default
    lastBeatTime = sampleCounter;          // bring the lastBeatTime up to date        
    firstBeat = true;                      // set these to avoid noise
    secondBeat = false;                    // when we get the heartbeat back
  }
  sei();     
  // enable interrupts when youre done!
}// end isr`
This is just a simple code, I have yet to develop it further. Basically I am designing a fatigue detection system, as you can see I'm not using getBeatsPerMinute() function because somehow I can't get any BPM value out of it. May I know what are the requirements for me to be able to activate the getBeatsPerMinute() function? The codes above can show BPM value but is not accurate nor consistent.

biomurph, 2019-03-18:
@Teeblackgold97 
You seem to be using an older version of our code. Are you not able to use the PulseSensor Playground library with the OLED?
Does this code work?

closed 2019-06-13 by biomurph