WorldFamousElectronics/PulseSensorPlayground #81: pulse  sensor readings are above 200 bps

semagesss, 2019-03-06:
i tried on different people but bpm is always above 200 bps any ideas???
#getting bpm  to monitor

biomurph, 2019-03-06:
@semagesss 
Your issue post is too vague to even begin to start debugging.
What hardware are you targeting? What example sketch are you running?
Please take a pic of your Pulse Sensor setup so we can see if you have legit hardware.

preksha02, 2019-05-17:
https://photos.app.goo.gl/DppBTW5CYzm7Bkj87
this is the picture of the sensor which i am using. 

code-
#define USE_ARDUINO_INTERRUPTS true    // Set-up low-level interrupts for most acurate BPM math.
#include <PulseSensorPlayground.h>     // Includes the PulseSensorPlayground Library.   

//  Variables
const int PulseWire = 0;       // PulseSensor PURPLE WIRE connected to ANALOG PIN 0
const int LED13 = 13;          // The on-board Arduino LED, close to PIN 13.
int Threshold = 518;           // Determine which Signal to "count as a beat" and which to ignore.
                               // Use the "Gettting Started Project" to fine-tune Threshold Value beyond default setting.
                               // Otherwise leave the default "550" value. 
                               
PulseSensorPlayground pulseSensor;  // Creates an instance of the PulseSensorPlayground object called "pulseSensor"

void setup() {   

  Serial.begin(9600);          // For Serial Monitor

  // Configure the PulseSensor object, by assigning our variables to it. 
  pulseSensor.analogInput(PulseWire);   
  pulseSensor.blinkOnPulse(LED13);       //auto-magically blink Arduino's LED with heartbeat.
  pulseSensor.setThreshold(Threshold);   

  // Double-check the "pulseSensor" object was created and "began" seeing a signal. 
   if (pulseSensor.begin()) {
    Serial.println("We created a pulseSensor Object !");  //This prints one time at Arduino power-up,  or on Arduino reset.  
  }
}

void loop() {

 int myBPM = pulseSensor.getBeatsPerMinute();  // Calls function on our pulseSensor object that returns BPM as an "int".
                                               // "myBPM" hold this BPM value now. 

if (pulseSensor.sawStartOfBeat()) {            // Constantly test to see if "a beat happened". 
 //Serial.println("♥  A HeartBeat Happened ! "); // If test is "true", print a message "a heartbeat happened".
 Serial.print("BPM: ");                        // Print phrase "BPM: " 
 Serial.println(myBPM);                        // Print the value inside of myBPM. 
}

  delay(20);                    // considered best practice in a simple sketch.

}

in the starting i get the correct readings but later i get more than 150 
Can you guide me ?

biomurph, 2019-05-17:
@preksha02 
I can see by the photo that you sent that you are using a cloned Pulse Sensor.
Since I have no idea of what is on that part, I really can't help you debug your problem.

There are many imitations, but only one original.
Please support the original makers of the Open Source Pulse Sensor by purchasing legit hardware.
here are links to verified sellers.
www.pulsesensor.com
www.adafruit.com
www.sparkfun.com
www.amazon.com
www.mouser.com
www.digikey.com

closed 2019-05-17 by biomurph