WorldFamousElectronics/PulseSensorPlayground #82: How to get Pulse Sensor working properly

RhizomaticNomad, 2019-03-17:
I attached a pulse sensor to an Arduino Uno as suggested here. As code I used the GettingStartedProject as well as PulseSensor_BPM as well as this easy sketch:

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600); 
}

void loop() {
  // put your main code here, to run repeatedly:
  Serial.println(analogRead(1));
}

Now to the problem: Somehow it seems to work, but not in the intended way. When I move i.e. the finger tip above the sensor, I get different values in the serial plotter. But when I put my finger on the sensor as intended, I get more or less the same value all the time. If the + is connected to 3.3V it ranges between 334 and 337, if it's connected to 5 V the values are between 507 and 511. 

I thought maybe the sensor is defect, but another one behaves the same way. And what ist strange to me is that it gives me other values if you change the distance betweeen the sensor and a thing in front of it. 

Any ideas what could be the problem?

biomurph, 2019-03-18:
In that easy sketch, are you plugging the Pulse Sensor purple wire into analog pin 1?

Please send a picture of your pulse sensor setup.

To answer your question about the values changing when you move the sensor around, that is because Pulse Sensor uses a light sensor to work. It's changing with the ambient light.

biomurph, 2019-05-08:
@preksha02 
In order to help you troubleshoot, please send a link to your code. Also, please send a picture of your Pulse Sensor setup.

Pulse Sensor is sensitive to movement noise. Where are you placing the Pulse Sensro?

preksha02, 2019-05-09:
https://photos.app.goo.gl/GyoJbcmUYpgnRCPc7
this is a setup of pulse sensor..i cover the sensor with black tape so as to avoid light, still it shows too much error.

code-

#define USE_ARDUINO_INTERRUPTS true    // Set-up low-level interrupts for most acurate BPM math.
#include <PulseSensorPlayground.h>     // Includes the PulseSensorPlayground Library.   

//  Variables
const int PulseWire = A0;       // PulseSensor PURPLE WIRE connected to ANALOG PIN 0
const int LED13 = 13;          // The on-board Arduino LED, close to PIN 13.
int Threshold = 517 ;           // Determine which Signal to "count as a beat" and which to ignore.
                               // Use the "Gettting Started Project" to fine-tune Threshold Value beyond default setting.
                               // Otherwise leave the default "550" value. 
                               
PulseSensorPlayground pulseSensor;  // Creates an instance of the PulseSensorPlayground object called "pulseSensor"

int period = 1000;
unsigned long time_now = 0;
void setup() {   

  Serial.begin(9600);          // For Serial Monitor

  // Configure the PulseSensor object, by assigning our variables to it. 
  pulseSensor.analogInput(PulseWire);   
  pulseSensor.blinkOnPulse(LED13);       //auto-magically blink Arduino's LED with heartbeat.
  pulseSensor.setThreshold(Threshold);   

  // Double-check the "pulseSensor" object was created and "began" seeing a signal. 
   if (pulseSensor.begin()) {
    Serial.println("We created a pulseSensor Object !");  //This prints one time at Arduino power-up,  or on Arduino reset.  
  }
}

int i=0;
int a[700];
float beat=0.00;

void loop() {
time_now= millis();
 int myBPM = pulseSensor.getBeatsPerMinute();  // Calls function on our pulseSensor object that returns BPM as an "int".
                                               // "myBPM" hold this BPM value now. 

if (pulseSensor.sawStartOfBeat()) {            // Constantly test to see if "a beat happened". 
 Serial.println("A HeartBeat Happened ! "); // If test is "true", print a message "a heartbeat happened".
 Serial.print("BPM: ");                        // Print phrase "BPM: " 
 Serial.println(myBPM);   // Print the value inside of myBPM. 
 int Signal= analogRead(PulseWire);
 Serial.println(Signal);
 beat= (float)60.00/myBPM;
 Serial.print("beat");
 Serial.println(beat);
 a[i]=Signal;
 i++;
 if (pulseSensor.sawNewSample()==true)
  {
    Serial.println("yes");
  }
  else
  {
    Serial.println("no");
  }
 int lastestBeatSample= pulseSensor.getLatestSample();
 Serial.println(lastestBeatSample);
 //long LB = pulseSensor.getLastBeatTime();
}
if (i==20)
//while(millis()< time_now + period)
{
 
 Serial.println("hello");
  for(i=0; i<20; i++)
  {
    Serial.println(a[i]);
  }
 // exit(0);
 
}

  delay(20);                    // considered best practice in a simple sketch.

}

biomurph, 2019-05-09:
@preksha02 
As I mentioned in the other issue you brought up, your array of 700 ints might be causing instability.
What are you using that for?
Thanks for sending the pic of your setup, but I would like to see a picture of the actual Pulse Sensor that you are using. Looking at the wires you have in that picture makes me thing that you are using a knockoff version of the hardware. Please send a pic.

preksha02, 2019-05-10:
https://photos.app.goo.gl/2PCi71z1Qyh2fEhp8
the set up of pulse sensor

biomurph, 2019-05-10:
Thanks for sharing that photo.
Unfortunately, I can see that the Pulse Sensor that you have is not made by us. It is a clone. 
I really can't help you troubleshoot, as I have no idea what is in that hardware.
Please support the original makers and purchase verified Pulse Sensor product.
These are verified sellers:
www.pulsesensor.com
www.adafruit.com
www.sparkfun.com
www.amazon.com

closed 2019-05-10 by biomurph