WorldFamousElectronics/PulseSensorPlayground #86: Pulse sensor amped with Teensy 3.2

h20180179, 2019-04-01:
I'm trying to interface Pulse sensor bpm alternative code with teensy 3.2 but the code is giving bpm value of 0 and IBI=600. what might be the possible cause of this? Can you please suggest a way to interface it on Teensy 3.2?

biomurph, 2019-04-01:
I need more info from you to troubleshoot this.
Are you able to view the Pulse Sensor data in the Arduino Serial Plotter?
Can you send a pic of your hardware setup? Have you made any modifications to the BPM_Alternative code?

h20180179, 2019-04-01:
Yes i'm able to view the data on serial plotter 
The code i'm using i've just changed the value of analog pin as the input pin is connected on pin 6 of teensy

p.s. im experimenting with the threshold value as i was getting bpm of 0 with 550. now when i have decreased the value of threshold the sensor gives bpm of 230 which is not possible(catching on noise)

#define USE_ARDUINO_INTERRUPTS false
#include <PulseSensorPlayground.h>

/*
   The format of our output.

   Set this to PROCESSING_VISUALIZER if you're going to run
    the Processing Visualizer Sketch.
    See https://github.com/WorldFamousElectronics/PulseSensor_Amped_Processing_Visualizer

   Set this to SERIAL_PLOTTER if you're going to run
    the Arduino IDE's Serial Plotter.
*/
const int OUTPUT_TYPE = SERIAL_PLOTTER;

/*
   Pinout:
     PULSE_INPUT = Analog Input. Connected to the pulse sensor
      purple (signal) wire.
     PULSE_BLINK = digital Output. Connected to an LED (and 220 ohm resistor)
      that will flash on each detected pulse.
     PULSE_FADE = digital Output. PWM pin onnected to an LED (and resistor)
      that will smoothly fade with each pulse.
      NOTE: PULSE_FADE must be a pin that supports PWM.
       If USE_INTERRUPTS is true, Do not use pin 9 or 10 for PULSE_FADE,
       because those pins' PWM interferes with the sample timer.
*/
const int PULSE_INPUT = A6;
const int PULSE_BLINK = 13;    // Pin 13 is the on-board LED
const int PULSE_FADE = 5;
const int THRESHOLD = 335;   // Adjust this number to avoid noise when idle

/*
   samplesUntilReport = the number of samples remaining to read
   until we want to report a sample over the serial connection.

   We want to report a sample value over the serial port
   only once every 20 milliseconds (10 samples) to avoid
   doing Serial output faster than the Arduino can send.
*/
byte samplesUntilReport;
const byte SAMPLES_PER_SERIAL_SAMPLE = 10;

/*
   All the PulseSensor Playground functions.
*/
PulseSensorPlayground pulseSensor;

void setup() {
  /*
     Use 115200 baud because that's what the Processing Sketch expects to read,
     and because that speed provides about 11 bytes per millisecond.

     If we used a slower baud rate, we'd likely write bytes faster than
     they can be transmitted, which would mess up the timing
     of readSensor() calls, which would make the pulse measurement
     not work properly.
  */
  Serial.begin(115200);

  // Configure the PulseSensor manager.
  pulseSensor.analogInput(PULSE_INPUT);
  pulseSensor.blinkOnPulse(PULSE_BLINK);
  pulseSensor.fadeOnPulse(PULSE_FADE);

  pulseSensor.setSerial(Serial);
  pulseSensor.setOutputType(OUTPUT_TYPE);
  pulseSensor.setThreshold(THRESHOLD);

  // Skip the first SAMPLES_PER_SERIAL_SAMPLE in the loop().
  samplesUntilReport = SAMPLES_PER_SERIAL_SAMPLE;

  // Now that everything is ready, start reading the PulseSensor signal.
//  if (!pulseSensor.begin()) {
//    /*
//       PulseSensor initialization failed,
//       likely because our Arduino platform interrupts
//       aren't supported yet.
//
//       If your Sketch hangs here, try changing USE_PS_INTERRUPT to false.
//    */
////    for(;;) {
////      // Flash the led to show things didn't work.
////      digitalWrite(PULSE_BLINK, LOW);
////      delay(50);
////      digitalWrite(PULSE_BLINK, HIGH);
////      delay(50);
////    }
//  }
}

void loop() {

  /*
     See if a sample is ready from the PulseSensor.

     If USE_INTERRUPTS is true, the PulseSensor Playground
     will automatically read and process samples from
     the PulseSensor.

     If USE_INTERRUPTS is false, this call to sawNewSample()
     will, if enough time has passed, read and process a
     sample (analog voltage) from the PulseSensor.
  */
  if (pulseSensor.sawNewSample()) {
    /*
       Every so often, send the latest Sample.
       We don't print every sample, because our baud rate
       won't support that much I/O.
    */
    if (--samplesUntilReport == (byte) 0) {
      samplesUntilReport = SAMPLES_PER_SERIAL_SAMPLE;

      pulseSensor.outputSample();

      /*
         At about the beginning of every heartbeat,
         report the heart rate and inter-beat-interval.
      */
      if (pulseSensor.sawStartOfBeat()) {
        pulseSensor.outputBeat();
      }
    }

    /*******
      Here is a good place to add code that could take up
      to a millisecond or so to run.
    *******/
  }

  /******
     Don't add code here, because it could slow the sampling
     from the PulseSensor.
  ******/
}

h20180179, 2019-04-01:
in the hardware i have connected + to 3.3 of teensy and - to ground (i have other hardware also connected to it's hard to see) do i have to use some noise cancellation code with it or is it already running?
[image]

h20180179, 2019-04-01:
i'm not able to open the serial plotter for teensy 3.2 though.

biomurph, 2019-04-02:
@h20180179 
Changing the analog pin to the one that your sensor is plugged into is clutch ;)

I can see by the hardware, that you are using a Pulse Sensor that is a clone of the original.
Since I am not the one who made that Pulse Sensor, I cannot help you troubleshoot. Who knows what's on that board?

Please support the original makers of Pulse Sensor and purchase legit hardware.
www.pulsesensor.com
www.adafruit.com
www.sparkfun.com
www.mouser.com
www.digikey.com
These suppliers sell the Original and Best Pulse Sensor.

h20180179, 2019-04-02:
woops. my bad. even i think this sensor is faulty. i bought a new one from adafruit and will update you on the same. thanks 💯

h20180179, 2019-04-19:
Hey, The original PulseSensor works absolutely fine with the Teensy board!! Although a kind request would be to make it more wearable as it gives accurate values when held between the fingers but not on the wrist. Thank you!

biomurph, 2019-04-19:
@h20180179 
The issue that you raise is as old as the Pulse Sensor. 
Wrist is not the best part of the body to measure PPG. You *can* do it, but it is hard.
PPG measures change in tissue density as your blood sloshes around your body when your heart beats. Best places are where you have capillary tissue. Fingertips, earlobes, etc.
We have gotten pulse wave from places like the wrist, but it's dodgy.

closed 2019-04-19 by biomurph