WorldFamousElectronics/PulseSensorPlayground #87: Pulse sensor w/Teensy board

dsiah, 2019-04-05:
Hi, 

First time on the forum and I have very little Arduino experience, but after running into some technical limitations I picked up a teensy board to experiment with. I'm using the pulse-sensor peripheral and I've found some good programs that I'd like to run on my Teensy 3.5 board. I've confirmed that all my pieces work together by setting up my circuit on my breadboard but when I use the Teensy on the Arduino programs that I'd like to run I run into an exception like:

```
Arduino: 1.8.8 (Mac OS X), Board: "Teensy 3.5, Serial, 120 MHz, Faster, US English"

Interrupt:28:5: error: expected constructor, destructor, or type conversion before '(' token
 ISR(TIMER2_COMPA_vect){                         // triggered when Timer2 counts to 124
     ^
/Users/dsiah/Documents/Arduino/pamp/PulseSensorAmped_Arduino_1.5.0/Interrupt.ino: In function 'void interruptSetup()':
Interrupt:18:3: error: 'TCCR2A' was not declared in this scope
   TCCR2A = 0x02;     // DISABLE PWM ON DIGITAL PINS 3 AND 11, AND GO INTO CTC MODE
   ^
Interrupt:19:3: error: 'TCCR2B' was not declared in this scope
   TCCR2B = 0x06;     // DON'T FORCE COMPARE, 256 PRESCALER
   ^
Interrupt:20:3: error: 'OCR2A' was not declared in this scope
   OCR2A = 0X7C;      // SET THE TOP OF THE COUNT TO 124 FOR 500Hz SAMPLE RATE
   ^
Interrupt:21:3: error: 'TIMSK2' was not declared in this scope
   TIMSK2 = 0x02;     // ENABLE INTERRUPT ON MATCH BETWEEN TIMER2 AND OCR2A
   ^
/Users/dsiah/Documents/Arduino/pamp/PulseSensorAmped_Arduino_1.5.0/Interrupt.ino: At global scope:
Interrupt:28:4: error: expected constructor, destructor, or type conversion before '(' token
 ISR(TIMER2_COMPA_vect){                         // triggered when Timer2 counts to 124
    ^
exit status 1
expected constructor, destructor, or type conversion before '(' token
/Users/dsiah/Documents/Arduino/pamp/PulseSensorAmped_Arduino_1.5.0/PulseSensorAmped_Arduino_1.5.0.ino
```

After some research i tried some things, including adding the correct imports:
```
#include <avr/io.h>
#include <avr/interrupt.h>
```

And when I tried to see if it wasn't picking up the avr header files i tried a bogus import and it sniped that, so I know the compiler is at least finding the header files, but this doesn't explain why the exception remains. Any help is greatly appreciated!

biomurph, 2019-04-05:
@dsiah Well, the Teensy does have interrupts that _would_ work with Pulse Sensor, but they're not implemented yet. We plan to, and we're working toward this eventual hardware support.
For now, please use the PulseSensor_BPM_Alternative code example to start with. It uses a software timer interrupt, and it can be implemented with very little jitter.

dsiah, 2019-04-06:
Lovely, I was able to compile that, as a follow up:

1) What is the key difference between using the software timer interrupt and the hardware support?
2) Is there a timeline for the hardware support? 

Thank you!

biomurph, 2019-04-08:
@dsiah 
1. When using the hardware timer, you are sure to get a very regular sample rate, which is important for doing DSP on analog signals. That ensures that our derivation of BPM values is accurate. Using a software timer means that the main loop in the program has to check the time to see if it's time to read the sensor. This can cause jitter in the timing of the analog readings, which will effect accuracy of the BPM values. Both our hardware and software timers are set up to read at 500Hz (2mS sample rate). That, according to some science that I read, is good enough to get accurate heart rate data. If you're using the software timer, you have to be mindful in your code not to get too caught up in some activity for more than 2mS to ensure that you don't miss a sample reading. 

We do have some statistical analysis code in the Playground to get a baseline of your sample accuracy when running either timer. A tutorial on that is forthcoming.

2. Right now, I'm wrapping up a tutorial on how to connect Pulse Sensor to Raspberry Pi. When I'm done with that, I will jump on the Teensy family.

biomurph, 2019-06-13:
@dsiah 
I'm jumping into this and noticing some things about your original issue.
First, it looks like you're compiling an unsupported code base.
The PulseSensorAmped_Arduino_1.5.0.ino has been superseded by the PulseSensor Playground Library. Right now, the best example sketch to use is the PulseSensor_BPM_Alternative.ino sketch.
This will compile and run on Teensy boards, but we don't have the timer interrupt worked out yet. 
Please use the new library.

closed 2019-06-13 by biomurph

dsiah, 2019-07-13:
Thanks @biomurph I've been using this and hope the issue directs others in my situation to use that sketch.