WorldFamousElectronics/PulseSensorPlayground #88: PulseSensor 3.3V

FrancoisCosta, 2019-04-05:
Hi!
Great work with the PulseSensorPlayground library :) 
However, I'm having some issues implementing the method described below to use the sensor with 3.3V rather than 5V.

[image]

The page says that the PulseSensor_Amped_Arduino arduino code (where I found the interrupt tab) 'has been superseded' by this library: Link (https://www.eecs.yorku.ca/~jr/res/m/MD/PulseSensor.pdf)
However, I have been unable to find the code I need to change in the PulseSensorPlayground library. 

Any guidance would be greatly appreciated. 
Thanks!

FrancoisCosta, 2019-04-05:
I should mention I am using the Arduino UNO R3 (5V works good, 3.3V limits the output range from to 0-675.
I aim to use the 3.3V of the Feather (ATmega32u4, 8MHz)

biomurph, 2019-04-05:
@FrancoisCosta 
When you are using the UNO, why not simply use the 5V output?
If you must use the 3.3V, then you should simply also wire the AREF pin (just near the pin13) also to the 3.3V. Then, add the line `analogReverence(EXTERNAL);` to your setup()

When you run the Pulse Sensor on the Feather, you won't need to do a thing, since the Feather and the Pulse Sensor will be under the same power supply.

FrancoisCosta, 2019-04-06:
OK! thanks you!

When using the 3.3V of the Feather rather than the 5V of the arduino, the signal often gets 'clipped' rather than clear peaks. 

[image]

Is there something else I should be changing in the code?

FrancoisCosta, 2019-04-06:
Actually, I've started to get the same problem with the Arduino UNO (running the PulseSensor_BPM sketch)!
Any idea what is going wrong? Was working fine previously

[image]

biomurph, 2019-04-06:
nothing is wrong. 
The Pulse Sensor is essentially a light sensor. It measures the change in tissue density when the heartbeat pulse wave travels under it. Your pulse is rather 'bright', so when it happens the sensor gets saturated. This is not effecting the accuracy of the BPM value. 
Try the Pulse Sensor on different parts of your body and see how the intensity of the pulse varies.

Good reference for the analogReference() can be found here
https://www.arduino.cc/reference/en/language/functions/analog-io/analogreference/

closed 2019-04-06 by biomurph