WorldFamousElectronics/PulseSensorPlayground #92: getLastBeatTime()

preksha02, 2019-05-08:
Member function getLastBeatTime() is not working in arduino. While using this member function, arduino shows an error- 'class PulseSensorPlayground' has no member named 'getLastBeatTime'. kindly help!

biomurph, 2019-05-08:
@preksha02 
I need more information to troubleshoot your problem.
What is your target hardware?
Show me your code, so I can see how you are implementing the function.

preksha02, 2019-05-09:
#define USE_ARDUINO_INTERRUPTS true    // Set-up low-level interrupts for most acurate BPM math.
#include <PulseSensorPlayground.h>     // Includes the PulseSensorPlayground Library.   

//  Variables
const int PulseWire = A0;       // PulseSensor PURPLE WIRE connected to ANALOG PIN 0
const int LED13 = 13;          // The on-board Arduino LED, close to PIN 13.
int Threshold = 517 ;           // Determine which Signal to "count as a beat" and which to ignore.
                               // Use the "Gettting Started Project" to fine-tune Threshold Value beyond default setting.
                               // Otherwise leave the default "550" value. 
                               
PulseSensorPlayground pulseSensor;  // Creates an instance of the PulseSensorPlayground object called "pulseSensor"

int period = 1000;
unsigned long time_now = 0;
void setup() {   

  Serial.begin(9600);          // For Serial Monitor

  // Configure the PulseSensor object, by assigning our variables to it. 
  pulseSensor.analogInput(PulseWire);   
  pulseSensor.blinkOnPulse(LED13);       //auto-magically blink Arduino's LED with heartbeat.
  pulseSensor.setThreshold(Threshold);   

  // Double-check the "pulseSensor" object was created and "began" seeing a signal. 
   if (pulseSensor.begin()) {
    Serial.println("We created a pulseSensor Object !");  //This prints one time at Arduino power-up,  or on Arduino reset.  
  }
}

int i=0;
int a[700];
float beat=0.00;

void loop() {
time_now= millis();
 int myBPM = pulseSensor.getBeatsPerMinute();  // Calls function on our pulseSensor object that returns BPM as an "int".
                                               // "myBPM" hold this BPM value now. 

if (pulseSensor.sawStartOfBeat()) {            // Constantly test to see if "a beat happened". 
 Serial.println("A HeartBeat Happened ! "); // If test is "true", print a message "a heartbeat happened".
 Serial.print("BPM: ");                        // Print phrase "BPM: " 
 Serial.println(myBPM);   // Print the value inside of myBPM. 
 int Signal= analogRead(PulseWire);
 Serial.println(Signal);
 beat= (float)60.00/myBPM;
 Serial.print("beat");
 Serial.println(beat);
 a[i]=Signal;
 i++;
 if (pulseSensor.sawNewSample()==true)
  {
    Serial.println("yes");
  }
  else
  {
    Serial.println("no");
  }
 int lastestBeatSample= pulseSensor.getLatestSample();
 Serial.println(lastestBeatSample);
 //long LB = pulseSensor.getLastBeatTime();
}
if (i==20)

{
 
 Serial.println("hello");
  for(i=0; i<20; i++)
  {
    Serial.println(a[i]);
  }
 // exit(0);
 
}

  delay(20);                    // considered best practice in a simple sketch.

}

sir, i want to analyse the the single heart beat generated in serial plotter and want to study further about it thats why i want to use member function getLastBeatTime().
here in this code i use different different member functions , just to check how it is used.

biomurph, 2019-05-09:
@preksha02 
I still don't quite know what your issue is. 
The getLastBeatTime function works for me.
One thing that I see is that you are creating an array int a[700]
700! that's a very large array. You may be using up too much memory and getting instability that way.

preksha02, 2019-05-09:
my issue is when i apply pulseSensor.getLastBeatTime() , it shows an error- 'class PulseSensorPlayground' has no member named 'getLastBeatTime'.
i used such a big array to store the values of of analogRead to perform some operations on it.

biomurph, 2019-05-13:
I can compile your code against an Arduino UNO and I don't get any errors. Just a warning that memory is low due to the 700 member array.
Can you post the errors you are getting?

closed 2019-06-13 by biomurph