WorldFamousElectronics/PulseSensorPlayground #93: Attiny85 wrong intereupt timings?

MrMabulous, 2019-05-13:
https://github.com/WorldFamousElectronics/PulseSensorPlayground/blob/1e478b25c211b91ec6bb6f54b32be843d0086c6e/src/utility/Interrupts.h#L210

Shouldn’t that be the other way around? 0x89 for 16mhz and 0x88 for 8 mhz...

biomurph, 2019-05-13:
You may be correct.
This portion of the library likely has more cobwebs that other portions.
What ATtiny85 board are you targeting? I have a LilyTiny and a LilyTwinkle that I just got in house to verify these settings. Also have some blank ATtiny85 DIP chips to play with.
This issue will get resolved soon.

MrMabulous, 2019-05-13:
Attiny85 specificaion document (http://www.farnell.com/datasheets/1698186.pdf) page 89

1000 (0x88) yields 128 clock prescaler
1001  (0x89) yields 256 clock prescaler

At the target counter of 125, a prescaler of 256 is needed at 16 Mhz and one of 128 at 8Mhz

biomurph, 2019-05-17:
@MrMabulous 
Just updated the library here on the repo with example sketches for ATtiny85.
I used the SparkFun Tiny Programmer and a ATtiny85 DIP chip.
Works like a champ. Simply Marbulous ;)

I will roll out another version release soon. For now, you can download or clone this repo to use it.
Let me know how it goes. I am thinking of putting up a speaker example. Found a tone library that I want to modify to do Timerless software tone generation.

I also have a LilyPad Protosnap and also a Tiny Twinkle that I want to play with.

Tutorial also getting rolled out soon

closed 2019-05-17 by biomurph