WorldFamousElectronics/PulseSensorPlayground #99: nRF52832 support

baramuse, 2019-06-05:
Hi,

I'm trying to get the library work with my Adafruit Bluefruit Feather (nRF52832) but can't compile the sketches, I tried the PulseSensor_BPM_Alternative :

```
In file included from PulseSensor_Playground/src/PulseSensorPlayground.h:439:0,
                 from PulseSensor_Playground/src/PulseSensorPlayground.cpp:16:
PulseSensor_Playground/src/PulseSensorPlayground.cpp: In member function 'boolean PulseSensorPlayground::sawNewSample()':
PulseSensor_Playground/src/utility/Interrupts.h:71:45: error: 'cli' was not declared in this scope
 #define DISABLE_PULSE_SENSOR_INTERRUPTS cli()
                                             ^
PulseSensor_Playground/src/PulseSensorPlayground.cpp:105:5: note: in expansion of macro 'DISABLE_PULSE_SENSOR_INTERRUPTS'
     DISABLE_PULSE_SENSOR_INTERRUPTS;
     ^
PulseSensor_Playground/src/utility/Interrupts.h:72:44: error: 'sei' was not declared in this scope
 #define ENABLE_PULSE_SENSOR_INTERRUPTS sei()
                                            ^
PulseSensor_Playground/src/PulseSensorPlayground.cpp:108:5: note: in expansion of macro 'ENABLE_PULSE_SENSOR_INTERRUPTS'
     ENABLE_PULSE_SENSOR_INTERRUPTS;
     ^
exit status 1
Error compiling for board Adafruit Bluefruit nRF52832 Feather.
```

Any idea how can I get out of this ?

Cheers

biomurph, 2019-06-05:
@baramuse 
We are aware of this issue, and we are working on a solution for it.
Likely it will first be a fix for the BPM_Alternative sketch, with support for interrupts coming after that.

baramuse, 2019-06-05:
Thanks for your update, I'll keep on eye on this issue then, and will continue my work with other compatible boards (avr ones ?)

biomurph, 2019-06-13:
@baramuse 
I have updated the [PulseSensor Playground](https://github.com/WorldFamousElectronics/PulseSensorPlayground) library with support for the Adafruit Bluetooth Feather (nRF52832). I did a cursory check on 'generic' nRF52832 boards and they don't seem to throw the same error.
Please download the library from this repository. I have not created a new release yet. I will try to get some more changes made before I update the release that can be accessed through the Arduino Libraries Manager tool. If you have any issues, please re-open this thread.

closed 2019-06-13 by biomurph