WorldFamousElectronics/PulseSensorPlayground #100: Pulse Sensor priting high values

FernandoDiogo99, 2019-06-11:
I connected the pulse sensor in my arduino, and its sending high values( From 300 to 500 bpms). I think it might be ASCII, but i'm not sure. 

And I think the problem isn't the pulse sensor since I have already tested 2 different ones

`

//  Variables
int PulseSensorPurplePin = 1;       
int LED13 = 13;  

int Signal;                
int Threshold = 550;            

// The SetUp Function:
void setup() {
  pinMode(LED13,OUTPUT);         
   Serial.begin(9600);         

}

// The Main Loop Function
void loop() {

  Signal = analogRead(PulseSensorPurplePin);  
                                              

   Serial.println(Signal);                    

delay(100);

}
`

biomurph, 2019-06-11:
@FernandoDiogo99 
I'm not sure I understand your issue. 
The code that you posted reads the signal from the Pulse Sensor every 100ms (10Hz) and sends it over the serial port. That's all. It does not calculate BPM. It's best used with the Arduino serial plotter. Click on `Tools > Serial Plotter` to open it, and make sure that the baud rate is 9600. Baud rate selector is in the lower right hand corner of the plotter window.

biomurph, 2019-06-13:
@FernandoDiogo99 
Please let me know if my comment cleared up your issue.

Perprintim, 2019-06-23:
@biomurph Hi! I have the same problem. My pulse sensor sends me high values and beside that it prints me values even though I don't have the finger placed onto the sensor. Please I really need to know how to fix these two issues.
I am using this code:

//Variables
const int PulseWire = A0;       
const int LED13 = 13;          // The on-board Arduino LED, close to PIN 13.
int Threshold = 550;           //for heart rate sensor
float myTemp;
int myBPM;
String BPM;
String temp;
int error;
int panic;
int raw_myTemp;
float Voltage;
float tempC;
void setup()
{
 
  Serial.begin(9600); 
  esp8266.begin(115200);
  pulseSensor.analogInput(PulseWire);   
  pulseSensor.blinkOnPulse(LED13);       //auto-magically blink Arduino's LED with heartbeat.
  pulseSensor.setThreshold(Threshold);   

  // Double-check the "pulseSensor" object was created and "began" seeing a signal. 
   if (pulseSensor.begin()) {
    Serial.println("We created a pulseSensor Object !");  //This prints one time at Arduino power-up,  or on Arduino reset.  
  }
  Serial.println("AT");
  esp8266.println("AT");

  delay(3000);

  if(esp8266.find("OK"))
  {
    connectWiFi();
  }
  t.every(10000, getReadings);
   t.every(10000, updateInfo);
}

void loop()
{
  panic_button();
start: //label
    error=0;
   t.update();
    //Resend if transmission is not completed
    if (error==1)
    {
      goto start; //go to label "start"
    } 
 delay(4000);
}

void updateInfo()
{
  String cmd = "AT+CIPSTART=\"TCP\",\"";
  cmd += IP;
  cmd += "\",80";
  Serial.println(cmd);
  esp8266.println(cmd);
  delay(2000);
  if(esp8266.find("Error"))
  {
    return;
  }
  cmd = msg ;
  cmd += "&field1=";    //field 1 for BPM
  cmd += BPM;
  cmd += "&field2=";  //field 2 for temperature
  cmd += temp;
  cmd += "\r\n";
  Serial.print("AT+CIPSEND=");
  esp8266.print("AT+CIPSEND=");
  Serial.println(cmd.length());
  esp8266.println(cmd.length());
  if(esp8266.find(">"))
  {
    Serial.print(cmd);
    esp8266.print(cmd);
  }
  else
  {
    Serial.println("AT+CIPCLOSE");
    esp8266.println("AT+CIPCLOSE");
    //Resend...
    error=1;
  }
}

boolean connectWiFi()
{
  Serial.println("AT+CWMODE=1");
  esp8266.println("AT+CWMODE=1");
  delay(2000);
  String cmd="AT+CWJAP=\"";
  cmd+=SSID;
  cmd+="\",\"";
  cmd+=PASS;
  cmd+="\"";
  Serial.println(cmd);
  esp8266.println(cmd);
  delay(5000);
  if(esp8266.find("OK"))
  {
    return true;
  }
  else
  {
    return false;
  }
}

void getReadings(){
  raw_myTemp = analogRead(A1);
  Voltage = (raw_myTemp / 1023.0) * 5000; // 5000 to get millivots.
  tempC = Voltage * 0.1; 
  //myTemp = (tempC* 1.8) + 32; // conver to F
  Serial.println(tempC);
  int myBPM = pulseSensor.getBeatsPerMinute();  // Calls function on our pulseSensor object that returns BPM as an "int".
                                               // "myBPM" hold this BPM value now. 
if (pulseSensor.sawStartOfBeat()) {            // Constantly test to see if "a beat happened". 
Serial.println(myBPM);                        // Print the value inside of myBPM. 
}

  delay(20);            
    char buffer1[10];
     char buffer2[10];
    BPM = dtostrf(myBPM, 4, 1, buffer1);
    temp = dtostrf(myTemp, 4, 1, buffer2);  
  }

void panic_button(){
  panic = digitalRead(8);
    if(panic == HIGH){
    Serial.println(panic);
      String cmd = "AT+CIPSTART=\"TCP\",\"";
  cmd += IP;
  cmd += "\",80";
  Serial.println(cmd);
  esp8266.println(cmd);
  delay(2000);
  if(esp8266.find("Error"))
  {
    return;
  }
  cmd = msg ;
  cmd += "&field3=";    
  cmd += panic;
  cmd += "\r\n";
  Serial.print("AT+CIPSEND=");
  esp8266.print("AT+CIPSEND=");
  Serial.println(cmd.length());
  esp8266.println(cmd.length());
  if(esp8266.find(">"))
  {
    Serial.print(cmd);
    esp8266.print(cmd);
  }
  else
  {
    Serial.println("AT+CIPCLOSE");
    esp8266.println("AT+CIPCLOSE");
    //Resend...
    error=1;
  }
}
}

biomurph, 2019-06-24:
@Perprintim 
It looks like you're targeting an ESP8266. That means that the PulseSensor Playground library will be using a software interrupt. It's important that you do the same in your code. Take a look at the BlinkWithoutDelay sketch in the `File > Examples > Digital` drop down for reference.

Also, try adjusting the Threshold variable upwards to avoid erroneous noise when not touching the sensor. 

Also, send a pic of your Pulse Sensor and setup for verification.

closed 2020-02-20 by biomurph