WorldFamousElectronics/PulseSensorPlayground #101: BPM work good and TFT work good but when I combine them, not good!

noajm, 2019-06-12:
So all works good but as I said and now when I want to see BPM number on the TFT. My TFT act wired like it will print different color and the letter "B" from BPM will be missing and the 5s maybe later it will turn all white I don't know why but after long debugging I was able to find the line causing that!  which is this -- >`pulseSensor.analogInput(PulseWire); `
However, if there is If a take the wire off from A0 things good but will print _BPM: 0_

```
#define USE_ARDUINO_INTERRUPTS true    // Set-up low-level interrupts for most acurate BPM math.
#include <PulseSensorPlayground.h>     // Includes the PulseSensorPlayground Library.   
#include <Adafruit_GFX.h>    // Core graphics library
#include <Adafruit_ST7735.h> // Hardware-specific library for ST7735
#include <Adafruit_ST7789.h> // Hardware-specific library for ST7789
#include <SPI.h>
#include <SD.h>

  #define TFT_CS        10
  #define TFT_RST        9 // Or set to -1 and connect to Arduino RESET pin
  #define TFT_DC         8

// OPTION 1 (recommended) is to use the HARDWARE SPI pins, which are unique
// to each board and not reassignable. For Arduino Uno: MOSI = pin 11 and
// SCLK = pin 13. This is the fastest mode of operation and is required if
// using the breakout board's microSD card.

// For 1.44" and 1.8" TFT with ST7735 (including HalloWing) use:
//Adafruit_ST7735 tft = Adafruit_ST7735(TFT_CS, TFT_DC, TFT_RST);

// For 1.54" TFT with ST7789:
//Adafruit_ST7789 tft = Adafruit_ST7789(TFT_CS, TFT_DC, TFT_RST);

// OPTION 2 lets you interface the display using ANY TWO or THREE PINS,
// tradeoff being that performance is not as fast as hardware SPI above.
#define TFT_MOSI 11  // Data out
#define TFT_SCLK 13  // Clock out

Adafruit_ST7735 tft = Adafruit_ST7735(TFT_CS, TFT_DC, TFT_MOSI, TFT_SCLK, TFT_RST);
float p = 3.1415926;
int Variable1;

//  Variables
int PulseWire = 0;       // PulseSensor PURPLE WIRE connected to ANALOG PIN 0
const int LED13 = 13;          // The on-board Arduino LED, close to PIN 13.
int Threshold = 550;           // Determine which Signal to "count as a beat" and which to ignore.
PulseSensorPlayground pulseSensor;  // Creates an instance of the PulseSensorPlayground object called "pulseSensor"
int myBPM = 100;
static uint32_t tsLastReport = 0;  
     
void setup(void) {
  Serial.begin(9600);
 //PulseWire = 3;
  pulseSensor.analogInput(PulseWire);   
  pulseSensor.blinkOnPulse(LED13);       //auto-magically blink Arduino's LED with heartbeat.
  pulseSensor.setThreshold(Threshold);   
  pulseSensor.begin();

  Serial.print(F("Hello! ST77xx TFT Test"));

#ifdef ADAFRUIT_HALLOWING
  // HalloWing is a special case. It uses a ST7735R display just like the
  // breakout board, but the orientation and backlight control are different.
  tft.initR(INITR_HALLOWING);        // Initialize HalloWing-oriented screen
  pinMode(TFT_BACKLIGHT, OUTPUT);
  digitalWrite(TFT_BACKLIGHT, HIGH); // Backlight on
#else
  // Use this initializer if using a 1.8" TFT screen:
 // tft.initR(INITR_BLACKTAB);      // Init ST7735S chip, black tab

  // OR use this initializer (uncomment) if using a 1.44" TFT:
  tft.initR(INITR_144GREENTAB); // Init ST7735R chip, green tab

  // OR use this initializer (uncomment) if using a 0.96" 180x60 TFT:
  //tft.initR(INITR_MINI160x80);  // Init ST7735S mini display

  // OR use this initializer (uncomment) if using a 1.54" 240x240 TFT:
  //tft.init(240, 240);           // Init ST7789 240x240
#endif

  Serial.println(F("Initialized"));

  uint16_t time = millis();
  tft.fillScreen(ST77XX_BLACK);
  time = millis() - time;
  
  Serial.println(time, DEC);
  delay(500);

  Serial.println("done");
  delay(1000);

  Variable1 = 90;
  
 
}

void loop() {
  

   int myBPM = pulseSensor.getBeatsPerMinute();  // Calls function on our pulseSensor object that returns BPM as an "int".                                 
       
      if (pulseSensor.sawStartOfBeat()) {  
     
 
         Serial.println(myBPM);                        // Print the value inside of myBPM. 
    
          } 
          
         delay(20);         

  testdrawtext(" BPM: ", ST77XX_WHITE);
  delay(600);
  tft.setTextColor(ST77XX_RED);  
  tft.setTextSize(3); 
  tft.print(myBPM); 
   delay(600);
 tft.fillScreen(ST77XX_BLACK);
  delay(500);

}

void testdrawtext(char *text, uint16_t color) {
  

  tft.setCursor(0, 0);
  //  tft.setCursor(10, 40);
  tft.setTextSize(2);
 // tft.setTextColor(color);
  tft.setTextColor(ST7735_YELLOW);
  tft.setTextWrap(true);
  tft.print(text);
}
```

noajm, 2019-06-12:
I actually here is the problem `#define USE_ARDUINO_INTERRUPTS true ` when I set `false` all good but no number gets to print out.

biomurph, 2019-06-12:
@noajm 
I think I understand what you're trying to do, but the problem seems to be that you are using delays of some length in your TFT code. When you use the BPM_Alternative sketch, it relies on software timing and the delays you have are getting in the way of the PulseSensor Playground accurately reading the sensor. 
When you do use interrupts in the PulseSensor_BPM sketch, the library should be getting accurate samples even through the delays. However, you may be bumping up against a conflict between the TFT and the PulseSensor code.

What Arduino are you targeting?
What Actual Adafruit TFT are you using?

Try declaring myBPM as a global variable and moving the value assignment into the if statement, like this:
`                              
       
      if (pulseSensor.sawStartOfBeat()) {  
     
         myBPM = pulseSensor.getBeatsPerMinute();  // Calls function on our pulseSensor object that returns BPM as an "int".  
         Serial.println(myBPM);                        // Print the value inside of myBPM. 
    
          } `

Since you are printing to the TFT, you don't really need the Serial.print statement in there either?

Why are you using those long delays in the loop? I see a total of 1700mS delay. Can you shorten those? Or use a software timer instead? Check out the BlinkWithoutDelay.ino in the 
`Arduino > File > Examples > 02.Digital` menu to see an example of this.

noajm, 2019-06-12:
I have made it global and did not think it worked but  I will try again with the taking the delay off. I put them because I was not sure what is going on. my `1.44" TFT Display with 128x128 Color Pixels` was overwriting and acting wired so I thought maybe the tft need time to clear or print. 

So you think there is no problem if I was using ADC0 pin and PWM pin simultaneously? I just find  this block of code:
```
void setup() {}

void loop() {
  noInterrupts();
  // critical, time-sensitive code here
  interrupts();
  // other code here
}
```
Do I have critical code ? 

I will try it with the delay off as you suggested as soon as I go back home from work.

biomurph, 2019-06-12:
I don't see any issues with using ADC0 and PWM. We do that with the Fade LED in our example sketch.

noajm, 2019-06-13:
Alright, I apologize for taking longer to get back but I have been trying a lot of stuff and none worked. I'm posting codes I have tried and I have recorded a video for each code.

**First Part**

Here is code `TFT Part1` and this is the [video](https://imgur.com/a/NYBdNcE) for it - the First video in there.

```
#define USE_ARDUINO_INTERRUPTS true    // Set-up low-level interrupts for most acurate BPM math.
#include <PulseSensorPlayground.h>     // Includes the PulseSensorPlayground Library.   
#include <Adafruit_GFX.h>    // Core graphics library
#include <Adafruit_ST7735.h> // Hardware-specific library for ST7735
#include <Adafruit_ST7789.h> // Hardware-specific library for ST7789
#include <SPI.h>
#include <SD.h>

  #define TFT_CS        10
  #define TFT_RST        9 // Or set to -1 and connect to Arduino RESET pin
  #define TFT_DC         8
  
#define TFT_MOSI 11  // Data out
#define TFT_SCLK 13  // Clock out

Adafruit_ST7735 tft = Adafruit_ST7735(TFT_CS, TFT_DC, TFT_MOSI, TFT_SCLK, TFT_RST);
float p = 3.1415926;
int Variable1;
int ledState = HIGH;  
//  Variables
int PulseWire = 0;       // PulseSensor PURPLE WIRE connected to ANALOG PIN 0
int myBPM = 0;
const int LED13 = 13;          // The on-board Arduino LED, close to PIN 13.
int Threshold = 550;           // Determine which Signal to "count as a beat" and which to ignore.
// constants won't change:
const long interval = 1000;   

unsigned long previousMillis = 0;
static uint32_t tsLastReport = 0;  
     
void setup(void) {
  Serial.begin(9600); 

  Serial.print(F("Hello! ST77xx TFT Test"));

#ifdef ADAFRUIT_HALLOWING

  tft.initR(INITR_HALLOWING);        // Initialize HalloWing-oriented screen
  pinMode(TFT_BACKLIGHT, OUTPUT);
  digitalWrite(TFT_BACKLIGHT, HIGH); // Backlight on
#else

  tft.initR(INITR_144GREENTAB); // Init ST7735R chip, green tab
  
#endif

  Serial.println(F("Initialized"));

  uint16_t time = millis();
  tft.fillScreen(ST77XX_BLACK);
  time = millis() - time;
  
  Serial.println(time, DEC);
  delay(500);

  Serial.println("done");
  delay(1000);

 
}

void loop() {

  testdrawtext(" BPM: ", ST77XX_WHITE);
  tft.setTextColor(ST77XX_RED);  
  tft.setTextSize(2); 
 
  int sensorValue = analogRead(A0);
  
  tft.print(sensorValue); 
    delay(300);
  tft.fillScreen(ST77XX_BLACK);
 
 

 

}

void testdrawtext(char *text, uint16_t color) {
   tft.setCursor(0, 0);
 tft.fillScreen(ST77XX_BLACK);

  
  tft.setCursor(0, 0);
  //  tft.setCursor(10, 40);
  tft.setTextSize(2);
 // tft.setTextColor(color);
  tft.setTextColor(ST7735_YELLOW);
  tft.setTextWrap(true);
  tft.print(text);
   delay(200);
}
```

**Second Part**

Here is code `TFT Part2` and this is the [video](https://imgur.com/a/NYBdNcE) for it - The second video in there

```
#define USE_ARDUINO_INTERRUPTS true    // Set-up low-level interrupts for most acurate BPM math.
#include <PulseSensorPlayground.h>     // Includes the PulseSensorPlayground Library.   
#include <Adafruit_GFX.h>    // Core graphics library
#include <Adafruit_ST7735.h> // Hardware-specific library for ST7735
#include <Adafruit_ST7789.h> // Hardware-specific library for ST7789
#include <SPI.h>
#include <SD.h>

  #define TFT_CS        10
  #define TFT_RST        9 // Or set to -1 and connect to Arduino RESET pin
  #define TFT_DC         8
.
#define TFT_MOSI 11  // Data out
#define TFT_SCLK 13  // Clock out

Adafruit_ST7735 tft = Adafruit_ST7735(TFT_CS, TFT_DC, TFT_MOSI, TFT_SCLK, TFT_RST);
float p = 3.1415926;
int Variable1;
int ledState = HIGH;  
//  Variables
int PulseWire = 0;       // PulseSensor PURPLE WIRE connected to ANALOG PIN 0
const int LED13 = 13;          // The on-board Arduino LED, close to PIN 13.
int Threshold = 550;           // Determine which Signal to "count as a beat" and which to ignore.
int myBPM = 0;
PulseSensorPlayground pulseSensor;  // Creates an instance of the PulseSensorPlayground object called "pulseSensor"

static uint32_t tsLastReport = 0;  
     
void setup(void) {
  Serial.begin(9600);
 // .  PulseWire = 3;
  pulseSensor.analogInput(PulseWire);   
  pulseSensor.blinkOnPulse(LED13);       //auto-magically blink Arduino's LED with heartbeat.
  pulseSensor.setThreshold(Threshold);   
  pulseSensor.begin();

  Serial.print(F("Hello! ST77xx TFT Test"));

#ifdef ADAFRUIT_HALLOWING

  tft.initR(INITR_HALLOWING);        // Initialize HalloWing-oriented screen
  pinMode(TFT_BACKLIGHT, OUTPUT);
  digitalWrite(TFT_BACKLIGHT, HIGH); // Backlight on
#else
  // Use this initializer if using a 1.8" TFT screen:
 // tft.initR(INITR_BLACKTAB);      // Init ST7735S chip, black tab

  // OR use this initializer (uncomment) if using a 1.44" TFT:
  tft.initR(INITR_144GREENTAB); // Init ST7735R chip, green tab

  // OR use this initializer (uncomment) if using a 0.96" 180x60 TFT:
  //tft.initR(INITR_MINI160x80);  // Init ST7735S mini display

  // OR use this initializer (uncomment) if using a 1.54" 240x240 TFT:
  //tft.init(240, 240);           // Init ST7789 240x240
#endif

  Serial.println(F("Initialized"));

  uint16_t time = millis();
  tft.fillScreen(ST77XX_BLACK);
  time = millis() - time;
  
  Serial.println(time, DEC);
  delay(500);

  Serial.println("done");
  delay(1000);

  Variable1 = 90;
 
}

void loop() {

    
  testdrawtext(" BPM: ", ST77XX_WHITE);
  tft.setTextColor(ST77XX_RED);  
  tft.setTextSize(3); 
  
  myBPM = pulseSensor.getBeatsPerMinute();   
if (pulseSensor.sawStartOfBeat()) {                            
}

  tft.print(myBPM); 
      delay(300);
  tft.fillScreen(ST77XX_BLACK);
 
 

 

}

void testdrawtext(char *text, uint16_t color) {
   tft.setCursor(0, 0);
 tft.fillScreen(ST77XX_BLACK);

  
  tft.setCursor(0, 0);
  //  tft.setCursor(10, 40);
  tft.setTextSize(2);
 // tft.setTextColor(color);
  tft.setTextColor(ST7735_YELLOW);
  tft.setTextWrap(true);
  tft.print(text);
   delay(200);
}
```

biomurph, 2019-06-13:
@noajm 
It will take a bit to dig into your code.
You say you are working with an Arduino Micro, which has an ATmega32u4.
What exact TFT screen are you using? That will help to know if there are specific issues.

noajm, 2019-06-13:
This is exactly where I bought my TFT from `https://www.google.com/url?sa=t&source=web&rct=j&url=https://www.adafruit.com/product/2088&ved=2ahUKEwjhx7vAtufiAhUEXK0KHdG7BVEQFjAAegQIAhAB&usg=AOvVaw1hXxhfG8k_71Brpf5-9rzM`

noajm, 2019-06-14:
I made a Diagram circuit in case you have a doubt and uploaded to my google [drive](https://drive.google.com/drive/folders/19Xjb5hvgIGpxv_fP1SicVs21srQo8G5w?usp=sharing)

noajm, 2019-06-16:
@biomurph  
I'm sorry but I am wondering if you come up with any ideas to try, please

noajm, 2019-06-17:
Could be this the issue: 

> The relation between timers and PWM outputs is:
> Pins 5 and 6: controlled by timer0
> Pins 9 and 10: controlled by timer1
> Pins 11 and 3: controlled by timer2

Unfortunately, I don't know how to tell what `timer` is `Getting_BPM_to_Monitor` using!

noajm, 2019-06-17:
I found out that they using `Timer1` and `Timer2` and modified the TFT pins but that did not work the same issue presents.

closed 2020-02-20 by biomurph