WorldFamousElectronics/PulseSensorPlayground #104: Can't get the right values with the pulse sensor and the LM35 temperature sensor . Reading 2 anolog sesnors

Wolffyx, 2019-06-19:
I am trying to combine the two sensors (pulse sensor and LM35 temp),but there is a problem with them when is a noise at the pulse sensor .The values are fluctuates especially the LM35 it goes from 31 degrees (normal value ) to 25-50 and even the pulse one . 
Here is the [result](https://imgur.com/a/gCLESby) .
The code is this(I got the code form the  Getting_BPM_to_Monitor sketch ) :
```
#define USE_ARDUINO_INTERRUPTS true    // Set-up low-level interrupts for most acurate BPM math.
#include <PulseSensorPlayground.h>     // Includes the PulseSensorPlayground Library. 

const int sensor = A5; // Assigning analog pin A5 to variable ‘sensor’
float tempc; //variable to store temperature in degree Celsius
float vout; //temporary variable to hold sensor reading
//  Variables
const int PulseWire = 0;       // PulseSensor PURPLE WIRE connected to ANALOG PIN 0
const int LED13 = 13;          // The on-board Arduino LED, close to PIN 13.
int Threshold = 550;           // Determine which Signal to "count as a beat" and which to ignore.
// Use the "Gettting Started Project" to fine-tune Threshold Value beyond default setting.
// Otherwise leave the default "550" value.

PulseSensorPlayground pulseSensor;  // Creates an instance of the PulseSensorPlayground object called "pulseSensor"
void setup() {
  Serial.begin(9600);          // For Serial Monitor
  // Configure the PulseSensor object, by assigning our variables to it.
  pulseSensor.analogInput(PulseWire);
  pulseSensor.blinkOnPulse(LED13);       //auto-magically blink Arduino's LED with heartbeat.
  pulseSensor.setThreshold(Threshold);
  // Double-check the "pulseSensor" object was created and "began" seeing a signal.
  if (pulseSensor.begin()) {
    Serial.println("We created a pulseSensor Object !");  //This prints one time at Arduino power-up,  or on Arduino reset.
  }
}

void loop() {
  int temp = analogTemp();
  Serial.print(temp);
  Serial.println("Temp");
  
  if (pulseSensor.sawStartOfBeat()) {            // Constantly test to see if "a beat happened".
    // Serial.println("♥  A HeartBeat Happened ! "); // If test is "true", print a message "a heartbeat happened".
    int myBPM = pulseSensor.getBeatsPerMinute();  // Calls function on our pulseSensor object that returns BPM as an "int".
    // "myBPM" hold this BPM value now.
    Serial.print("BPM: ");                        // Print phrase "BPM: "
    Serial.print(myBPM);                        // Print the value inside of myBPM.
    Serial.println(" A HeartBeat Happened ! ");
  }
  delay(20);                    // considered best practice in a simple sketch.
}
float analogTemp() {
  vout = analogRead(sensor);
  vout = (vout * 500) / 1023;
  return vout; // Storing value in Degree Celsius

}
```

biomurph, 2019-06-19:
@Wolffyx 
This is quite a simple sketch, and I don't see how there could be a conflict between the temp sensor and the Pulse Sensor. Try increasing the baud rate? Set it at 115200 and see if that helps?
Also, you could try to use a different sketch, like the PulseSensor_BPM sketch. Modify it so that it only prints the Pulse Sensor BPM values and the Temp values, and then look at the Serial Plotter?
Also, you should try to have the sketch print the raw Pulse Sensor signal as well as the temp values to the serial plotter so you can see what is happening to the signals. That might provide a clue.

Also, please send along a picture of your Pulse Sensor so I can verify that it is legit.

Wolffyx, 2019-06-20:
Here is a pic with the [pulse sensor ](https://imgur.com/a/bHlNSAl).Tried the PulseSensor_BPM sketch same problem .

biomurph, 2019-06-20:
@Wolffyx 
Thanks for the picture. It explains alot. You are not using an Official Pulse Sensor. I cannot help you troubleshoot your problem, because I have no idea what is in that hardware.
Please support the original makers and purchase legit Pulse Sensor. You can find them at these internet locations.
www.pulsesensor.com
www.adafruit.com
www.sparkfun.com
www.amazon.com
www.mouser.com
www.digikey.com

closed 2019-06-20 by biomurph