WorldFamousElectronics/PulseSensorPlayground #107: Serial prints junk on ESP8266

katerss1, 2019-09-13:
Just using the simplest code

biomurph, 2019-09-13:
@katerss1 
you're not providing enough information for us to help troubleshoot your problem.
exactly what code are you using?
what is your physical setup?
send or attach a picture of your hardware for verification
what is this so-called 'junk' that you speak of?

katerss1, 2019-09-13:
Sorry for the brief explanation.

**exactly what code are you using?**
I used the Getting_BPM_to_Monitor example.

```
/*  Getting_BPM_to_Monitor prints the BPM to the Serial Monitor, using the least lines of code and PulseSensor Library.
 *  Tutorial Webpage: https://pulsesensor.com/pages/getting-advanced
 *
--------Use This Sketch To------------------------------------------
1) Displays user's live and changing BPM, Beats Per Minute, in Arduino's native Serial Monitor.
2) Print: "♥  A HeartBeat Happened !" when a beat is detected, live.
2) Learn about using a PulseSensor Library "Object".
4) Blinks LED on PIN 13 with user's Heartbeat.
--------------------------------------------------------------------*/

#define USE_ARDUINO_INTERRUPTS true    // Set-up low-level interrupts for most acurate BPM math.
#include <PulseSensorPlayground.h>     // Includes the PulseSensorPlayground Library.   

//  Variables
const int PulseWire = 0;       // PulseSensor PURPLE WIRE connected to ANALOG PIN 0
const int LED13 = 13;          // The on-board Arduino LED, close to PIN 13.
int Threshold = 550;           // Determine which Signal to "count as a beat" and which to ignore.
                               // Use the "Gettting Started Project" to fine-tune Threshold Value beyond default setting.
                               // Otherwise leave the default "550" value. 
                               
PulseSensorPlayground pulseSensor;  // Creates an instance of the PulseSensorPlayground object called "pulseSensor"

void setup() {   

  Serial.begin(9600);          // For Serial Monitor

  // Configure the PulseSensor object, by assigning our variables to it. 
  pulseSensor.analogInput(PulseWire);   
  pulseSensor.blinkOnPulse(LED13);       //auto-magically blink Arduino's LED with heartbeat.
  pulseSensor.setThreshold(Threshold);   

  // Double-check the "pulseSensor" object was created and "began" seeing a signal. 
   if (pulseSensor.begin()) {
    Serial.println("We created a pulseSensor Object !");  //This prints one time at Arduino power-up,  or on Arduino reset.  
  }
}

void loop() {

 int myBPM = pulseSensor.getBeatsPerMinute();  // Calls function on our pulseSensor object that returns BPM as an "int".
                                               // "myBPM" hold this BPM value now. 

if (pulseSensor.sawStartOfBeat()) {            // Constantly test to see if "a beat happened". 
 Serial.println("♥  A HeartBeat Happened ! "); // If test is "true", print a message "a heartbeat happened".
 Serial.print("BPM: ");                        // Print phrase "BPM: " 
 Serial.println(myBPM);                        // Print the value inside of myBPM. 
}

  delay(20);                    // considered best practice in a simple sketch.

}```

**what is your physical setup?**
I use the Wemos D1 MINI. 3 wires: 1 signal to A0, + to 5V, - to ground.

**send or attach a picture of your hardware for verification**
https://prnt.sc/p5vc15

**what is this so-called 'junk' that you speak of?**
`^EP⸮F⸮,MH⸮⸮G⸮h⸮` something like this inside my serial monitor even though i use the exact same baud rate: 9600

biomurph, 2019-09-16:
@katerss1 
Thanks for the clarification. There's a couple of problems with your setup.

First, based on the Wemos page I found [here](https://wiki.wemos.cc/products:d1:d1_mini) you are only allowed to put in 3.2V max into the analog input. So you should move the + connector to the 3.3V pin instead of the 5V pin. Not sure if you have damaged the input or not, but that is possible.

Second, you are using a knockoff Pulse Sensor. Please support the original makers of Pulse Sensor and purchase legit hardware. I cannot help you with any hardware related issues because I don't know what is on that clone.

Third, try simply sending some Serial print commands without the sensor plugged in to ensure that you can at least get that working.

closed 2020-02-20 by biomurph