WorldFamousElectronics/PulseSensorPlayground #111: Cannot work with ESP32 NODEMCU

johnny2231, 2019-11-19:
I was using the code "Getting_BPM_to_Monitor" to track my heart BPM using arduino UNO.
After that I plan to use ESP32 as it is small and can use it to send data to IOT platform. But I cant see any thing printed on the serial monitor like I used to have when I use UNO. 

`

#define USE_ARDUINO_INTERRUPTS false    // Set-up low-level interrupts for most acurate BPM math.
#include <PulseSensorPlayground.h>     // Includes the PulseSensorPlayground Library.   

//  Variables
const int PulseWire = 34;       // PulseSensor PURPLE WIRE connected to ANALOG PIN 0
const int LED2 = 2;          // The on-board Arduino LED, close to PIN 13.
int Threshold = 550;           // Determine which Signal to "count as a beat" and which to ignore.
                               // Use the "Gettting Started Project" to fine-tune Threshold Value beyond default setting.
                               // Otherwise leave the default "550" value. 
                               
PulseSensorPlayground pulseSensor;  // Creates an instance of the PulseSensorPlayground object called "pulseSensor"

void setup() {   

  Serial.begin(9600);          // For Serial Monitor

  // Configure the PulseSensor object, by assigning our variables to it. 
  pulseSensor.analogInput(PulseWire);   
  pulseSensor.blinkOnPulse(LED2);       //auto-magically blink Arduino's LED with heartbeat.
  pulseSensor.setThreshold(Threshold);   

  // Double-check the "pulseSensor" object was created and "began" seeing a signal. 
   if (pulseSensor.begin()) {
    Serial.println("We created a pulseSensor Object !");  //This prints one time at Arduino power-up,  or on Arduino reset.  
  }
}

void loop() {

 int myBPM = pulseSensor.getBeatsPerMinute();  // Calls function on our pulseSensor object that returns BPM as an "int".
                                               // "myBPM" hold this BPM value now. 

if (pulseSensor.sawStartOfBeat()) {            // Constantly test to see if "a beat happened". 
 Serial.println("♥  A HeartBeat Happened ! "); // If test is "true", print a message "a heartbeat happened".
 Serial.print("BPM: ");                        // Print phrase "BPM: " 
 Serial.println(myBPM);                        // Print the value inside of myBPM. 
}

  delay(20);                    // considered best practice in a simple sketch.

}

  `

biomurph, 2019-11-19:
@johnny2231 
There are many ESP 32 boards out there. In order to help troubleshoot your problem, please tell me exactly which board you are targeting? 
Also, can you get any simple serial code to work on the module you are using?

johnny2231, 2019-11-20:
@biomurph 
Hi, the esp32 board i am using is this 
[image]

biomurph, 2019-11-20:
@johnny2231 
Please tell me what board you are selecting from the arduino boards dropdown menu.
This sounds like an issue with the board itself. Please confirm that you can or cannot run a simple sketch that outputs to the serial port.

johnny2231, 2019-11-23:
@biomurph 
Hi there, sorry for the late reply.
I chose DOIT ESP32 DEVKIT V1.

I have tried to use analogRead(gpio) in new sketch to see if I can read any data from the pulse sensor or not. And I got the values. 
And I can see the ECG by using the "getting started project".

biomurph, 2019-11-24:
@johnny2231 
OK, so you have analog read happening on the pin 34?
And you see data on the serial port?
Does this mean your problem is solved?

johnny2231, 2019-11-25:
@biomurph 
Yea can see the data on serial monitor, but when i use the "getting bpm to monitor" code, because I want to know my heart bpm, the serial monitor is blank.

johnny2231, 2019-11-25:
Is it because of this code?
"#define USE_ARDUINO_INTERRUPTS false"

biomurph, 2019-11-25:
@johnny2231 
It is best to set USE_ARDUINO_INTERRUPTS to false because we don't support interrupts on the ESP units yet.

try putting this line in the code at the start of the loop 

pulseSensor.sawNewSample();

johnny2231, 2019-11-26:
@biomurph  Hi, I tried that and can see bpm on serial monitor, but it seems not accurate as it goes to 200something... and the speed seems slower...

biomurph, 2019-11-30:
@johnny2231 
Please send a photo of the Pulse Sensor that you are using so I can verify the hardware?

Did you ever get good heart rate from it?

rsorima, 2019-12-16:
hello, I am also using the same board with Johnny but my problem is that it cannot get into these statement.

`
pulseSensor.sawNewSample();
int myBPM = pulseSensor.getBeatsPerMinute();  
if (pulseSensor.sawStartOfBeat()) {            
 Serial.println("♥  A HeartBeat Happened ! "); 
 Serial.print("BPM: ");
 Serial.println(myBPM);
}
  delay(20);
`

I cannot read a BPM, Help please.

biomurph, 2019-12-17:
@rsorima 
Please use the PulseSensor_BPM_Alternative. We don't have support for hardware timer interrupts yet for the ESP32.

closed 2020-02-20 by biomurph

fitrahmaw, 2023-02-08:
> @biomurph Hi, I tried that and can see bpm on serial monitor, but it seems not accurate as it goes to 200something... and the speed seems slower...

Hello, can i ask about this?
how do you solve this problem?

biomurph, 2023-02-08:
@fitrahmaw this issue is closed.
please start a new issue with a clear description of what your question is.