WorldFamousElectronics/PulseSensorPlayground #112: Playground Sketch : Getting_BPM_to_Monitor : Not providing output on Adafruit feather HUZZAH

bkleynhans, 2019-12-06:
I'm building an alternate interface for a web-based game, whereby I'm planning to display the users heart rate on the screen.  I'm using the original PulseSensor (bought from pulsesensor.com) connected to the AdaFruit feather HUZZAH ESP8266.

NOTE: To ensure the sensor isn't damaged, I used it on a MEGA2560 and the sketch worked fine.

The PulseSensor provides a 1.6V output but the feather supports 1V max input on it's analog pin, so I'm using a voltage divider to achieve this.

Both of the projects referenced are part of the PulseSensor playground.

When running the "GettingStartedProject" (which uses analogRead) I can see an output in the serial monitor (varying between 600 and 1024) however when I run the "Getting_BPM_to_Monitor" sketch (which uses pulseSensor.analogInput) I don't receive any output.

The only configuration changes made to the sample code for both sketches are PulseWire and LED13

`const int PulseWire = A0;       // PulseSensor PURPLE WIRE connected to ANALOG PIN 0`
`const int LED13 = 0;          // The on-board AdaFruit feather HUZZAH ESP8266 LED`

based on the pin-outs for the feather.

[Adafruit Feather HUZZAH ESP8266](https://learn.adafruit.com/adafruit-feather-huzzah-esp8266)

biomurph, 2019-12-09:
@bkleynhans 
The problem you're having is that in the Getting_BPM_To_Monitor skets uses interrupts by default.
In this case, you want to use the PulseSensor_BPM_Alternative sketch. Start there, and modify to fit the pinouts of the HUZZAH you're using. I have done this successfully in the past, using an ESP8266. I also used a voltage divider off the Pulse Sensor purple wire with a 2.2K and 1K resistor, tapping the signal between the 2.2K and 1K.

bkleynhans, 2019-12-10:
Hi biomurph,

That was exactly it thank you.  I got it going with the alternative sketch as you suggested.

closed 2019-12-10 by bkleynhans