WorldFamousElectronics/PulseSensorPlayground #114: compilation error for digispark Attiny85

SaimShuja, 2019-12-23:
there is something wrong with the library.
compilation log below for reference 

libraries\PulseSensor_Playground\PulseSensorPlayground.cpp.o: In function `PulseSensorPlayground::PulseSensorPlayground(int)':

C:\Users\sohai\Documents\Arduino\libraries\PulseSensor_Playground\src/PulseSensorPlayground.cpp:28: undefined reference to `operator new[](unsigned int)'

collect2.exe: error: ld returned 1 exit status

Multiple libraries were found for "PulseSensorPlayground.h"
 Used: C:\Users\sohai\Documents\Arduino\libraries\PulseSensor_Playground
exit status 1
Error compiling for board Digispark (Default - 16.5mhz).

biomurph, 2019-12-25:
Try uninstalling the PulseSensor Playground library, or going to your Documents/Arduino/libraries folder and manually deleting it and then re installing with the Arduino Library manager.

SaimShuja, 2020-01-03:
after deleting the folder from "libraries" i tried reinstalling from library manager and also tried by pasting doenloaded folder from git, same issue

biomurph, 2020-01-04:
I know that the playground works for Attiny85. Perhaps try to select a different board version that uses the ATtiny85.

closed 2020-02-20 by biomurph