WorldFamousElectronics/PulseSensorPlayground #115: Working with Arduino Nano 33 IoT

cfawcett, 2020-01-23:
Ive been trying to get the sensor to work with the arduino nano 33 Iot. I understand the sensor is primarily designed for 5v, but working with the arduino uno, it works on both 5v and 3.3v absolutely fine. 

When i try and use it on the nano IoT, it doesn't trigger a reading with the default gettingBPMtoMonitor sketch? Within the gettingstarted project it seems to be working fine and the plotter is working fully but the sawStartOfBeat just doesn't seem to be triggering. Ive tried adjusting the threshold thinking it might be that but it hasn't changed anything.

Is there any compatibility issues with the Nano 3.3 IoT board? Or is it something stupid ive looked past?

biomurph, 2020-01-23:
@cfawcett 
I  have not worked with the Arduino Nano 33. I just loaded the board files, and tried to compile and got an error. Were you able to compile the gettingBPMtoMonitor scketch? The issue is known when targeting SAMD MCUs. A provision has to be made to the interrupt functions. I have made this change to the library that is in this repo. To use it, please download the zip or clone the current version. It has not been released yet, so you won't be able to get the changes if you use the arduino library manager. likely next week.

But that doesn't acutally solve your problem. One issue is that the gettingBPMtoMonitor does try to use the hardware interrupt function. In this case, you are better off using the PulseSensor_BPM_Alternative example sketch. Try that and let us know if it is working better.

Also, if you can share your code, we can help to point you in the right direction regarding timing and other functions that you might want to know.

cfawcett, 2020-01-24:
Everything compiled fine on the board, yeah. 

The BPM alternative sketch does work for displaying the BPM etc into the monitor.

This is the code im working with, although its in a very early stage of kinda testing it. Its for my final year uni project, where the device connects to WiFi and an MQTT client to send the BPM from the pulse sensor, along with the time and a primary key to an Amazon web services table. Its more to test the underlying methods of getting streaming data to a cloud platform, then on to a web application.

This is the loop before any adaptation to change it to the BPM alternative method. Basically the plan was to take the BPM on every beat and send it up to the cloud. Thinking about it now as im typing, it might be better to just take the BPM every second or so to limit the amount of messages being sent. 

Is there a suitable way using BPM Alternate to do this?

Thanks for the help

`void loop() {

 int myBPM = pulseSensor.getBeatsPerMinute();
 if (WiFi.status() != WL_CONNECTED) {
    connectWiFi();
  }

  if (!mqttClient.connected()) {
    // MQTT client is disconnected, connect
    connectMQTT();
  }

// poll for new MQTT messages and send keep alives
  mqttClient.poll();

    if (pulseSensor.sawStartOfBeat()) {

// --------
       Serial.println("HeartBeat!"); // If test is "true", print a message "a heartbeat happened".
       Serial.print("BPM: ");                        // Print phrase "BPM: " 
       Serial.println(myBPM);// Print the value inside of myBPM. 
       Serial.println("Publishing message");
// ------ For Debugging
       
       JSONVar BPMObject;
        BPMObject["bpm"] = myBPM;
        BPMObject["time"] = getTime();
        BPMObject["workoutInt"] = 0;

        mqttClient.beginMessage("my/test");
        mqttClient.print(BPMObject);
        mqttClient.endMessage();

       
    }

    
}`

biomurph, 2020-01-24:
OK, I can sorta see what you're doing. 
You are better off working from the BPM_Alternative function, as we don't yet support hardware timer interrupts on SAMD boards. 
Yes, you may be better off in not sending the pulse continuously, although you likely _could_ do it. The thing to watch out for is disturbing the timing of the samples when you do your mqtt and connection stuff. We sample the Pulse Sensor signal at 500Hz (every 2 milliseconds) and in BPM_Alternative, we are using a software timer to do this. That means that if you have some code that takes more than 2mS to execute, then you will be chomping on the Pulse Sensor signal timing and likely get a bad BPM calculation. The thing to do to work around this is to take some time to read the Pulse Sensor exclusively, so that the BPM settles. Also, note that the BPM value is calculated as an **average** of the current IBI and the past 9 IBIs. Total of 10 beats are used in a rolling average to determine BPM. When the beat finding code is first fired up, or when you use pause() and then resume(), the values that are used to calculate BPM are seeded with dummy values. In this case, it may be more valuable to do your own calculation of BPM by using the value of the IBI variable every time a beat is found. This is called 'instantaneous BPM', by the way. The IBI variable holds the time in milliseconds between each beat.

BTW, this _could_ be better handled with a hardware timer interrupt, but in my experience it is difficult to manage that when doing time sensitive activities such as wificonnect and data transfer stuff.  We added the pause() and resume() functions for this reason, so that you can stop the sampling when you want to do other stuff. Right now, pause() and resume() will only reset the pulse and bpm finding variables when you're _not_ using hardware interrupts. This is in the most recent library that you can download or clone.

Any questions about the above, please reach out.

cfawcett, 2020-02-04:
Just tried using the BPM Alternative in this scenario and the timing is definitley off, with the monitor showing a very high  BPM without even sending any MQTT messages and just checking that the connections to the wifi and the server are live.

How would I go about giving the sensor time to read exclusivley? Still somewhat new to the whole arduino coding things so managing things like timing is definitley not a skillset I have managed to pick up yet.

cfawcett, 2020-02-04:
Alternativley, how would I access the IBI variable directly? If i were to just run the sensor and every second or so use the IBI variable to find the BPM using a calculation and then send that to the server, would that be a viable alternative?

biomurph, 2020-02-04:
@cfawcett 
First, regarding IBI variable, check this resources [page](https://github.com/WorldFamousElectronics/PulseSensorPlayground/blob/master/resources/PulseSensor%20Playground%20Tools.md). There is a function called `getInterBeatIntervalMs()` that returns the latest IBI. You can use it after the call to `sawStartOfBeat()` returns true.

Second, regarding timing, things can get tricky when the micro is tied up with WiFi and MQTT, or any radio related stuff that takes priority. I think there is a way to take the pulse occasionally, and then report to MQTT and keep those activities siloed. Please share your code and I can point you in a direction?

cfawcett, 2020-02-06:
This is the code that loops. The other code is just simple setup code for the initial connections to the wifi network and the server for MQTT.  The idea is basically that its meant to ensure that the connections are set and then poll the sensor. then everytime a sample is recieved it would just send it.  The code is messy as its just prototype to see how it would work. 

`void loop() {
   
     if (WiFi.status() != WL_CONNECTED) {
    connectWiFi();
  }

  if (!mqttClient.connected()) {
    // MQTT client is disconnected, connect
    connectMQTT();
    
  }
  
// poll for new MQTT messages and send keep alives
  mqttClient.poll();
// put your main code here, to run repeatedly:

if (pulseSensor.sawNewSample()) {
    /*
       Every so often, send the latest Sample.
       We don't print every sample, because our baud rate
       won't support that much I/O.
    */
    if (--samplesUntilReport == (byte) 0) {
      samplesUntilReport = SAMPLES_PER_SERIAL_SAMPLE;

      pulseSensor.outputSample();

      /*
         At about the beginning of every heartbeat,
         report the heart rate and inter-beat-interval.
      */
      if (pulseSensor.sawStartOfBeat()) {
        pulseSensor.outputBeat();
      }
    }

    /*******
      Here is a good place to add code that could take up
      to a millisecond or so to run.
    *******/ 

  Serial.println(pulseSensor.getInterBeatIntervalMs());
  int myBPM = 0; // Will be the BPM once its figured out
  JSONVar BPMObject;
  BPMObject["bpm"] = myBPM;
  BPMObject["time"] = getTime();
  BPMObject["workoutInt"] = 0;
  mqttClient.beginMessage("my/test");
  mqttClient.print(BPMObject);
  mqttClient.endMessage();

      
    }
  }
`

The code there would obviously send a message every single time theres a beat, which was the idea originally. But even a message every few seconds with the current BPM would work well, maybe every 2 or 3 seconds ideally.  

Thank you for the help with this, its very much appreicated.

biomurph, 2020-02-06:
In my experience it's not necessary to be constantly checking the wifi status and polling MQTT.
That stuff is time consuming and causes problems when timing is critical.

In this case, I would set up to do two different things.
1 service the Pulse Sensor
2 connect to MQTT and send/receive data

This can be done with a software timer. For example.

```
unsigned long startPulseCapture;
int pulseCaptureDuration = 10000;	// 10 seconds of pulse capture

void loop(){
	startPulseCapture = millis();
        // this while loop will run for 10 seconds
	while(millis() - startPulseCapture < pulseCaptureDuration){
		capturePulse();
	}

	// insert wifi or MQTT or other code here
}

void capturePulse(){
	if (pulseSensor.sawNewSample()) {
    if (--samplesUntilReport == (byte) 0) {
      samplesUntilReport = SAMPLES_PER_SERIAL_SAMPLE;
      pulseSensor.outputSample();		// this ouptuts to the serial port, do you need that?
      if (pulseSensor.sawStartOfBeat()) {	// this is true when a beat happens
        pulseSensor.outputBeat();	// this outputs to the serial port, do you need that?
				myBPM = pulseSensor.getBeatsPerMinute();
				instantaneousBPM = 60000 / pulseSensor.getInterBeatIntervalMs(); // calculate instantaneous BPM
      }
    }
}
```

If you want to go with the Pulse Sensor Playground calculated BPM, then you don't need anything except the line `if(pulseSensor.sawNewSample(){}` as this function will sample the Pulse Sensor at 500Hz and process the data and update the variables (IBI, BPM, etc). If you want to derive your own BPM, then you will need the `pulseSensor.sawStartOfBeat()` conditional and then calculate your own IBI (run and average, or whatever).

Makes sense?

cfawcett, 2020-02-06:
This worked perfectly. Would there be any issue with only looping for 3 seconds rather than 10 assuming I calculated the IBI myself? 

Many thanks for all the help. Its very very appreciated.

biomurph, 2020-02-06:
Hey @cfawcett that's great!
Totally untested seat of the pants code works sometimes!

I guess the thing you will want to make sure of is that you do actually get legit heartbeats in the time you allow for measuring. When the Pulse Sensor is fired up from scratch, sometimes it will 'settle' a bit. 

Also, there is a pause() and a resume() functions that we're NOT using. you _could_ use them, but they reset all of the variables and you will for sure see some settling of the BPM variable (it's seeded with dummy values, and averaged every 10 beats). 

That makes me think that the BPM value that you're getting will be relying on 'old' averages. Not sure if that's a good thing... Might be best to do your own averaging of fresh IBI values so that you get more accurate? Also, you likely want to throw away the first IBI you get when you enter the timed loop. That one would be suspect, and likely way off.

biomurph, 2020-02-20:
@cfawcett Are you still having success? If so, we'll close this issue.

cfawcett, 2020-04-18:
Apologies for the very late reply. Don't use github as often as i should.

Have had some very great success with it. 

Can't thank you enough for the help, helped a great deal with my honours project and I very very much appreciate it.

biomurph, 2020-04-18:
Great!

closed 2020-04-18 by biomurph