WorldFamousElectronics/PulseSensorPlayground #118: Problem calling PULSE_SENSOR_TIMING_ANALYSIS

Jardu, 2020-02-20:
Dear PulseSensorTeam:

I would like to activate the PULSE_SENSOR_TIMING_ANALYSIS function in the PulseSensorPlayground library. I´m doing this, following carefully your instructions:

I have a teensy 3.6. in use, Arduino IDE 1.8.10 and using an external editor (TextWrangler) to edit the PulseSensorPlayground.h file.
I uncomment the line: "#define PULSE_SENSOR_TIMING_ANALYSIS true" and comment „//#define PULSE_SENSOR_TIMING_ANALYSIS false“ 
I save the file, the editor maintains open. 
Now I open the original „PulseSensor_BPM_Alternative“ sketch, compile and upload the file to my teensy, open the Serial Monitor,  but unfortunately I don´t get the values related to the analysis after 30 seconds. The teensy keeps writing Serial data: "pulseSensor.outputSample();“ 

Am I doing something fundamentally wrong? Thank you for a short advice.
Jardu

biomurph, 2020-02-20:
@Jardu This is a great question. 
@bneedhamia can you chime in on this one? I will have to dig into the library to follow the timing analysis thread, but since you wrote this portion you may have immediate insight?

Jardu, 2020-02-21:
Dear Pulse Sensor Team,

fortunately I could solve the issue and actually it was my fault, since I wasn´t working with the right file/in the right directory. I used the *.h and *cpp files, that where in my proyect folder and not in the original library folder. I´m really sorry :-((( Hopefully you didn´t invest to much time!!! I already want to thank you for the effort!!! Now I´m getting really good values: -1 0 2 :-)
All the best
Jardu

biomurph, 2020-02-21:
Oh that's fantastic! I'll close the issue.
Great luck with your projects!

closed 2020-02-21 by biomurph