WorldFamousElectronics/PulseSensorPlayground #119: Flat Top on Amplitude in Serial Plotter

Jardu, 2020-02-20:
Dear PulseSensor Team:

first of all thank you for developing the great pulse sensor kit, the coding and all the good documentation. I got several devices and I´m quite happy with them. I got my first pulse sensor in 2017, which is still in function and bought a number of 5 in summer 2019.

Regularly I use them on an Arduino Uno and on a Teensy 3.2/3.6. Recently I made the following observations:
The sensor I bought in 2017 is showing the expected curve in the serial plotter, both on arduino and teensy. But if I connected the newer ones from 2019 on those devices, I get curves with a flat top in the serial plotter. Since I would like to get the most accurate results, I was wondering what could be the reason for flat top in the serial plotter you see in the attached screenshots? This occurs on all the new ones. I still had two sensor in their original package and opened one yesterday and it shows the same curve. I just bought the original ones, with the longer cable and all the pieces you show in the documentation.

Thank you in advance for a feedback on the subject.
Jardu

P.D.: I´m working with the Arduino IDE 1.8.10, to show the curves I just used the „GettingStartedProject“, but all the same occurs using other examples.

ARDUINO/UNO/REV.3 on 5V:
2017
[image]
2019
[image]

TEENSY 3.6. on 3.3V:
2017
[image]
2019
[image]

biomurph, 2020-02-20:
@Jardu 
Thanks for being a long-term customer ;)

The 'flat top' effect that you're seeing is called clipping. The Pulse Sensor signal is saturating the ADC input. The reason why it's not getting up to 1023 is because there is a protection diode in the circuit power input, so the max voltage that the sensor puts out is just a bit under 5V.

The reason why you are seeing a difference between the Pulse Sensors is likely due to a difference in the brightness of the LED we use. When Kingbright, the LED company, produces the LED the manufacturing process results in a range of possible brightnesses. We work with our manufacturer to accommodate that range, and it turns out some Pulse Sensors are more 'responsive' than others.

On a side note, I notice that there is some high frequency noise in a couple of your data traces. This is likely due to powerline noise. Sometimes if you have your laptop connected to a power supply, you can see this ripple. It's a known issue that we're going to address in the next hardware revision. To clear that up, either use the Arduino 3.3V supply (adding `analogReference(EXTERNAL);` to your setup and connectin 3.3V to the `AREF` pin) or power from a battery.

Jardu, 2020-02-21:
Dear Biomurph, 

thank you so much for your comments. Concerning the variation in brightness of the LED, do you think this could be solved by just simply changing the R1/470 OHM resistor by myself? Maybe using 1K or even a higher value? 

I will attempt to get a smoother curve without ripples using 3.3V/analogReference on the arduino. Do you think using "aRef" on the teensy 3.6 will be usefull? Since the native operating voltage is 3.3?

Thank you in advance!!!
Jardu.

biomurph, 2020-02-21:
@Jardu Our manufacturing parter already does some compensation for the different brightnesses. Each reel of LEDs has a BIN code, and that code has information about the brightness. Depending on the BIN code, they populate a different value resistor that we have specified. Unfortunately, we don't know what the BIN code is for different production runs. That's just TMI. It is likely NOT 470 ohm.

To answer your question, yes, you could rework the R1 resistor and put a different value there, but first be sure to measure the existing one to know where to shoot for.

To answer your original question, I don't know if you need to worry about the clipping effecting the results. The Pulse Sensor beat finder algorithm determines that a pulse has happened when the rising edge of the pulse signal crosses a threshold. The threshold is dynamically updated to be positioned in the midpoint between the trough and the peak of the signal.

It sounds like you are thinking that we are locating the 'beat' at the peak of the waveform, but that's not how we do it. After researching PPG, it turns out that the 'moment of heart beat' is more closely associated with the steepest part of the rising edge of the pulse wave. Since that part of the wave is so steep, I don't think that it will make a difference in the IBI or BPM values even if there is clipping. The clipping usually remains consistent, or drifts slowly in and out, so your relative IBI values are going to be consistent too.

Does that help with your question?

Jardu, 2020-02-24:
Dear Biomurph,

thank you for the detailed information on the subject. I´ll make some tests with different resistor to find a value that would fit. 
Modifing the sketch I realized, as you mentioned, that the beat ist not determinded by the peak but by the rising edge of the pulse crossing the threshold. I asked because I´m specifically using the peak value for a proyect. 
Thank you so much!
Best regards.
Jardu

biomurph, 2020-02-24:
Good luck!

closed 2020-02-24 by biomurph