WorldFamousElectronics/PulseSensorPlayground #121: Error compiling for board Arduino Nano 33 BLE

Sofyanjm, 2020-03-14:
I am trying to use pulse sensor with Arduino Nano 33 BLE sense, but when I verify it, it shows Error compiling for board Arduino Nano 33 BLE. The code is from the pulse sensor website. The code works with all Arduino board, but it is not working with the Arduino Nano 33 BLE. Code and Error text are down.

**This is the code:**

`/* Getting_BPM_to_Monitor prints the BPM to the Serial Monitor, using the least lines of code and PulseSensor Library.

Tutorial Webpage: https://pulsesensor.com/pages/getting-advanced
--------Use This Sketch To------------------------------------------

Displays user's live and changing BPM, Beats Per Minute, in Arduino's native Serial Monitor.
Print: "♥ A HeartBeat Happened !" when a beat is detected, live.
Learn about using a PulseSensor Library "Object".
Blinks LED on PIN 13 with user's Heartbeat.
--------------------------------------------------------------------*/
#define USE_ARDUINO_INTERRUPTS true // Set-up low-level interrupts for most acurate BPM math.
#include <PulseSensorPlayground.h> // Includes the PulseSensorPlayground Library.

// Variables
const int PulseWire = 0; // PulseSensor PURPLE WIRE connected to ANALOG PIN 0
const int LED13 = 13; // The on-board Arduino LED, close to PIN 13.
int Threshold = 550; // Determine which Signal to "count as a beat" and which to ignore.
// Use the "Gettting Started Project" to fine-tune Threshold Value beyond default setting.
// Otherwise leave the default "550" value.

PulseSensorPlayground pulseSensor; // Creates an instance of the PulseSensorPlayground object called "pulseSensor"

void setup() {

Serial.begin(9600); // For Serial Monitor

// Configure the PulseSensor object, by assigning our variables to it.
pulseSensor.analogInput(PulseWire);
pulseSensor.blinkOnPulse(LED13); //auto-magically blink Arduino's LED with heartbeat.
pulseSensor.setThreshold(Threshold);

// Double-check the "pulseSensor" object was created and "began" seeing a signal.
if (pulseSensor.begin()) {
Serial.println("We created a pulseSensor Object !"); //This prints one time at Arduino power-up, or on Arduino reset.
}
}

void loop() {

int myBPM = pulseSensor.getBeatsPerMinute(); // Calls function on our pulseSensor object that returns BPM as an "int".
// "myBPM" hold this BPM value now.

if (pulseSensor.sawStartOfBeat()) { // Constantly test to see if "a beat happened".
Serial.println("♥ A HeartBeat Happened ! "); // If test is "true", print a message "a heartbeat happened".
Serial.print("BPM: "); // Print phrase "BPM: "
Serial.println(myBPM); // Print the value inside of myBPM.
}

delay(20); // considered best practice in a simple sketch.

}
`

**Error text :**

`Arduino: 1.8.7 (Mac OS X), Board: "Arduino Nano 33 BLE"

In file included from /Users/sofyanjamal/Documents/Arduino/libraries/PulseSensor_Playground/src/PulseSensorPlayground.h:439:0,
from /Users/sofyanjamal/Documents/Arduino/libraries/PulseSensor_Playground/src/PulseSensorPlayground.cpp:16:
/Users/sofyanjamal/Documents/Arduino/libraries/PulseSensor_Playground/src/PulseSensorPlayground.cpp: In member function 'boolean PulseSensorPlayground::sawNewSample()':
/Users/sofyanjamal/Documents/Arduino/libraries/PulseSensor_Playground/src/utility/Interrupts.h:72:41: error: 'cli' was not declared in this scope
#define DISABLE_PULSE_SENSOR_INTERRUPTS cli()
^
/Users/sofyanjamal/Documents/Arduino/libraries/PulseSensor_Playground/src/PulseSensorPlayground.cpp:105:5: note: in expansion of macro 'DISABLE_PULSE_SENSOR_INTERRUPTS'
DISABLE_PULSE_SENSOR_INTERRUPTS;
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/sofyanjamal/Documents/Arduino/libraries/PulseSensor_Playground/src/utility/Interrupts.h:73:40: error: 'sei' was not declared in this scope
#define ENABLE_PULSE_SENSOR_INTERRUPTS sei()
^
/Users/sofyanjamal/Documents/Arduino/libraries/PulseSensor_Playground/src/PulseSensorPlayground.cpp:108:5: note: in expansion of macro 'ENABLE_PULSE_SENSOR_INTERRUPTS'
ENABLE_PULSE_SENSOR_INTERRUPTS;
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
exit status 1
Error compiling for board Arduino Nano 33 BLE.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

biomurph, 2020-03-14:
@Sofyanjm 
The latest Pulse Sensor Library does support the Nano 33 BLE, but it has not been officially released yet. Please either download or clone this repository and move it to your Arduino>Libraries folder. 
We do NOT support interrupts on the Nano 33 BLE yet, so you should use the BPM_Alternative example sketch. 
Will be updating the release next week, but for now the easiest thing to do us download or clone.

Sofyanjm, 2020-04-02:
@biomurph 
Thank you. It is working for the serial plotter, but I get very high values in the serial monitor. Is there an explanation to get the correct values of BPM?

biomurph, 2023-10-26:
PulseSensor Playground Library supports nRF52 based boards.

closed 2023-10-26 by biomurph