WorldFamousElectronics/PulseSensorPlayground #126: Ticking sound on speaker with Arduino Pro Micro

ralphcrutzen, 2020-06-07:
I'm making a heart rate monitor which produces a sound when the heart rate is above a certain value. I connected an 8 Ohm speaker to pin 2 and an LED to pin 5 of an Arduino Pro Micro (5V).

The problem is that the speaker is only producing short ticking sounds instead of beeps. The ticks aren't by far loud enough.

Things I have tried to rule out a malfunctioning speaker:
- Use another speaker -> results in the same problem
- Try the toneMelody sketch in the standard Arduino IDE examples -> This sketch works

I also tried an Arduino Uno with the same sketch and the same circuit. In this case most of the "beeps" are still "ticks", but sometimes, when the duration of the pulse's peak of the heartbeat is longer, the right loud beeping sound is produced.

I'm not sure if I'm correct, but it seems that on the Pro Micro the peak of the pulse drops too soon, so the noTone() function breaks off the sound too early. So I changed the Treshold value to some values below 550; that also didn't solve it.

So I removed the noTone() function and used tone(PIN_SPEAKER, 1047, 16), but I think this causes timing issues because there are some kind of irregular delays between heartbeats.

Does this have something to do with the difference in timing between the Pro Micro and the Uno? I would really like to use the Pro Micro because of the small form factor.

The code: (I combined the PulseSensor_BPM and PulseSensor_Speaker examples)

```
#define USE_ARDUINO_INTERRUPTS true    // Set-up low-level interrupts for most acurate BPM math.
#include <PulseSensorPlayground.h>     // Includes the PulseSensorPlayground Library.   

//  Variables
const int PulseWire = 0;       // PulseSensor PURPLE WIRE connected to ANALOG PIN 0
const int PIN_LED = 5;
const int PIN_SPEAKER = 2;
int Threshold = 550;           // Determine which Signal to "count as a beat" and which to ignore.
// Use the "Gettting Started Project" to fine-tune Threshold Value beyond default setting.
// Otherwise leave the default "550" value.

PulseSensorPlayground pulseSensor;  // Creates an instance of the PulseSensorPlayground object called "pulseSensor"

void setup() {

  Serial.begin(9600);          // For Serial Monitor

  // Configure the PulseSensor object, by assigning our variables to it.
  pulseSensor.analogInput(PulseWire);
  pulseSensor.fadeOnPulse(PIN_LED);
  pulseSensor.setThreshold(Threshold);

  // Double-check the "pulseSensor" object was created and "began" seeing a signal.
  if (pulseSensor.begin()) {
    Serial.println("We created a pulseSensor Object !");  //This prints one time at Arduino power-up,  or on Arduino reset.
  }
}

void loop() {

  int myBPM = pulseSensor.getBeatsPerMinute();  // Calls function on our pulseSensor object that returns BPM as an "int".
  // "myBPM" hold this BPM value now.

  if (pulseSensor.sawStartOfBeat()) {            // Constantly test to see if "a beat happened".
    Serial.print("♥ "); // If test is "true", print a message "a heartbeat happened".
    Serial.println(myBPM);                        // Print the value inside of myBPM.
    if (myBPM > 50) {
      tone(PIN_SPEAKER, 1047);
    }
  }

  if (pulseSensor.isInsideBeat() == false) {
    noTone(PIN_SPEAKER);
  }

  delay(20);                    // considered best practice in a simple sketch.
}

```

biomurph, 2020-06-07:
@ralphcrutzen 
Please share a photo or diagram of your circuit. It may be that you are driving the speaker incorrectly? Do you have a resistor and a capacitor in series with the speaker? The resistor will adjust the volume of the sound, and the capacitor will "AC couple" the audio signal.

I'm out of the lab today, but I can try to replicate your problem tomorrow. Until then, try modifying the circuit if you don't have those components installed.

ralphcrutzen, 2020-06-08:
This is my circuit. It's the same as on your website (https://pulsesensor.com/pages/pulse-sensor-speaker-tutorial)

[image]

I also made a short video. After 5 seconds you see me changing the orange wire to bypass the resistor, because it prevents the sound to be loud enough to be noticed.

[https://youtu.be/UmgQhFkoVho](https://youtu.be/UmgQhFkoVho)

As I mentioned before. The tone() function works in other sketches, like the toneMelody example. So that's why I don't suspect the circuit to be wrong.

ralphcrutzen, 2020-06-08:
I just tried it again with an Arduino Uno again. This now suddendly works like it should!
So apparently it had something to do with the difference between an Uno and a Pro Micro.

biomurph, 2020-06-08:
@ralphcrutzen 
I have a Leonardo clone here (which should the same micro as the ProMicro 5V, an ATmega32u4). and I've programmed it with our example code that beeps the speaker. I have verified that pin 2 is outputting a square wave with my o-scope. This is as expected. Who makes your ProMicro? Can you link to that?

Can you try a different pin to do the tone on? like pin 4, just for sanity's sake?

ralphcrutzen, 2020-06-09:
Changing pins from 2 to 4 didn't solve it, unfortunately.

I bought the Pro Micro here: https://www.tinytronics.nl/shop/nl/arduino/main-boards/pro-micro-5v-16mhz. They guarantee that it's an exact copy of the original Arduino, but I'll try to get my hands on an original too, just to check.

It seems to me that the duration of the heart pulse isn't long enough, so
```
  if (pulseSensor.isInsideBeat() == false) {
    noTone(PIN_SPEAKER);
  }
````
is executed too soon.

A way to solve my problem would be to be able to use the optional third argument (duration) in the tone() function, so that every beep lasts, say, exactly 250 milliseconds. Apparently, this doesn't work, but I'm not sure why. Is this because you use threading in the library and a Pro Micro handles this differently than an Uno?

biomurph, 2020-06-09:
That's interesting that it won't work with the duration argument. 

There is no 'threading' in Arduino. These micros are step-instruction based. We use a hardware interrupt, or a software interrupt, to get the most accurate pulse rate. If you don't have good sample rate integrity, your data is garbage.
I thought initially that there might be an issue with the hardware timer, but according to the files in Arduino software they use a different timer for tone than we do in PulseSensor Playground.
a quick way to check this would be to build off of the BPM_Alternative sketch instead of the BPM sketch. the Alternative sketch does not use hardware timer and instead uses software interrupts.

Just to be clear, it sounds like you used the same hardware setup (with PulseSensor and LED and Speaker) and ran some basic tone sketch. Is that right?

Do you have a Leonardo handy to test on? The only thing different from the Leonardo and the Pro Micro is a few pin definitions, it seems. That and the 5V-3V option. Looking at your product link it does seem to be a clone of the SparkFun version.

Another quick test to make is to change the hardware timer that is used by the PulseSensor Playground. This is easy, since we make a change whenever the Servo library is instantiated. You don't need to use the Servo library, just place
`#include <Servo.h>`
at the top of the sketch, before anything else. When PulseSensor Playground is included, it looks to see if the Servo library is also included, then it picks a different hardware timer.

ralphcrutzen, 2020-06-11:
Unfortunately I don't have a Leonardo.

I tried the BPM_Alternative sketch and added the tone() function on line 163:
```
if (pulseSensor.sawStartOfBeat()) {
   pulseSensor.outputBeat();
   tone(4, 1047, 200);
}
```
The result is the same problem: a "ticking" sound, like the sound is cut off to soon (after a few ms or so).

Finally, when I include the Servo library, it starts to get interesting (I hope...)
I get the following compiler errors:

```
Arduino:1.8.12 (Windows Store 1.8.33.0) (Windows 10), Board:"Arduino Leonardo"

In file included from D:\Documenten\Arduino\libraries\PulseSensor_Playground\src/PulseSensorPlayground.h:439:0,

                 from D:\Documenten\Arduino\PulseSensor\PulseSensor.ino:13:

D:\Documenten\Arduino\libraries\PulseSensor_Playground\src/utility/Interrupts.h: In function 'boolean PulseSensorPlaygroundSetupInterrupt()':

D:\Documenten\Arduino\libraries\PulseSensor_Playground\src/utility/Interrupts.h:144:7: error: 'TCCR2A' was not declared in this scope

       TCCR2A = 0x02;          // Disable PWM and go into CTC mode

       ^~~~~~

D:\Documenten\Arduino\libraries\PulseSensor_Playground\src/utility/Interrupts.h:144:7: note: suggested alternative: 'TCCR0A'

       TCCR2A = 0x02;          // Disable PWM and go into CTC mode

       ^~~~~~

       TCCR0A

D:\Documenten\Arduino\libraries\PulseSensor_Playground\src/utility/Interrupts.h:145:7: error: 'TCCR2B' was not declared in this scope

       TCCR2B = 0x05;          // don't force compare, 128 prescaler

       ^~~~~~

D:\Documenten\Arduino\libraries\PulseSensor_Playground\src/utility/Interrupts.h:145:7: note: suggested alternative: 'TCCR0B'

       TCCR2B = 0x05;          // don't force compare, 128 prescaler

       ^~~~~~

       TCCR0B

D:\Documenten\Arduino\libraries\PulseSensor_Playground\src/utility/Interrupts.h:147:9: error: 'OCR2A' was not declared in this scope

         OCR2A = 0XF9;         // set count to 249 for 2mS interrupt

         ^~~~~

D:\Documenten\Arduino\libraries\PulseSensor_Playground\src/utility/Interrupts.h:147:9: note: suggested alternative: 'OCR3A'

         OCR2A = 0XF9;         // set count to 249 for 2mS interrupt

         ^~~~~

         OCR3A

D:\Documenten\Arduino\libraries\PulseSensor_Playground\src/utility/Interrupts.h:151:7: error: 'TIMSK2' was not declared in this scope

       TIMSK2 = 0x02;          // Enable OCR2A match interrupt DISABLE BY SETTING TO 0x00

       ^~~~~~

D:\Documenten\Arduino\libraries\PulseSensor_Playground\src/utility/Interrupts.h:151:7: note: suggested alternative: 'TIMSK0'

       TIMSK2 = 0x02;          // Enable OCR2A match interrupt DISABLE BY SETTING TO 0x00

       ^~~~~~

       TIMSK0

D:\Documenten\Arduino\libraries\PulseSensor_Playground\src/utility/Interrupts.h: In function 'boolean PulseSensorPlaygroundDisableInterrupt()':

D:\Documenten\Arduino\libraries\PulseSensor_Playground\src/utility/Interrupts.h:239:7: error: 'TIMSK2' was not declared in this scope

       TIMSK2 = 0x00;          // Disable OCR2A match interrupt

       ^~~~~~

D:\Documenten\Arduino\libraries\PulseSensor_Playground\src/utility/Interrupts.h:239:7: note: suggested alternative: 'TIMSK0'

       TIMSK2 = 0x00;          // Disable OCR2A match interrupt

       ^~~~~~

       TIMSK0

In file included from D:\Documenten\Arduino\libraries\PulseSensor_Playground\src/PulseSensorPlayground.h:439:0,

                 from D:\Documenten\Arduino\PulseSensor\PulseSensor.ino:13:

D:\Documenten\Arduino\libraries\PulseSensor_Playground\src/utility/Interrupts.h: In function 'boolean PulseSensorPlaygroundEnableInterrupt()':

D:\Documenten\Arduino\libraries\PulseSensor_Playground\src/utility/Interrupts.h:285:7: error: 'TIMSK2' was not declared in this scope

       TIMSK2 = 0x02;          // Enable OCR2A match interrupt

       ^~~~~~

D:\Documenten\Arduino\libraries\PulseSensor_Playground\src/utility/Interrupts.h:285:7: note: suggested alternative: 'TIMSK0'

       TIMSK2 = 0x02;          // Enable OCR2A match interrupt

       ^~~~~~

       TIMSK0

exit status 1
Fout bij het compileren voor board Arduino Leonardo

````

biomurph, 2020-06-11:
@ralphcrutzen 
I'm getting that error when I compile for Leonardo too...
I will have to dig into that. Give me a day to get back to you.
I have a Leonardo here, and will test it.

As for the BPM Alternative problem, I don't understand why that won't work, and will have to try and replicate it.

ralphcrutzen, 2020-06-13:
Hi,
I have decided to use an Arduino Uno, so this issue isn't that important for me personally anymore.
But thanks a lot for your help so far! 👍

biomurph, 2020-06-13:
@ralphcrutzen totally understand. I've solved the compile error for using servo with Leonardo. Going to update the library version this coming week. 
I will also try to replicate your problem and get a fix together.

closed 2023-10-26 by biomurph