WorldFamousElectronics/PulseSensorPlayground #127: get the BPM data

lucasroth, 2020-06-26:
Hi, how can I get the value of the data plotted in the sample PulseSensor_BPM? can´t find documentation of the library´s functions.
Can I retrieve the BPM value, and do it without getting into the library .cpp files?

biomurph, 2020-06-26:
@lucasroth 
What do you mean by 'get the data plotted in the sample'?

You can retrieve the BPM values easily by making simple modification to the example sketch.

lucasroth, 2020-06-27:
hi, indeed realized about the functions available, just use: pulseSensor.getBeatsPerMinute()
by the way, I am using esp32, so changed the function of the mapping data for output:
`
void PulseSensor::readNextSample() {
  // We assume assigning to an int is atomic.
  Signal = map(analogRead(InputPin), 0, 4095, 0, 1023);
}`

the data received doesn´t seem to be near 550 is low as 450, and with low range between high and low levels, any advice on how this can be managed, or how to improve it?

biomurph, 2020-06-27:
@lucasroth 
Instead of modifying the library, try setting the analog resolution
`analogReadResolution(10);`

Also, please send a picture of your Pulse Sensor so I can verify the hardware.

lucasroth, 2020-08-03:
> @lucasroth
> Instead of modifying the library, try setting the analog resolution
> `analogReadResolution(10);`
> 
> Also, please send a picture of your Pulse Sensor so I can verify the hardware.

Hi, thanks for help! sorry for delay have been busy, here are the photos:
[image]
[image]

When using latest lib V1.5 the values go very high near 2000, then just adding analogReadResolution(10); the values drop to 450-500 range. The same happens modifying the lib on: void PulseSensor::readNextSample() { // We assume assigning to an int is atomic. Signal = map(analogRead(InputPin), 0, 4095, 0, 1023); }
[image]

Any advice to make it work using esp32 wroom?

lucasroth, 2020-08-03:
[image]
[image]
better photos

biomurph, 2020-08-03:
Hi Lucas,

I can see by the phots that you have a knock-off Pulse Sensor. Since I don't know what is in that hardware, I really can't help you troubleshoot.
Please support the original makers of Open Source hardware and purchase only legit products.
You can find the Pulse Sensor at
www.pulsesensor.com
www.adafruit.com
www.sparkfun.com
www.mouser.com
www.digikey.com

You will likely get a better result when you get the real deal.

closed 2020-08-03 by biomurph