WorldFamousElectronics/PulseSensorPlayground #128: It does not work on D1mini or esp8266,

YoYoHC, 2020-06-30:
It can run on uno but it does not work on D1mini or Esp8266. Can you help me?

biomurph, 2020-06-30:
I need more information to help you.
What sketch are you trying to run?
Please also attach a picture of your Pulse Sensor to verify the hardware.

YoYoHC, 2020-07-15:
[image]
This is my settings. 
Serial can`t print anything.
I don`t know what`s wrong.

biomurph, 2020-07-15:
Sounds like more of an issue with esp... what sketch are you trying to run? Do you have the right baud rate setting on the serial monitor?

YoYoHC, 2020-07-15:
run the  PulseSensor_BPM   code and i make the right serial monitor  used 115200,but cant print any data.

biomurph, 2020-07-15:
OK, That might be your problem.
Please try to use the PulseSensor_BPM_Alternative code. 
The regular BPM code uses hardware interrupts, and that is likely what the problem is. We are not set up to run hardware interrupts on the ESP modules. The BPM_Alternative uses software timing to get the samples, and it works great.
Let us know how it goes.

YoYoHC, 2020-07-16:
It works.  Thanks!

closed 2020-07-16 by biomurph