WorldFamousElectronics/PulseSensorPlayground #129: Hardware timer interrupts for Adafruit Bluefruit Feather (nRF52832) and Adafruit ItsyBitsy Express BLE (nRF52840)

amanm27, 2020-07-03:
Hello! I'm working on a project looking to extract BPM and related data using the example sketches in the PulseSensor Playground library. Down the road, the project will use an Adafruit ItsyBitsy Express BLE (nRF52840) board, but for the moment I have access to an Adafruit Bluefruit Feather (nRF52832) and I have been experimenting with that. 

I downloaded the most recent code from the GitHub yesterday. I've been messing around with the sketches and realize that only the Alternative sketch (which uses software timer interrupts) is compatible with the Feather; I'm able to view the plot of pulse, BPM, and IBI in the PulseSensor_BPM_Alternative sketch.

Is there any way to run all of the PulseSensor Playground sketches on a Feather or ItsyBitsy using hardware timer interrupts, or is it only possible using software interrupts like in the Alternative sketches? We'd ideally like to use hardware timer interrupts over software timer interrupts for our project if possible.

Thanks!

biomurph, 2020-07-04:
@amanm27 
There have been some implementations of timers on the nRF52840, and from what I've seen they involve digging into the nordic end of things. At least, Adafruit has made is possible to access some of the nordic tools, and also opened up the JTAG so you can go full on nordic if you want.

That said, one of the things that I've run into trouble with in doing timer interrupts on radio modules is that the radio stack will also be running on interrupts, and those are usually held in much higher regard than your use of the lowly timer interrupt, so you will get jitter in your sample rate. Sometimes super bad, not good at all, jitter. That's why we use the software timer, because even if we go down the rabbit hole of getting the timers to work consistently, we still have this problem. 

There are a few work arounds.
Accept the soft timer accuracy as meh, and go with it.
Try to determine the jitter and know when you have samples that are out of sync and accommodate
Turn off the radio (basically try to shut down the BT stack) when you are taking samples

We do have some tools in the library that are not used often to determine jitter. These are in the TimingStatistics files. Exploring that or at least setting up your own method of determining jitter would be the first step, just to see how bad it actually is. Might be acceptable, might not, but that would be the best first thing to do in deciding if you even need the timer. You can get decent machine time using the microseconds() function, and since we're sampling at 500Hz, you'll be able to see any drift or glitches, blorps, in the sample rate pretty clearly.

Next idea is to run the Pulse Sensor acquisition on a different uC so you know you're getting accurate sample rates and then bundle them into packets to send to a dedicated radio. Not sure how accurate you need to be, but if accurate sample rate is mission critical, then I would suggest not doing it on a radio, or at least include the sample rate jitter in the data stream so you know when things get weird.

If you do want to dig into the TimingStatistics and implement it, please let us know as we can help out. That part of the library has got some cobwebs that might need clearing.

Hope that helps!

amanm27, 2020-07-06:
@biomurph Thank you for your quick response! I have a few follow-up questions about taking the software timer approach.

First of all, are there any other major drawbacks in using a software timer approach versus a hardware timer approach besides pinpoint accuracy?

Also, our ItsyBitsy board will be part of a haptic breathing pacer device, and is going to connect with an iPhone app over bluetooth (BLE). If we use software timer interrupts, would any simultaneous bluetooth activity/advertisement be an issue? In other words, in order to get an accurate reading with software interrupts, would we have to temporarily disable any bluetooth connectivity? 

Any guidance on this front would be very much appreciated. Thank you!

biomurph, 2020-07-06:
@amanm27 
Your application sound very interesting, and it also sounds familiar. We know many people have used Pulse Sensor in this kind of application. The only useful answer that I can give you is test test test. You will likely be bundling samples into come kind of compressed data packet for the Bluetooth transmission, and you will also want to do empirical jitter testing. I would suggest that you use software timer interrupts first, because they are easier, and you can easily get a sense of the kind of jitter you are facing. A simple test would be to set up a simple sketch that uses a timer (set to 500Hz, or other frequency) and then have the uC send over serial the actual time between 'samples'. It's best to use a high baud rate when doing these kind of tests so that the serial coms don't get in the way. Alternately, you could only send a serial com if your timing is over a certain amount, like if it's over 1mS or something, but it's probably best to serial log all of the timings to get a sense of the jitter, and do it with various other tings going on. Like advertising, being paired, etc. 
No real difference between hardware timer and software timer if your hardware timer is getting stomped on my the radio...

paris007, 2020-07-07:
@biomurph Thank you for your thoughtful messages.
I wanted clarify a few things here. 
In nRF52832 say I use NRF_TIMER2 to collect the IBIs set NRF_TIMER2->CC[0] = 235;      // value for compare register 0: (1e6/CC[0])/50 = 85Hz(MM3C Resonant Frequency).   Are you suggesting that I will get a lot noise because somehow NRF_TIMER2 and BLE will have issues?

biomurph, 2020-07-07:
@paris007 
That's a good question. It's kind of getting into the weeds of the nRF stuff, and a bit further down the rabbit hole than I am ready to get into right now. Please know that this is an issue that I am very interested in resolving, but that it is also very particular and truly solved on a case by case basis.

That said, here's a few notes on your post.
* I don't know if I can verify your CC setting for the nRF. You have to triple check the data sheet and then empirically test the frequency.
* The Pulse Sensor Playground is set up to sample at 500Hz. It's possible to change that, but you will have to dig into the library and make changes to the way the IBI is calculated. We chose 500Hz because it was recommended as a frequency that is verified to work with PPG. For more on the basic skeleton of the code, please refer to this [old blog post](https://pulsesensor.com/pages/pulse-sensor-amped-arduino-v1dot1), and the reference links at the bottom. 
* I don't know if you will get a lot of noise because of the timer and the BLE interrupts crashing. I do know that when the BLE stack is running, the interrupts that it throws will be of the highest importance. I think that there are ways to set your timer interrupts with very high importance settings, but I don't know if they are high enough to avoid getting stomped by the BLE. This is something that you will have to test.

The good thing is that testing is not very hard. You can even test with dummy data and just report the time between samples in order to determine if you are getting consistent sample time. You may find that it is OK for your application, or you may find that you will need to shut down the stack when you sample the Pulse Sensor.

biomurph, 2020-07-15:
@amanm27 
As I mentioned, I'm working on a project that uses the nRF52840, and I've got a module from Adafrfuit.
In their hardware libraries, they have an example sketch called `hw_systick.ino`. If you load their board platform with the boards manager tool, and then select an nRF52 board from Adafruit, you will see the sketch in 

`File > Examples > Adafruit Bluefruit nRF52 Libraries > Hardware`

It looks like this could be the way to go for setting up a reliable timer to take Pulse Sensor samples at 500Hz. I'll do some testing, and see if I can't get a 'quick and dirty' fix thrown together.

amanm27, 2020-07-17:
@biomurph we were actually able to set up reliable hardware timer interrupts at 500 Hz, which we validated to indeed fire off every 2ms by toggling a GPIO pin. We implemented the interrupts following this example: https://github.com/NordicPlayground/nrf51-TIMER-examples/blob/master/timer_example_timer_mode/main.c

biomurph, 2020-07-17:
That's great! How does it work when you run the BLE stack?

amanm27, 2020-07-17:
@biomurph Our sketch also includes bluetooth advertising at the same time, so looks like the interrupts seemed to work fine with that as well!

biomurph, 2023-10-26:
PulseSensor Playground Library supports nRF52

closed 2023-10-26 by biomurph