WorldFamousElectronics/PulseSensorPlayground #130: Arduino due doesn't work

arashsoftco, 2020-07-20:
hi this library doesn't work in arduino due:
```
In file included from C:\Users\itosoft\Documents\Arduino\libraries\PulseSensorPlayground-master\src/PulseSensorPlayground.h:439:0,

                 from C:\Users\itosoft\Documents\Arduino\libraries\PulseSensorPlayground-master\src\PulseSensorPlayground.cpp:16:

C:\Users\itosoft\Documents\Arduino\libraries\PulseSensorPlayground-master\src\PulseSensorPlayground.cpp: In member function 'boolean PulseSensorPlayground::sawNewSample()':

C:\Users\itosoft\Documents\Arduino\libraries\PulseSensorPlayground-master\src/utility/Interrupts.h:72:45: error: 'cli' was not declared in this scope

 #define DISABLE_PULSE_SENSOR_INTERRUPTS cli()

                                             ^

C:\Users\itosoft\Documents\Arduino\libraries\PulseSensorPlayground-master\src\PulseSensorPlayground.cpp:105:5: note: in expansion of macro 'DISABLE_PULSE_SENSOR_INTERRUPTS'

     DISABLE_PULSE_SENSOR_INTERRUPTS;

     ^

C:\Users\itosoft\Documents\Arduino\libraries\PulseSensorPlayground-master\src/utility/Interrupts.h:73:44: error: 'sei' was not declared in this scope

 #define ENABLE_PULSE_SENSOR_INTERRUPTS sei()

                                            ^

C:\Users\itosoft\Documents\Arduino\libraries\PulseSensorPlayground-master\src\PulseSensorPlayground.cpp:108:5: note: in expansion of macro 'ENABLE_PULSE_SENSOR_INTERRUPTS'

     ENABLE_PULSE_SENSOR_INTERRUPTS;

     ^

exit status 1
Error compiling for board Arduino Due (Programming Port).

```

biomurph, 2020-07-20:
@arashsoftco What sketch example are you using? Please try the BPM_Alternative.ino sketch.

Sudomarko, 2023-01-20:
I'm also having this issue when using the Due, I went to the Interrupts.h file and changed all instances of cli() to interrupts(), sli() to noInterrupts(). Now it works but it is reading extreme values when I'm 100% positive that the Arduino connections are correct due to the fact that it works on the Uno. 

When I set the USE_ARDUINO_INTERRUPTS to false it started giving different extreme values. This leads me to the conclusion that this is a library issue. What could be the cause of this in the library other than interrupts?

Sudomarko, 2023-01-20:
I'll look for more possible reasons, probably something with the timer code and upload it here. It would be really nice if you could direct me to the right place though.

biomurph, 2023-01-20:
@Sudomarko You are on the right track. I don't know if we need to use those defines. In the Interrupt.h file there are a number of board architectures specified that don't define a value for those items.
This forum post has what looks like a good srtarting point for adding hardware timer support for the Due.
https://forum.arduino.cc/t/timer-interrupts-on-due/127643/17

We want to sample the PulseSensor at 500Hz. (every 2 milliseconds). There are other places the code will need to be adjusted to add support for Due hardware timers, but just getting the timer to function is a big step.

Sudomarko, 2023-01-20:
Thanks, will keep you posted over the next couple of days.

biomurph, 2023-10-26:
PulseSenor Playground Library supports Arduino Due

closed 2023-10-26 by biomurph