WorldFamousElectronics/PulseSensorPlayground #132: Does this work in Arduino Nano 33 IOT.

bleedweedsuz, 2020-08-26:
Product Url: [Arduino Nano 33 IOT](https://store.arduino.cc/usa/nano-33-iot)

I tried Getting_BPM_to_Monitor.ino example where interrupt is true. **pulseSensor.begin()** gives false. 
Does this means interrupt version does not work in Arduino Nano 33 IOT? 

There is also Problem with Serial initialization too i need to do this in this product: (my be this is issue with Serial in this device)
```
Serial.begin(9600);
while(!Serial){  } //basically waiting for to Serial to initialize
```

so I m thinking like do i need to wait for pulseSensor to initialize? 
I m confuse. Can u test it in "Arduino Nano 33 IOT" post some result here?

p.s i used alternative one it is working.

biomurph, 2020-08-26:
@bleedweedsuz 
You must use the BPM_Alternative example code. Hardware timer interrupts are not yet supported for the nRF52840 devices.

Yes, you must wait for the serial to be connected in order to use the serial port. The two lines of code you put will work to do that.

bleedweedsuz, 2020-08-26:
Thanks for the reply mate, i will use alternative one.

closed 2020-08-26 by bleedweedsuz