WorldFamousElectronics/PulseSensorPlayground #133: PulseSensor + Feather nRF52840 Bluefruit LE

marcjan, 2020-08-26:
Hi,

I bought original PulseSensor and Feather nRF52840 Bluefruit LE, but values which I get are very bad - doesnt looks like heartbeat.
[image]
Heartbeats are not detected, condition `if sawStartOfBeat` is always false. I use the code from alternative sketch.
I read other issues which describe problem with timer, is there any easy workaround for that?

Thanks
Marcin

biomurph, 2020-08-26:
@marcjan 
That does not look like a good signal at all.
Please send a picture of your physical setup. That can help sometimes. What pin are you connecting the Pulse Sensor to?
Also, what exact Adafruit module are you using? They have a few nRF52840 boards.

marcjan, 2020-08-26:
@biomurph Thanks for quick response :)

https://www.adafruit.com/product/4516
[image]

I soldered directly GND and VCC to board and results are far better, but they are too high. A2 Has signal from Pulsesensor.

[image]

I should play with THRESHOLD parameter?

Thanks!

biomurph, 2020-08-26:
OK, that looks much better on the plotter.

OK, what will you compare it to?

marcjan, 2020-08-26:
With some professional medical device. So there is no timer problems on nRF52840 (Ive seen some issues)?

biomurph, 2020-08-26:
Some folks have worked out hardware timer issues, but they are not tested by us so not in the library. Check the issues tab for the thread that goes into it. Reach out to them there, as they may share code

marcjan, 2020-08-30:
Hi again ;)

It looks like results from pulsesensor are wrong. While correct result on medical device is 59bpms pulsesensor shows ~83. I left only pulsesensor code. Results usually are 30bpms higher. Board works with 3.3V. I get the results in following way:

```
const int THRESHOLD = 500;   // Adjust this number to avoid noise when idle
const byte SAMPLES_PER_SERIAL_SAMPLE = 10;

if (pulseSensor.sawNewSample()) {

  if (--samplesUntilReport == (byte) 0) {

    samplesUntilReport = SAMPLES_PER_SERIAL_SAMPLE;

    if (pulseSensor.sawStartOfBeat()) {

      // I use this function when I send it over Bluetooth

      uint16_t heartBeatValue = pulseSensor.getBeatsPerMinute();

      // I use this function to debug it in serial monitor

      pulseSensor.outputBeat();

    }
  }
}
```

Thank you for  help
Marcin

biomurph, 2020-08-30:
@marcjan 
Ah yes, I'm remembering now. 
There is a known issue with the nRF52 timing. The micros() function (which we use in the library) is stunningly low resolution
Here's a post about the issue
https://github.com/adafruit/Adafruit_nRF52_Arduino/issues/451
Also, There is an other issue thread that discusses the problem here
https://github.com/WorldFamousElectronics/PulseSensorPlayground/issues/129

This is a feature that we want to include in future library functionality. It's not there yet.
If you go to that other issue (129) and @ the author, you may get their attention and ask them to share their solution?

Our timeline for addressing and testing this for the library may be longer than you need for your purpose.

marcjan, 2020-08-30:
@biomurph Ok, thx for clarification and help.

Best regards
Marcin

marcjan, 2020-09-05:
@biomurph I have ported the code to xmega and setup 2ms interrupts with 12 bits ADC (0-4096). Could you tell me how I should calibrate these variables, cause my readings are now about 108 BPM which are much to high:
P, T, threshold and IBI.
Currently I have:
P- 3100, T- 3100, threshold - 3201, amplitude - 100, IBI - 750,

Thanks
Marcin

biomurph, 2020-09-05:
@marcjan   
I don't think I understand what you mean by 'ported to xmega'
The serial feed you're showing has the numbers going from very high to very low. 
There's not enough information to really help you debug. Can you share your code? Do you have a link to a repo with the code? Also, details on the hardware you're using. 
Can you verify that you are getting a 500Hz interrupt?

marcjan, 2020-09-05:
@biomurph 
Xmega128a3-AU
Screens above show in one line BPM and in next one IBI and again and again...
I found that internal oscillator is very inaccurate, so I put external one (16 Mhz)
here is the code:
https://gist.github.com/marcjan/3203eb519b16e9735288f3e3799cca4a

Thanks
Marcin

biomurph, 2020-09-08:
@marcjan it looks like the link to your repo is broken.
Can you tell me what your bug was? I'm very curious.

marcjan, 2020-09-09:
@biomurph Thanks for your time again. 
I was turning off/on timer instead of disabling and enabling interrupts :)
BTW when you plan to make your code compataible with Adafruit Feather nRF52840 Bluefruit Sense LE?

biomurph, 2020-09-09:
Thanks for letting me know your bug

We will be testing this fall

closed 2020-09-11 by marcjan

marcjan, 2023-01-06:
Hello @biomurph :)

It's been some time since last post :)
Did you manage to make a code compatible with the Adafruit Feather nRF52840?

Thanks,
Marcin

biomurph, 2023-01-06:
@marcjan i think this is the best lead for a reliable timer interrupt source for the nRF52 boards. I think it also includes the Feather variant that your are talking about in the issue title.
This would be my first route for testing with Pulse Sensor Playground. 
Would you like to try?