WorldFamousElectronics/PulseSensorPlayground #135: I cannot get the peak sometimes

DoanThu, 2020-12-06:
Hi, I have four sensors as below:
[image]
For the third one, I have no idea why sometimes I cannot get peaks as they are totally flat at around 966-968 and cannot reach higher. I'm using PTT sketch and Processing together.
I also took a look at the code of both Arduino and Processing but found nothing that capped these signals.
Please help me find a way to fix this.

biomurph, 2020-12-06:
What your experiencing is a phenomenon known as clipping. The signal itself does conceivably want to go higher, but the voltage range available is capped at 5:00 or 3.3 depending upon your system. If you really need to get the peak, you can make adjustments to the position and or pressure of the post sensor on your body. In any case, the signal is so big and bright that it is impossible to miss. In our research on finding the moment of heartbeat, we discovered that the steepest slope of the rising edge of the pulse wave is a very close approximation to the moment of heartbeat. So finding the peak is not necessary for our algorithm.

DoanThu, 2020-12-06:
> What your experiencing is a phenomenon known as clipping. The signal itself does conceivably want to go higher, but the voltage range available is capped at 5:00 or 3.3 depending upon your system. [...]

Thanks for your info. I just tested and realize if I use 3.3V, it is capped at 600 something and at 969 when it comes to 5V. I'm wondering if I use another board that supports higher voltage, for example, [this one](https://sg.element14.com/arduino/a000066/arduino-uno-evaluation-board/dp/2075382?gclid=CjwKCAiAn7L-BRBbEiwAl9UtkPp0Pjs-RLtUe2FtVzRGuU1dVwUoiBLw4GDYErLuZfNZYL6EjdPazhoC5b4QAvD_BwE&mckv=sOeVvab4e_dc|pcrid|432679002659|pkw||pmt||slid||product|2075382|pgrid|100843414295|ptaid|pla-900378677730|&CMP=KNC-GSG-GEN-SHOPPING-SBC-EMBEDDED). will it be capped at higher numbers and still work with the given code?

biomurph, 2020-12-08:
Take a look at this page in Arduino site
https://www.arduino.cc/reference/en/language/functions/analog-io/analogreference/
it describes the analog reference pin. What you are learning is that the output of the analog sensor (in this case, PulseSensor) will not go above the voltage that is powering it. When you power PulseSensor with 3.3V, you will only get the highest value of 3.3V. The analog to digital converter reads the voltage and makes it a number between 0 and 1023 (10 bit). If the analog read is powered by 5V, and you put a sensor powered by 3V, then you will get a reduced range of values. That's why you would use the Analog Reference pin (ARef) where you put the voltage of the sensor so that the Arduino knows what the highest value is.

DoanThu, 2020-12-08:
> What your experiencing is a phenomenon known as clipping. The signal itself does conceivably want to go higher, but the voltage range available is capped at 5:00 or 3.3 depending upon your system. [...]

One more thing, may I know if the peak finding algorithm is not necessary to your team, is it still correct in case the signals are clipped?

biomurph, 2020-12-08:
Please read this pages
https://pulsesensor.com/pages/open-hardware
https://pulsesensor.com/blogs/news/6877119-announcing-code-version-1-1-its-out-of-beta
and follow the links from them to other documentation.

We do our best to provide accurate working code, but we don't warranty anything and we don't recommend the PulseSensor for medical uses. 

To answer your question, yes. The algorithm does find the BPM to be accurate enough for our goals.

closed 2023-10-26 by biomurph