WorldFamousElectronics/PulseSensorPlayground #136: Possible to control green light on PulseSensor?

amanm27, 2020-12-17:
Hello! I was wondering if there's a way to control (ideally through code in Arduino IDE) when the green light on the PulseSensor is on or off? We have a PulseSensor connected to an Adafruit ItsyBitsy, which is part of a vibrotactile breathing pacer. We have a feeling that the PulseSensor might be drawing a good amount of energy through the LED, which is causing our PulseSensor-integrated device to misbehave compared to another device which isn't connected to a PulseSensor. Any guidance on this would be greatly appreciated. Thank you!

biomurph, 2020-12-18:
@amanm27 The itsy bitsy has an ATmega 32u4 on it running at 5V, 16Mz. It's basically a shrunken Leonardo.

The Pulse Sensor draws just around 3mA of current, so it's not taking too much of a gulp off your power supply.
What is the behavior that you're seeing? Please discribe.

Pulse Sensor can be powered by any of the digital pins from the microcontroller. HIGH on the pin will power the Pulse Sensor, LOW will turn it off. Easy peasy.

The Pulse Sensor is known to be susceptible to power line noise. It is recommended that you place a capacitor accross the rails right next to where the Pulse Sensor is connected to the board. A value of 0.1uF to 1uF should be OK. This could help in cleaning up your issue? What is your issue? What is your code?

paris007, 2020-12-23:
@biomurph Thank you for your quick response. We are still trying to figure out what is causing the extreme power draw in our device after we added the pulse sensor. I guess our main question is how to turn off the green light on and off as needed through software. We would like the green light to be on only when we are collecting data and off when we are not.  Thank you in advance.

biomurph, 2020-12-24:
@amanm27 

As mentioned, you can connect the Pulse Sensor RED wire to any GPIO, and then set that GPIO HIGH to turn it on, and LOW to turn it off. 
If you can send a schematic of your project, I am happy to troubleshoot with you as much as I can.

paris007, 2023-05-19:
Hi @biomurph I am following up on this after couple of years :) 

We are using https://www.adafruit.com/product/4481  
In our code we have 

The pulseSensorPin is connected to A0. 

and I am using it in

 void TIMER3_IRQHandler()
 {
    if (NRF_TIMER3->EVENTS_COMPARE[0] ) { // if ((NRF_TIMER3->EVENTS_COMPARE[0] != 0) && ((NRF_TIMER3->INTENSET & TIMER_INTENSET_COMPARE0_Msk) != 0))... commonly used, do we need this condition?
      Signal = analogRead(pulseSensorPin);
      sampleCounter += 2;                           // keep track of the time in mS with this variable
      N = sampleCounter - lastBeatTime;             // monitor the time since the last beat to avoid noise

A0 is a GPIO pin. 

I can set it to LOW in the  void setup() 

  digitalWrite(pulseSensorPin, LOW);

so it can be off by default. Then when I receive a command from my app, I set the pin to HIGH. I am ssume that then eventHandler attached to A0 for the pulse sensor will start reading and so does not need modification in code. Then when I am done, I will send another command from my app to set the pulseSensorPin to Low again. 

Is this the right way of thinking about it?

biomurph, 2023-05-22:
@paris007 
I am not sure that you have it right.
In order to control the power to the PulseSensor, you need to connect the red wire to a GPIO pin. Say pin 9 for example.
Then, when pin 9 is HIGH, the PulseSensor is on, and when pin 9 is LOW, the PulseSensor is off.

You don't want to drive the pin A0 HIGH or LOW. A0 is the analog pin that you should connect the PulseSensor purple wire to.

Also, we have a couple of examples for [nRF52840](https://github.com/WorldFamousElectronics/PulseSensorPlayground/tree/master/examples) in the PulseSensor Playground that have been tested against Adafruit Feather. Check them out!

closed 2023-10-26 by biomurph