WorldFamousElectronics/PulseSensorPlayground #138: Brand New Pulse Sensor Not Working

AnyaSengupta, 2021-01-18:
I got a new pulse sensor today in the mail and I was running "Getting_BPM_to_Monitor" to get inaccurate data like this. I tried other  example files and it's still inaccurate. 
[image]

biomurph, 2021-01-18:
@AnyaSengupta 
I need more information to help you with your problem.
What arduino hardware are you using?
You've modified the serial output. How else have you modified the code?
Send a picture of your hardware setup and the Pulse Sensor that you got.

AnyaSengupta, 2021-01-18:
I am using Arduino Uno. The black wire goes to ground, red goes to power, and purple goes to Analog 0. This is the [model ](https://www.amazon.com/gp/product/B01CPP4QM0/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1)I have. Here is the simple code. 

`
/*  Getting_BPM_to_Monitor prints the BPM to the Serial Monitor, using the least lines of code and PulseSensor Library.
 *  Tutorial Webpage: https://pulsesensor.com/pages/getting-advanced
 *
--------Use This Sketch To------------------------------------------
1) Displays user's live and changing BPM, Beats Per Minute, in Arduino's native Serial Monitor.
2) Print: "♥  A HeartBeat Happened !" when a beat is detected, live.
2) Learn about using a PulseSensor Library "Object".
4) Blinks LED on PIN 13 with user's Heartbeat.
--------------------------------------------------------------------*/

#define USE_ARDUINO_INTERRUPTS true    // Set-up low-level interrupts for most acurate BPM math.
#include <PulseSensorPlayground.h>     // Includes the PulseSensorPlayground Library.   

//  Variables
const int PulseWire = 0;       // PulseSensor PURPLE WIRE connected to ANALOG PIN 0
int Threshold = 550;           // Determine which Signal to "count as a beat" and which to ignore.
                               // Use the "Gettting Started Project" to fine-tune Threshold Value beyond default setting.
                               // Otherwise leave the default "550" value. 
                               
PulseSensorPlayground pulseSensor;  // Creates an instance of the PulseSensorPlayground object called "pulseSensor"

void setup() {   

  Serial.begin(9600);          // For Serial Monitor

  // Configure the PulseSensor object, by assigning our variables to it. 
  pulseSensor.analogInput(PulseWire);  
  pulseSensor.setThreshold(Threshold);   

  // Double-check the "pulseSensor" object was created and "began" seeing a signal. 
   if (pulseSensor.begin()) {
    Serial.println("We created a pulseSensor Object !");  //This prints one time at Arduino power-up,  or on Arduino reset.  
  }
}

void loop() {

 int myBPM = pulseSensor.getBeatsPerMinute() - 30;  // Calls function on our pulseSensor object that returns BPM as an "int".
                                               // "myBPM" hold this BPM value now. 

if (pulseSensor.sawStartOfBeat()) {            // Constantly test to see if "a beat happened". 
 //Serial.println("♥  A HeartBeat Happened ! "); // If test is "true", print a message "a heartbeat happened".
 Serial.print("BPM: ");                        // Print phrase "BPM: " 
 Serial.println(myBPM);                        // Print the value inside of myBPM. 
}

  delay(20);                    // considered best practice in a simple sketch.

}

  `
Thank you!

biomurph, 2021-01-19:
Thanks for the info.

Where are you placint the Pulse Sensor on your body? Sometimes the sensor can be held too tight or too loosely on your finger.
Please try uploading the PulseSensor_BPM.ino sketch, and open up the Arduino Serial Plotter to take a look at the signal.
Follow this written tutorial to see the data stream, and take a screen shot to share it here.

AnyaSengupta, 2021-01-19:
I am placing it on my thumb. Here is the screenshot. Thank you
[image]

biomurph, 2021-01-19:
@AnyaSengupta while that is a very noisy signal, I can see your pulse in there. It's the higher peaks that don't saturate.there may be a few things contributing to the noise. Are you moving around at all? How is your computer powered? If it is plugged into the wall outlet, that may add noise, although this does not look too much like that. Try using different pressure and body parts to put the sensor and watch the signal trace to see what the best signal is

closed 2023-10-26 by biomurph