WorldFamousElectronics/PulseSensorPlayground #139: Extremely High Raw Sensor Values

sam-shridhar1950f, 2021-01-24:
I am getting values over 900 consistently. This is a new product. I am using the "Getting Started" code from the playground. 
[image]
[image]

[image]

bolek19, 2021-01-24:
Was ist das ja ein Produkt

sam-shridhar1950f, 2021-01-24:
> Was ist das ja ein Produkt
Yes, it is the Pulse sensor

biomurph, 2021-01-25:
@sam-shridhar1950f 
That is some noise for sure.
 
Interesting that it is all at the high end of the ADC and the range of values is under 100 points. 
Can you tell me more about your setup? Your Pulse Sensor does look like it's legit, but it's super out of focus photo.

How do you have it connected? Do you have the clear vinyl sticker on the front of the sensor?

sam-shridhar1950f, 2021-01-25:
@biomurph 

It is connected to an Arduino Uno with all the recommended connections (A0, 5V, GND). I am "goldilocks" pressing the sensor. Yes, I see the vinyl sticker.

biomurph, 2021-01-25:
@sam-shridhar1950f 
When you are *not* touching the sensor, the signal should settle around V/2 and the ADC result should be somewhere about 500 counts. Do you find that is the case?
What kind of computer are you using? Are you plugged into a wall outlet? Sometimes Pulse Sensor can be susceptible to power line noise.

sam-shridhar1950f, 2021-01-25:
@biomurph 

No. When I am not touching the sensor the values stay at the 900s-1000s. I am using a Microsoft Surface Book 3, and the arduino is connected via USB to the laptop.

biomurph, 2021-01-25:
This does sound like something is wrong with the sensor. Can you dm me? Or email joel-at-pulsesensor-dot-com

sam-shridhar1950f, 2021-01-25:
Sure, sent you a "hello."

closed 2023-10-26 by biomurph