WorldFamousElectronics/PulseSensorPlayground #140: Getting BPM to Monitor not Working on Arduino Nano 33 BLE Sense??

LavenderLemonade, 2021-03-25:
Hello, so I've been working on getting the 'Getting BPM to Monitor' code to work on the Arduino Nano 33 BLE Sense. Looking through some of the other code here, I made sure to clone the repository into the library directory, so that it's all at least compiling. But once the code is uploaded, nothing happens, no LED flashing or anything going to the Serial monitor. What's going on? 

`/*  Getting_BPM_to_Monitor prints the BPM to the Serial Monitor, using the least lines of code and PulseSensor Library.
 *  Tutorial Webpage: https://pulsesensor.com/pages/getting-advanced
 *
--------Use This Sketch To------------------------------------------
1) Displays user's live and changing BPM, Beats Per Minute, in Arduino's native Serial Monitor.
2) Print: "♥  A HeartBeat Happened !" when a beat is detected, live.
2) Learn about using a PulseSensor Library "Object".
4) Blinks LED on PIN 13 with user's Heartbeat.
--------------------------------------------------------------------*/

#define USE_ARDUINO_INTERRUPTS true    // Set-up low-level interrupts for most acurate BPM math.
#include <PulseSensorPlayground.h>  // Includes the PulseSensorPlayground Library.   

//  Variables
const int PulseWire = 0;       // PulseSensor PURPLE WIRE connected to ANALOG PIN 0
const int LED13 = 13;          // The on-board Arduino LED, close to PIN 13.
int Threshold = 550;           // Determine which Signal to "count as a beat" and which to ignore.
                               // Use the "Gettting Started Project" to fine-tune Threshold Value beyond default setting.
                               // Otherwise leave the default "550" value. 
                               
PulseSensorPlayground pulseSensor;  // Creates an instance of the PulseSensorPlayground object called "pulseSensor"

void setup() {   

  Serial.begin(9600);          // For Serial Monitor

  // Configure the PulseSensor object, by assigning our variables to it. 
  pulseSensor.analogInput(PulseWire);   
  pulseSensor.blinkOnPulse(LED13);       //auto-magically blink Arduino's LED with heartbeat.
  pulseSensor.setThreshold(Threshold);   

  // Double-check the "pulseSensor" object was created and "began" seeing a signal. 
   if (pulseSensor.begin()) {
    Serial.println("We created a pulseSensor Object !");  //This prints one time at Arduino power-up,  or on Arduino reset.  
  }
}

void loop() {

 int myBPM = pulseSensor.getBeatsPerMinute();  // Calls function on our pulseSensor object that returns BPM as an "int".
                                               // "myBPM" hold this BPM value now. 

if (pulseSensor.sawStartOfBeat()) {            // Constantly test to see if "a beat happened". 
 Serial.println("♥  A HeartBeat Happened ! "); // If test is "true", print a message "a heartbeat happened".
 Serial.print("BPM: ");                        // Print phrase "BPM: " 
 Serial.println(myBPM);                        // Print the value inside of myBPM. 
}

  delay(20);                    // considered best practice in a simple sketch.

}

  `

I haven't changed the code or anything, and this code works perfectly well with an Arduino UNO. I hope to eventually send this serial output to MATLAB via Bluetooth, but I'm not sure how I'll be able to do this if it's not even seeming to work here. All of my libraries are updated, there are no error messages, and my personal laptop is up-to-date as well. Any ideas? Does the library still not support interrupts on the Arduino Nano 33 BLE Sense?

biomurph, 2021-03-25:
@LavenderLemonade 
The sketch you are using is set up to function with hardware interrupts.
We don't support hardware interrupts on the Nano 33 BLE yet.
I suggest you use the BPM_Alternative.ino sketch from the library, and modify it if you want to output the BPM only.

tom-villanueva, 2023-10-23:
Hello! Any updates for the Arduino Nano 33 BLE?

biomurph, 2023-10-24:
@tom-villanueva 
The Arduino Nano33 uses a uBlox radio that is based on the nRF52840
We have code that works very well for the nRF35840, called PulseSensor_nRF52.ino
This code example is kind of like two examples in one, because we have different methods for targeting different hardware in the same example. We will be changing that to avoid confusion in the future, but for now you have to setup the example correctly for your hardware.

In the example as it is shipped, it is designed to run on Adafruit or Seeed boards.
To make it run on the MBED (like Nano 33 BLE) you have to comment out some lines and comment in others.
Here are the relevant parts to look for in the example sketch. Make sure that the code specified for MBED is uncommented, and also that the code specified for Adafruit or Seeed is commented out.

```
/*
    If you are using an Adafruit or Seeed nRF52 platform,
    uncomment the next line to use the NRF52TimerInterrupt library
*/
#include "NRF52TimerInterrupt.h"

/*
    If you are using an MBED nRF52 platform, like Nano 33 BLE,
    uncomment the next lines to use the NRF52_MBED_TimerInterrupt library
*/
// #include <NRF52_MBED_TimerInterrupt.h>
// #include <NRF52_MBED_TimerInterrupt.hpp>
// #include <NRF52_MBED_ISR_Timer.h>
// #include <NRF52_MBED_ISR_Timer.hpp>

#define TIMER3_INTERVAL_US        2000 // critical fine tuning here!

/*
    If you are using an Adafruit or Seeed nRF52 platform,
    uncomment the next line to use the NRF52TimerInterrupt library
*/
NRF52Timer sampleTimer(NRF_TIMER_3);

/*
    If you are using an MBED nRF52 platform, like Nano 33 BLE,
    uncomment the next lines to use the NRF52_MBED_TimerInterrupt library
*/
// NRF52_MBED_Timer sampleTimer(NRF_TIMER_3);

```

tom-villanueva, 2023-10-26:
> @tom-villanueva The Arduino Nano33 uses a uBlox radio that is based on the nRF52840 We have code that works very well for the nRF52840, called PulseSensor_nRF52.ino This code example is kind of like two examples in one, because we have different methods for targeting different hardware in the same example. We will be changing that to avoid confusion in the future, but for now you have to setup the example correctly for your hardware.
> 
> In the example as it is shipped, it is designed to run on Adafruit or Seeed boards. To make it run on the MBED (like Nano 33 BLE) you have to comment out some lines and comment in others. Here are the relevant parts to look for in the example sketch. Make sure that the code specified for MBED is uncommented, and also that the code specified for Adafruit or Seeed is commented out.
> 
> ```
> /*
>     If you are using an Adafruit or Seeed nRF52 platform,
>     uncomment the next line to use the NRF52TimerInterrupt library
> */
> #include "NRF52TimerInterrupt.h"
> 
> /*
>     If you are using an MBED nRF52 platform, like Nano 33 BLE,
>     uncomment the next lines to use the NRF52_MBED_TimerInterrupt library
> */
> // #include <NRF52_MBED_TimerInterrupt.h>
> // #include <NRF52_MBED_TimerInterrupt.hpp>
> // #include <NRF52_MBED_ISR_Timer.h>
> // #include <NRF52_MBED_ISR_Timer.hpp>
> 
> #define TIMER3_INTERVAL_US        2000 // critical fine tuning here!
> 
> /*
>     If you are using an Adafruit or Seeed nRF52 platform,
>     uncomment the next line to use the NRF52TimerInterrupt library
> */
> NRF52Timer sampleTimer(NRF_TIMER_3);
> 
> /*
>     If you are using an MBED nRF52 platform, like Nano 33 BLE,
>     uncomment the next lines to use the NRF52_MBED_TimerInterrupt library
> */
> // NRF52_MBED_Timer sampleTimer(NRF_TIMER_3);
> ```

Hi, can't really make the code work on Nano 33 BLE, nor rev1 or rev2. The program just hangs and can't even open the Serial Monitor on Arduino IDE. Not really sure how can I make it work.
I tried the BPM_alternative sketch, it outputs BPM values but are way off. Any leads that you can think of?

biomurph, 2023-10-26:
PulseSenor Playground Library supports nRF52 boards

biomurph, 2023-10-26:
@tom-villanueva 
I will have to dig up a Nano33 BLE to see if I can replicate your issue.
Are you making any other modifications to the code? Trying to interface with other hardware?
Are you trying to use the Bluetooth?

tom-villanueva, 2023-10-26:
@biomurph 
Tried with my other sketch that uses the BLE, didn't work, so I tried with the example (commenting and uncommenting as you said), but had no luck. The only thing I commented out were the blink and fade on pulse.

biomurph, 2023-10-26:
I have poked around the lab and I don't have a Nano 33 BLE on hand. Must have wound up in a project somewhere.

I have a board on order. When it comes in I will try to replicate your issue. 
Do you have any other hardware in hand to test to at least see that you have the PulseSensor working?
Most likely, I will be able to reply next week.

tom-villanueva, 2023-10-26:
Thanks for your time!! Really apprecciate it. I tested with the UNO and it's outputting normal BPM values.

biomurph, 2023-11-06:
@tom-villanueva 
I have gotten a couple of Nano 33 BLE boards, and I have been messing around with them.
Unfortunately, I can reproduce your issue. Worse, I think I may have bricked them, as they don't show up on the serial port.
Will have to re-bootload them and hopefully get them back on line.

I also uploaded our BPM_Alternative code example and it works well.
Please use that code base to work from?
Or choose different hardware?

I am not sure what the issue we are having might be. I used different timers (3 and 4) to test and both are not fun.

tom-villanueva, 2023-11-08:
Perfect! Thanks, I'll try to work with that example

biomurph, 2024-01-05:
@tom-villanueva 
Please switch the Playground repository branch to the 'Version-2.0-beta' branch.
Then, delete the library you have installed, and download (or fork) the Version-2.0-beta branch into your Arduino/libraries folder.
The v2 has vast improvements and will work much better with your target hardware.

biomurph, 2024-03-05:
This is closed because V2 solves the problem

closed 2024-03-05 by biomurph