WorldFamousElectronics/PulseSensorPlayground #141: Arduino Pro mini 3V 8MHz - Pulse sensor connected on port A1 - ONLY NOISE

yboujraf, 2021-06-02:
Dear @biomurph 

Setup :

**Arduino pro mini 3V 8MHz ATM328P but powered through usb for debugging.**

**The port used for the sensor is : A1**

[image]

I tried with PulseSensorPlayground library and examples I capturing noise and inaccurate BPM value

7:51:28.978 -> We created a pulseSensor Object !
17:51:29.876 -> ♥  A HeartBeat Happened ! 
17:51:29.923 -> BPM: 146
17:51:30.111 -> ♥  A HeartBeat Happened ! 
17:51:30.158 -> BPM: 152
17:51:30.342 -> ♥  A HeartBeat Happened ! 
17:51:30.389 -> BPM: 158
17:51:30.621 -> ♥  A HeartBeat Happened ! 
17:51:30.666 -> BPM: 165
17:51:30.900 -> ♥  A HeartBeat Happened ! 
17:51:30.900 -> BPM: 172
17:51:31.134 -> ♥  A HeartBeat Happened ! 
17:51:31.180 -> BPM: 181
17:51:31.417 -> ♥  A HeartBeat Happened ! 
17:51:31.463 -> BPM: 186
17:51:31.745 -> ♥  A HeartBeat Happened ! 
17:51:31.791 -> BPM: 194
17:51:32.501 -> ♥  A HeartBeat Happened ! 
17:51:32.501 -> BPM: 175
17:51:32.924 -> ♥  A HeartBeat Happened ! 
17:51:32.971 -> BPM: 172
17:51:33.208 -> ♥  A HeartBeat Happened ! 
17:51:33.254 -> BPM: 179
17:51:33.486 -> ♥  A HeartBeat Happened ! 
17:51:33.486 -> BPM: 179
17:51:35.590 -> ♥  A HeartBeat Happened ! 
17:51:35.636 -> BPM: 115
17:51:36.887 -> ♥  A HeartBeat Happened ! 
17:51:36.934 -> BPM: 96
17:51:37.791 -> ♥  A HeartBeat Happened ! 
17:51:37.838 -> BPM: 86
17:51:38.339 -> ♥  A HeartBeat Happened ! 
17:51:38.392 -> BPM: 83
17:51:38.693 -> ♥  A HeartBeat Happened ! 
17:51:38.693 -> BPM: 83
17:51:38.995 -> ♥  A HeartBeat Happened ! 
17:51:38.995 -> BPM: 83
17:51:39.245 -> ♥  A HeartBeat Happened ! 
17:51:39.245 -> BPM: 89
17:51:39.497 -> ♥  A HeartBeat Happened ! 
17:51:39.497 -> BPM: 92
17:51:39.745 -> ♥  A HeartBeat Happened ! 
17:51:39.745 -> BPM: 92
17:51:40.000 -> ♥  A HeartBeat Happened ! 
17:51:40.047 -> BPM: 92
17:51:40.247 -> ♥  A HeartBeat Happened ! 
17:51:40.247 -> BPM: 129
17:51:40.548 -> ♥  A HeartBeat Happened ! 
17:51:40.602 -> BPM: 164
17:51:40.803 -> ♥  A HeartBeat Happened ! 
17:51:40.850 -> BPM: 200
17:51:41.051 -> ♥  A HeartBeat Happened ! 
17:51:41.051 -> BPM: 222
17:51:41.304 -> ♥  A HeartBeat Happened ! 
17:51:41.351 -> BPM: 229
17:51:41.605 -> ♥  A HeartBeat Happened ! 
17:51:41.652 -> BPM: 228
17:51:41.853 -> ♥  A HeartBeat Happened ! 
17:51:41.906 -> BPM: 228
17:51:42.556 -> ♥  A HeartBeat Happened ! 
17:51:42.556 -> BPM: 198
17:51:42.958 -> ♥  A HeartBeat Happened ! 
17:51:43.011 -> BPM: 187
17:51:43.212 -> ♥  A HeartBeat Happened ! 
17:51:43.245 -> BPM: 186
17:51:43.459 -> ♥  A HeartBeat Happened ! 
17:51:43.513 -> BPM: 186
17:51:43.714 -> ♥  A HeartBeat Happened ! 
17:51:43.760 -> BPM: 189
17:51:44.062 -> ♥  A HeartBeat Happened ! 
17:51:44.116 -> BPM: 184
17:51:44.317 -> ♥  A HeartBeat Happened ! 
17:51:44.364 -> BPM: 183
17:51:44.619 -> ♥  A HeartBeat Happened ! 
17:51:44.619 -> BPM: 182
17:51:44.866 -> ♥  A HeartBeat Happened ! 
17:51:44.866 -> BPM: 186
17:51:45.120 -> ♥  A HeartBeat Happened ! 
17:51:45.167 -> BPM: 184
17:51:45.422 -> ♥  A HeartBeat Happened ! 
17:51:45.468 -> BPM: 208
17:51:45.669 -> ♥  A HeartBeat Happened ! 
17:51:45.722 -> BPM: 221
17:51:45.923 -> ♥  A HeartBeat Happened ! 
17:51:45.923 -> BPM: 222
17:51:46.171 -> ♥  A HeartBeat Happened ! 
17:51:46.224 -> BPM: 222
17:51:46.425 -> ♥  A HeartBeat Happened ! 
17:51:46.472 -> BPM: 222
17:51:46.672 -> ♥  A HeartBeat Happened ! 
17:51:46.725 -> BPM: 229
17:51:47.577 -> ♥  A HeartBeat Happened ! 
17:51:47.635 -> BPM: 185
17:51:48.132 -> ♥  A HeartBeat Happened ! 
17:51:48.179 -> BPM: 170
17:51:48.433 -> ♥  A HeartBeat Happened ! 
17:51:48.480 -> BPM: 166
17:51:48.834 -> ♥  A HeartBeat Happened ! 
17:51:48.834 -> BPM: 163
17:51:49.082 -> ♥  A HeartBeat Happened ! 
17:51:49.082 -> BPM: 164
17:51:49.316 -> ♥  A HeartBeat Happened ! 
17:51:49.351 -> BPM: 164
17:51:49.583 -> ♥  A HeartBeat Happened ! 
17:51:49.637 -> BPM: 164
17:51:49.837 -> ♥  A HeartBeat Happened ! 
17:51:49.837 -> BPM: 164
17:51:50.084 -> ♥  A HeartBeat Happened ! 
17:51:50.138 -> BPM: 164
17:51:50.385 -> ♥  A HeartBeat Happened ! 
17:51:50.438 -> BPM: 163
17:51:50.639 -> ♥  A HeartBeat Happened ! 
17:51:50.639 -> BPM: 197
17:51:50.886 -> ♥  A HeartBeat Happened ! 
17:51:50.886 -> BPM: 218
17:51:51.274 -> ♥  A HeartBeat Happened ! 
17:51:51.321 -> BPM: 213
17:51:51.542 -> ♥  A HeartBeat Happened ! 
17:51:51.542 -> BPM: 221
17:51:51.805 -> ♥  A HeartBeat Happened ! 
17:51:51.805 -> BPM: 221
17:51:52.042 -> ♥  A HeartBeat Happened ! 
17:51:52.089 -> BPM: 221
17:51:52.289 -> ♥  A HeartBeat Happened ! 
17:51:52.342 -> BPM: 221
17:51:52.543 -> ♥  A HeartBeat Happened ! 
17:51:52.590 -> BPM: 220
17:51:52.922 -> ♥  A HeartBeat Happened ! 
17:51:52.960 -> BPM: 212
17:51:53.145 -> ♥  A HeartBeat Happened ! 
17:51:53.192 -> BPM: 216
17:51:53.445 -> ♥  A HeartBeat Happened ! 
17:51:53.445 -> BPM: 216
17:51:53.946 -> ♥  A HeartBeat Happened ! 
17:51:53.993 -> BPM: 197
17:51:54.247 -> ♥  A HeartBeat Happened ! 
17:51:54.294 -> BPM: 202
17:51:54.495 -> ♥  A HeartBeat Happened ! 
17:51:54.548 -> BPM: 202
17:51:55.091 -> ♥  A HeartBeat Happened ! 
17:51:55.150 -> BPM: 181
17:51:55.450 -> ♥  A HeartBeat Happened ! 
17:51:55.497 -> BPM: 176
17:51:55.852 -> ♥  A HeartBeat Happened ! 
17:51:55.852 -> BPM: 169
17:51:56.099 -> ♥  A HeartBeat Happened ! 
17:51:56.152 -> BPM: 170
17:51:56.500 -> ♥  A HeartBeat Happened ! 
17:51:56.554 -> BPM: 166
17:51:56.801 -> ♥  A HeartBeat Happened ! 
17:51:56.801 -> BPM: 166
17:51:57.054 -> ♥  A HeartBeat Happened ! 
17:51:57.054 -> BPM: 166
17:51:57.302 -> ♥  A HeartBeat Happened ! 
17:51:57.355 -> BPM: 179
17:51:57.603 -> ♥  A HeartBeat Happened ! 
17:51:57.656 -> BPM: 179
17:51:57.957 -> ♥  A HeartBeat Happened ! 
17:51:58.004 -> BPM: 175
17:51:58.204 -> ♥  A HeartBeat Happened ! 
17:51:58.204 -> BPM: 194
17:51:58.458 -> ♥  A HeartBeat Happened ! 
17:51:58.505 -> BPM: 201
17:51:59.060 -> ♥  A HeartBeat Happened ! 
17:51:59.107 -> BPM: 186
17:51:59.461 -> ♥  A HeartBeat Happened ! 
17:51:59.508 -> BPM: 179
17:51:59.709 -> ♥  A HeartBeat Happened ! 
17:51:59.709 -> BPM: 189
17:52:00.063 -> ♥  A HeartBeat Happened ! 
17:52:00.110 -> BPM: 184
17:52:00.311 -> ♥  A HeartBeat Happened ! 
17:52:00.364 -> BPM: 184
17:52:00.565 -> ♥  A HeartBeat Happened ! 
17:52:00.612 -> BPM: 184
17:52:00.812 -> ♥  A HeartBeat Happened ! 
17:52:00.866 -> BPM: 188
17:52:01.066 -> ♥  A HeartBeat Happened ! 
17:52:01.113 -> BPM: 192
17:52:01.329 -> ♥  A HeartBeat Happened ! 
17:52:01.367 -> BPM: 192
17:52:01.567 -> ♥  A HeartBeat Happened ! 
17:52:01.624 -> BPM: 192
17:52:01.815 -> ♥  A HeartBeat Happened ! 
17:52:01.868 -> BPM: 218
17:52:02.068 -> ♥  A HeartBeat Happened ! 
17:52:02.115 -> BPM: 229
17:52:02.469 -> ♥  A HeartBeat Happened ! 
17:52:02.469 -> BPM: 220
17:52:02.670 -> ♥  A HeartBeat Happened ! 
17:52:02.717 -> BPM: 229
17:52:02.918 -> ♥  A HeartBeat Happened ! 
17:52:02.971 -> BPM: 229
17:52:03.166 -> ♥  A HeartBeat Happened ! 
17:52:03.206 -> BPM: 229
17:52:03.469 -> ♥  A HeartBeat Happened ! 
17:52:03.469 -> BPM: 229

I appreciate if you could support me 

Best Regards,

biomurph, 2021-06-02:
@yboujraf 
Please shorten the length of your last post to make it easy to process this issue thread.

Please program the PulseSensor_BPM.ino sketch example. It sends the pulse data to the Arduino Serial Plotter.
Then open the Arduino serial plotter and send a screenshot.

Also, please send a photo of your physical setup with the Pulse Sensor clearly visible.

yboujraf, 2021-06-02:
Dear @biomurph 

Thanks for your quick reply.

Please find below two screenshots :

1. with finger 
[image]

2. no finger

[image]

3. the schematic

[image]

[image]
[image]
[image]

Best Regards

biomurph, 2021-06-02:
@yboujraf 
You are unfortunately using a knockoff Pulse Sensor. I can tell by the way the board is parted and the large vias.
Our Pulse Sensor also comes with Purple, Red and Black wires.
I can't really troubleshoot the noise that you're seeing because I don't know the hardware.
It looks like there is a pulse wave in the signal, but our algorithm is having trouble finding it.

Please support original makers of Pulse Sensor and purchase legit parts.
You can get Pulse Sensor at our website www.pulsesensor.com
Also,
Adafruit
Sparkfun
Mouser
Digikey

Where did you get yours?

yboujraf, 2021-06-03:
Dear @biomurph 

Just received my new sensor this morning.
I have tested with Arduino UNO and now I have my pulse :

[image]

https://user-images.githubusercontent.com/6151258/120633978-0b983000-c46b-11eb-9433-43ded001ace6.MOV

Will test with Arduino pro mini 3V/8MHz this afternoon and will let you know.

Best Regards

biomurph, 2021-06-03:
Looks much better!
The clipping is normal. If you change the finger pressure, you will see the amplitude of the pulse wave change.

yboujraf, 2021-06-03:
Dear @biomurph 

knowing that I made a connected watch. Is the wrist of the hand that would be used to measure the pulses? I noticed that the level was not terrible. do you have any idea how to amplify the signal? or is this sensor not suitable?

best regards

biomurph, 2021-06-03:
You will have to test and see how the signal works for you in your application.
I have been able to get good signal from the wrist using PPG and using Pulse Sensor.
Yes, you could add more hardware to amplify the signal.

yboujraf, 2021-06-03:
Dear @biomurph 

Finally, when connected to an arduino pro mini 3V/8MHz, I still have the same issue.
But if using the same pulse sensor on an arduino UNO R3, we get the coorect pulse as shown on my previous email.

Do I need to change anything in the code to fix this ?

Best Regards

yboujraf, 2021-06-03:
Dear @biomurph 

Regarding the THRESHOLD value, do I need to decrease or increase the value ?

Best Regards

biomurph, 2021-06-03:
@yboujraf you would need to test with different values.
If you do, you should use numbers over 512, as the signal settles to 512 when the signal is idle.
The threshold is the ADC value that the signal has to cross in order to qualify as a beat (along with some other requirements of the algorithm).

The threshold *should* self-adjust when beats are happening consistently. When there is no beat for 2.5 seconds, the threshold *should* be reset to the default (or what you set it to)

yboujraf, 2021-06-03:
Dear @biomurph 

These are screenshot of the pulse with arduino pro mini 3v/8MHz

no finger :

[image]

With finger : 
[image]

I tried to adjust the threshold but nothing changed.

same sensor on arduino uno r3 and it works.

question, we added a Bluetooth transmitter. Could this interface could interfere with the sensor ?

With uno we used 5v, pro mini we used 3.3v or battery.

This is the homework of my son expect to use the sensor and using Bluetooth to transmit to a base central where  all the sensors value are displayed to a waveform. Now two weeks without success even we replaced all the items and build new board.

best regards

yboujraf, 2021-06-03:
Dear @biomurph 

insee the green led is blinking?

Uploading IMG_0020.MOV…

Best regards

biomurph, 2021-06-03:
@yboujraf 
Oh my. 
Can you try using the ProMini with no bluetooth?
Should for sure be OK with the 3.3V and the 5V no problem. 
It appears that the bluetooth might be causing some kind of current spike?

yboujraf, 2021-06-04:
Dear @biomurph 

I think I found the issue.
This is the USB adapter used to upload the code and using as power supply.

=> I can see the green led is blinking.

if I use my battery to supply the arduino, no more blinking and value are correct.

[image]

now I d like to zoom a little bit the pulse to show the heartbeat pulse better.

To display the heartbeat I am doing this :

oled : 128x64

calculate pixel :

y = 62 - (analog a1 value / 16)
For x I am not doing anything just x +=x.

maybe with a better formula will be more accurate or the sample rate is too small.

we are transmitting at 9600 baudrate through Bluetooth.

best regards

yboujraf, 2021-06-04:
> You will have to test and see how the signal works for you in your application.
> I have been able to get good signal from the wrist using PPG and using Pulse Sensor.
> Yes, you could add more hardware to amplify the signal.

PPG ?

biomurph, 2021-06-04:
Photophlesmography
https://en.m.wikipedia.org/wiki/Photoplethysmogram
It's what the pulse Sensor does. We add some hardware signal conditioning, but that's basically it.

Can you share your code? Harder to help with out of context snippets

yboujraf, 2021-06-04:
Dear @biomurph 

Please find the source code.
Need to clean and comment it.

FYI with wrist, we have a pulse but very low.

Best regards,

[210604 - BASE_CENTRALE_MONTRE_CONNECTEE_INRACI_RELEASE.zip](https://github.com/WorldFamousElectronics/PulseSensorPlayground/files/6596202/210604.-.BASE_CENTRALE_MONTRE_CONNECTEE_INRACI_RELEASE.zip)

yboujraf, 2021-06-04:
Dear @biomurph 

i bought also max30102 but not find the correct lib or issue when compile.

Otherwise if you know a sugar sensor for glicemy will be nice.

best regards

biomurph, 2021-06-04:
@yboujraf I think I see what you're doing with the OLED display.
By dividing down the pulse signal by 16, your range is becoming 0 - 63 instead of 0 - 1023.
That could be effecting the look of the data you're showing. 
You could try to go back to the basic example sketch and do the divide by 16 and then display it on the Arduino serial plotter for a sanity check.
I think that if you expand your 'x' you may get the kind of 'zoom' that you are looking for?
Try `x += 3` or something?

Yes, pulse on the wrist will be lower and a bit noisier. You may need a tight fit.

The only blood sugar sensor I know of is invasive (implant)

Your son will likely get an A on this homework. I hope he's having a chance to do some code too ;)

yboujraf, 2021-06-04:
Dear @biomurph 

Now the pulse is better.

Question, we installed the pulse sensor in a box with Arduino, Bluetooth oled and the pulse is perfect but the BPM value is not matching.

The threshold = 520 when the sensor is outside the box and the two fingers press the sensor.
When included in the box, the BPM is 2x times higher.

Best Regards

biomurph, 2021-06-04:
Make sure that you are getting good close contact between the Pulse Sensor and the fingertip or other body part.

yboujraf, 2021-06-05:
Dear @biomurph 
Do you think the hole needs to be greater ?

Best regards

biomurph, 2021-06-05:
@yboujraf 
I will leave that design decision up to you and your team.
If you are getting good values with the Pulse Sensor in one scenario, but not good in another scenario, the troubleshooting should be straight forward.

yboujraf, 2021-06-08:
@biomurph 

When we designed the box, the pulse sensor is behind the Bluetooth HC05.
I suspect the transmission of datas is interfering with the pulse sensor.

Best regards

closed 2023-10-26 by biomurph