WorldFamousElectronics/PulseSensorPlayground #142: Lines commented

Leoriem-code, 2021-06-12:
Don't know if that's where I should put it but in the 1.5.1 version, in the interrupts.h file, the line 272 and 318 are commented. A such, when using Arduino board not defined before, we receive an error.

biomurph, 2021-06-12:
For the Arduino Nano 33 BLE, you need to use the PulseSensor_BPM_Alternative.ino example sketch

closed 2021-06-12 by biomurph

Leoriem-code, 2021-06-12:
> 
> 
> For the Arduino Nano 33 BLE, you need to use the PulseSensor_BPM_Alternative.ino example sketch

Doesn't change anything. The problem is with the lines 72 and 73 where cli() and sei() are not declared.
I used PulseSensor_BPM_Alternative.ino without touching it (my wiring was already correct) and I get the error that I reported

biomurph, 2021-06-12:
@Leoriem-code that is odd. I have just compiled the Alternative sketch for the Arduino Nano 33 BLE and I don't get the same error. 

Stand by

biomurph, 2021-06-12:
Please delete the library that you have on your computer, and download the current version directly from this repository using the CODE button and selecting the Download ZIP option.

Leoriem-code, 2021-06-12:
It has compiled correctly, thank you