WorldFamousElectronics/PulseSensorPlayground #144: Generating Comparable Amplitudes with Two Pulse Sensors

kjolly02, 2021-08-13:
Hello,

I am currently working on a project that incorporates two pulse sensors. We are trying to compare whether one artery has greater blood flow than its counterpart on the other side of the body (i.e. left vs right sides of the body). However, when testing with our apparatus, one sensor consistently results in higher amplitude readings than another. 

What factors go into changing the amplitude of the readings... does greater blood flow result in higher readings? Is there a way to calibrate the sensors to start at some common baseline?

Thanks in advance for any help.

biomurph, 2021-08-13:
I don't think that it's possible to measure 'blood flow' with the Pulse Sensor. The Pulse Sensor functions as a PPG sensor. It measures the relative change in tissue density on the site under the sensor. So it depends on the site as to what kind of density change you will get.

Also, the Pulse Sensor is not a calibrated device. There may be great variation between different Pulse Sensors!
If you are curious about how the Pulse Sensors you have correlate to each other, try putting them on the same body site and see how the signal may or may not differ. That might be a simple way to get closer to a kind of calibration, even if you just understand how the difference is on different parts. 

But again, I am concerned that the premise of your post. Do you have any data on how a PPG can be used to measure volume of blood flow?

kjolly02, 2021-08-14:
Thanks so much for your response. If we are looking at the difference in tissue density, would increased blood flow to that area result in a change there? Is there a relationship between tissue density and the amplitude of the signal?

And as for figuring out the variation between our two sensors, that is a great idea, thank you. We will try that out and figure out if there may just be some inherent difference between the signal even if placed in the same location.